iiab/iiab #6: apache user vs data

tim-moody, 2017-06-02:
<OS>.yml var files have both apache user and apache data. what is the difference

for debian data is www-data and user is apache

georgejhunt, 2017-06-04:
Nosing around a little, {{ apache_user }} and {{ apache_data }} may be duplicated due to my confusion. I see that apache has user id 1001 on a raspbian machine.  I assumed that user apache was created by one of the packages we load, but it may be created by our playbook itself.  This will be hard to track down, and test, because we'll need to test across all platforms -- I'd rather not do for adam's imminent 6.3.

holta, 2017-06-04:
@georgejhunt 6.3 final release doesn't have to be imminent in early/mid-June, unless we choose it to be so, during tmrw's call -- or a few weeks later if that's best possibly?  Either way is fine, and whenever George/All choose to work on {{ apache_user }} versus {{ apache_data}} reconciliation, great!

SEPARATELY: what is imminent is that many deployments will occur across the Caribbean, Central America and West Africa in coming Days, and Weeks, regardless what we decide tmrw -- and these folks will (generally, at their own risk as always!) use whatever 6.3 pre-release they can get rolling -- and for good reason, as this is already so much more solid than 6.2, thanks to some very impressive work from all corners~

jvonau, 2017-07-04:
Think this is resolved and can be closed if you agree

closed 2017-07-05 by georgejhunt