iiab/iiab #10: Moodle forces erroneous redirect to schoolserver.lan/moodle (bad enough on LAN, worse on WAN)

holta, 2017-06-03:
Regardless of IIAB's host name (schoolserver.lan is deprecated in favor of box.lan, and can also be further customized beyond that) &mdash; and regardless of whether user is coming from the LAN or WAN of IIAB server &mdash; Moodle incorrectly redirects to http://schoolserver.lan/moodle in all conditions.

_Crude Interim Fix:_
Moodle could be hard-coded to redirect to http://box.lan/moodle/*

_Improved Interim Fix:_
Moodle could be hard-coded to redirect to http://172.18.96.1/moodle/*

_Better Interim Fix:_
Moodle could be redirected to http://<actual-server-host-name.lan>/moodle/*

_Real Fix:_
Moodle really should retain http://<IP-or-Name-specified-by-the-user>/moodle/* in all conditions, for WAN (campus) users as well as LAN (local hotspot) users.

georgejhunt, 2017-06-04:
Could this be fixed by letting an apache proxy rewrite rule delegate to localhost, and change the moodle config file specify localhost?

jvonau, 2017-06-04:
Better Interim Fix #17

holta, 2017-06-05:
Aside, but incredibly relevant+similar:

WordPress 4.7.3 (which instantly/forcibly auto-upgrades you to 4.7.5 in IIAB 6.3 currently, when online anyway) has a similar failure to Moodle, if not QUITE so severe -- as the WordPress failure only occurs on the WAN a.k.a. campus side of the server.

Consequence: WAN-side "campus" users visiting http://wan-ip-address/wordpress get an absolute mess of a page -- as most of its links/content depend on the inaccessible http://box/wordpress/*

(FWIW a couple of the links on the above page DO appear "correctly" as http://wan-ip-address/wordpress/* interestingly, so possibly/perhaps this too is hopefully fixable too, eventually!!)

tim-moody, 2017-06-05:
> wordpress get an absolute mess of a page -- as most of its links/content

I think owncloud will have similar problems.

jvonau, 2017-06-05:
The upgrade issue could be resolved by updating the the latest upstream
release, or disable the auto updater.  The main problem with any of the
php/db based CMS is that the solutions are based on a server with only a
single network interface. This problem is exasperated by saving the full
request url in the db backend. One solution is to have an upstream DNS
server like dyndns or cooperation from the WAN's admin resolve to the same
hostname.domainname as used on the service hosting box(xsce) but that would
require the use of a real domainname on the LAN not '.lan' along with the
hostname matching what is in upstream dns.

jvonau, 2017-09-11:
Think moodle's 'Real Fix:' from above, would be cleaner to deal with once nginx is integrated. #224

holta, 2019-10-09:
This is basically a duplicate of #941 so am closing.

closed 2019-10-09 by holta