iiab/iiab #8: box and box.lan need to work on "all" IIAB's, regardless of local_vars.html

holta, 2017-06-03:
http://box does not show up until forced thru local_vars.html or similar.  Hence all documentations's is flawed, as all our documentation assumes newbies instantly have access to http://box/admin (etc).

So let's make sure "box" and "box.lan" work truly instantly, no matter what (on Raspbian especially :)

holta, 2017-06-04:
Given that this bug renders most all documentation nearly useless to newbies (who seek to use http://box/admin etc) I'd suggest we change Line 46 of https://github.com/iiab/iiab/blob/master/vars/default_vars.yml as immediately as possible.

Line 46 currently reads:
xsce_hostname: schoolserver

It should read:
xsce_hostname: box

Or is there a better way?

holta, 2017-06-04:
@georgejhunt @tim-moody @jvonau

In the past these 3 files seemed to work:
https://github.com/iiab/iiab/blob/master/roles/network/templates/named/school.external.zone.db
https://github.com/iiab/iiab/blob/master/roles/network/templates/named/school.internal.zone.db
https://github.com/iiab/iiab/blob/master/roles/network/templates/named/school.local.zone.db

As documented in FAQ #13 @ http://FAQ.IIAB.IO
("Can I name my server something other than http://BOX.LAN ?")

However I can reconfirm this is not working on CentOS and Raspbian today (newbies cannot access http://box/admin etc, with a strong guarantee that this will work in all conditions, regardless of local_vars.yml).  Changing Line 46 of default_vars.yml would seem to be an immediate fix (to "xsce_hostname: box") but others may know a better way?

jvonau, 2017-06-04:
How are you accessing the webserver? From a remote client browser or from the server itself?

holta, 2017-06-04:
Clarif for @jvonau :

box and box.lan fail to work from the server itself :/

jvonau, 2017-06-04:
Yea I suspect that /etc/resolv.conf is wrong

jvonau, 2017-06-04:
bit of background reading: https://serverfault.com/questions/513271/why-dont-my-server-query-the-2nd-entry-in-my-resolv-conf 

The only nameserver that can resolve "box" is local so one would need to have it's ip address(I like to use 127.0.0.1 but 172.18.96.1 should work if br0 is present).

holta, 2017-06-04:
Thx @jvonau noting that box and box.lan are entirely missing from /etc/resolv.conf on Raspbian and CentOS, for reasons I do not understand.  I (really thought) this used to work, and it had been documented many months ago @ http://FAQ.IIAB.IO (number 13)

Certainly I get the following on CentOS:

# more /etc/hosts
127.0.0.1            localhost.localdomain   localhost holt-content.lan holt-content
::1         localhost localhost.localdomain localhost6 localhost6.localdomain6

And more recently on the same system resolv.conf has changed completely for reasons I don't know (I did not do an additional ./runansible, mystifying!) but FWIW resolve.conf now appears in this new form after sev reboots:

# more /etc/resolv.conf
# Generated by NetworkManager
search lan
nameserver 127.0.0.1

PS @georgejhunt may understand more why /etc/resolv.conf may be flip-flopping (if that's indeed the case!) as he helped me upgrade this system from 6.1 to 6.2, and knows infinitely more than I do on such topics :)

holta, 2017-06-04:
Oops @jvonau @georgejhunt I forgot to click "Insert code" above.
All fixed now (GitHub WikiText is no longer mangling the 2 snippets I posted above).

holta, 2017-06-04:
Ref @jvonau's #16 idea. possibly updating /etc/hosts with an alias (also a great idea, so newbies who override their hostname don't suddenly lose access to http://box/info documentation!)

jvonau, 2017-06-04:
For CentOS I would ensure that 'box' is present in the zone files mentioned above, but I can see a weakness in appliance mode where we are using 172. but may not actually have a second interface, that would cause a lookup failure.

jvonau, 2017-07-04:
Issue resolved?

holta, 2017-07-07:
> Issue resolved?

@jvonau it's looking much better than a month ago, even if I'd love to know with higher confidence that all IIAB operators (across most OS's) can change their Host Name and still be able to access docs @ http://box/info and Admin Console @ http://box/admin

If everyone perceives we're 99% of the way there, then yeah let's close this issue during Monday's call (July 10)

holta, 2017-09-02:
This seems 99% solved.

closed 2017-09-02 by holta

mitra42, 2019-04-24:
Note - I've never seen box or box.lan show up, it always comes up as box.local for me - both connected over the ethernet or to the WiFi. (I'm on a Mac, same thing happens on my iphone) - the assumption of box or box.lan is definitely situation specific.

ubuntupunk, 2024-08-04:
my pi install of iiab was available via box.local, now its only available via its IP address 192.168.10.29, what are the recommended settings for resolv.conf and /etc/hosts?

holta, 2024-08-04:
1. @ubuntupunk can you please run...

   `sudo iiab-diagnostics`

4. And then reply here, with the resulting **dpaste.com** URL?

Thanks!!

ubuntupunk, 2024-09-04:
using avahi-discover iiab shows up as linux.local for some reason:
```
Service Type: _ssh._tcp
Service Name: linux
Domain Name: local
Interface: enp2s0 IPv4
Address: linux.local/192.168.10.29:22
TXT Data: empty
```
Here is the pastebin diagnostic for what its worth https://dpaste.org/XAu5K

holta, 2024-09-04:
Thanks @ubuntupunk and that looks pretty normal!

Quick questions:

1. Would you be able to run `curl iiab.ip/install.txt | bash` on a clean/fresh install of the very latest (ideally 64-bit) Raspberry Pi OS?

2. Can your say more about what makes resolv.conf and /etc/hosts important?  e.g. are you trying to accomplish something specific with name resolution, or naming of the Raspberry Pi?