iiab/iiab #9: http://box/info does not show documentation

holta, 2017-06-03:
http://box/info does not show our great offline documentation on new installs.

That is until I manually run "cd /opt/schoolserver/iiab/scripts; ./refresh-wiki-docs.sh"

How should we enforce [programmatically, even if just socially for now?] that this is run during all installs?

PS box/info IT/Support docs originated in IIAB/XSCE 6.2.  This issue is slightly exacerbated by Issue #8 (box and box.lan don't appear on many fresh installations, per https://github.com/iiab/iiab/issues/8) but really that is separate!

georgejhunt, 2017-06-04:
proposal to move /scripts/refresh-wiki-docs to roles/3-base-server/templates, run it during install, and copy it to /usr/bin/xs-refresh-dos for later end user update.
maybe all the xs-* binaries should be replaced with iiab-*

holta, 2017-06-04:
Either xs-refresh-docs or iiab-refresh-docs are A-Ok with me!

As Tim outlined Thursday, there are many cases where "xs" and "xsce" should eventually be replaced wholesale with "iiab" in a methodical way, rather than in a piecemeal fashion?

closed 2017-06-13 by holta