iiab/iiab #11: rpi not detected

tim-moody, 2017-06-04:
detection uses
"ansible_cmdline": {
    "dwc_otg.lpm_enable": "0",
    "console": "tty1",
    "root": "PARTUUID=e00409af-02",
    "bcm2708_fb.fbwidth": "656",
    "rootwait": true,
    "vc_mem.mem_size": "0x3f000000",
    "8250.nr_uarts": "0",
    "fsck.repair": "yes",
    "elevator": "deadline",
    "bcm2708_fb.fbheight": "416",
    "vc_mem.mem_base": "0x3ea00000",
    "bcm2708_fb.fbdepth": "16",
    "rootfstype": "ext4",
    "bcm2708_fb.fbswap": "1"
  },

when:  ansible_cmdline["bcm2709.serial"]  is defined

but this may not be true for some rpi 3

tim-moody, 2017-06-04:
what about  "ansible_lsb":   "id": "Raspbian"

jvonau, 2017-07-04:
#58

closed 2017-07-04 by tim-moody