iiab/iiab #13: rpi clock setup require change to local_vars

tim-moody, 2017-06-04:
rtc_id: ds3231

must be in local_vars.yml or clock not setup

tim-moody, 2017-06-04:
we should have a Raspbian.yml vars file that takes care of this
can still be overridden in local_vars

tim-moody, 2017-06-04:
looks like ansible_distribution is Debian on rpi rather than Raspian and ansible_lsb['id'] is not defined on Centos!

tim-moody, 2017-06-04:
actually on Centos 7.3 I get

  },
        "ansible_lsb": {
            "codename": "Core",
            "description": "CentOS Linux release 7.3.1611 (Core)",
            "id": "CentOS",
            "major_release": "7",
            "release": "7.3.1611"

so using ansible_lsb id should work

georgejhunt, 2017-06-04:
There are many places in the playbook where we use ansible_distribution. I'd be in favor of changing them all  to ansible_lsb.id (smoke test required, I think)

tim-moody, 2017-06-04:
I am working on this.

I think we should start with xsce.yml and OS.yml

later we should find ansible_distribution, presumably when: ansible_distribution == OS and change to use our is_OS flags

tim-moody, 2017-06-04:
Another question is whether we should provide a default clock chip, which we didn't do in the past for some reason, e.g. 

rtc_id: ds3231

georgejhunt, 2017-06-04:
no reason not to default to the current rtc_id.  +1

holta, 2017-06-04:
DS3231 is the most common.

But we should support DS1307 as well: what instructions should I add to http://FAQ.IIAB.IO ?

(In 2016 I tested @georgejhunt's code worked with both chips, confirming the time was retained after yanking out Ethernet and rebooting: very oddly both worked!)

jvonau, 2017-07-04:
[second commit](https://github.com/iiab/iiab/pull/58/commits/9ea621297c88561691d1ba38a0c8a7587b707a12) of #58

tim-moody, 2017-07-04:
probably the default could go in raspian.yml, since we assume that os="raspian" is a marker for an rpi.

jvonau, 2017-07-04:
#58 pushed, this issue can be closed.

closed 2017-07-05 by georgejhunt