iiab/iiab #30: kiwix did not install

tim-moody, 2017-06-13:
on laptop with debian using http://download.iiab.io/6.3/rpi/load.txt

TASK [kiwix : Set kiwix source file name x86_64] *******************************
ok: [127.0.0.1]

TASK [kiwix : Set kiwix source file name armv7l] *******************************
skipping: [127.0.0.1]

TASK [kiwix : Get the kiwix software] ******************************************
changed: [127.0.0.1]

TASK [kiwix : Create various directories for zims] *****************************
ok: [127.0.0.1] => (item=/library/zims)
ok: [127.0.0.1] => (item=/library/zims/content)
ok: [127.0.0.1] => (item=/library/zims/index)

TASK [kiwix : Copy kiwix library file if needed] *******************************
ok: [127.0.0.1] => (item=library.xml)

TASK [kiwix : Set kiwix first pass] ********************************************
skipping: [127.0.0.1]

TASK [kiwix : debug] ***********************************************************
ok: [127.0.0.1] => {
    "kiwix_first_pass_test": {
        "changed": false,
        "msg": "All items completed",
        "results": [
            {
                "_ansible_item_result": true,
                "_ansible_no_log": false,
                "_ansible_parsed": true,
                "changed": false,
                "diff": {
                    "after": {
                        "path": "/library/zims/library.xml"
                    },
                    "before": {
                        "path": "/library/zims/library.xml"
                    }
                },
                "gid": 0,
                "group": "root",
                "invocation": {
                    "module_args": {
                        "backup": null,
                        "content": null,
                        "delimiter": null,
                        "dest": "/library/zims/library.xml",
                        "diff_peek": null,
                        "directory_mode": null,
                        "follow": true,
                        "force": false,
                        "group": "root",
                        "mode": "0644",
                        "original_basename": "library.xml",
                        "owner": "root",
                        "path": "/library/zims/library.xml",
                        "recurse": false,
                        "regexp": null,
                        "remote_src": null,
                        "selevel": null,
                        "serole": null,
                        "setype": null,
                        "seuser": null,
                        "src": null,
                        "state": null,
                        "unsafe_writes": null,
                        "validate": null
                    }
                },
                "item": "library.xml",
                "mode": "0644",
                "owner": "root",
                "path": "/library/zims/library.xml",
                "size": 70260,
                "state": "file",
                "uid": 0
            }
        ]
    }
}

TASK [kiwix : debug] ***********************************************************
ok: [127.0.0.1] => {
    "kiwix_first_pass": false
}

TASK [kiwix : Copy test.zim file] **********************************************
skipping: [127.0.0.1]

TASK [kiwix : Unarchive it to permanent location - not bin_only] ***************
skipping: [127.0.0.1]

TASK [kiwix : Create directory for kiwix bin] **********************************
changed: [127.0.0.1]

TASK [kiwix : enable the mods which permit apache to proxy] ********************
ok: [127.0.0.1] => (item=proxy)
ok: [127.0.0.1] => (item=proxy_html)
ok: [127.0.0.1] => (item=rewrite)

TASK [kiwix : Unarchive it to permanent location - bin only] *******************
skipping: [127.0.0.1]

TASK [kiwix : Set kiwix ownership] *********************************************
changed: [127.0.0.1]
 [WARNING]: Consider using file module with owner rather than running chown

TASK [kiwix : Make an entry in crontab to restart every hour] ******************
ok: [127.0.0.1]

TASK [kiwix : Make an entry in crontab to restart every hour] ******************
skipping: [127.0.0.1]

TASK [kiwix : Create kiwix-serve service] **************************************
ok: [127.0.0.1] => (item={u'dest': u'/etc/systemd/system/kiwix-serve.service', u'src': u'kiwix-serve.service.j2', u'mode': u'0655'})
ok: [127.0.0.1] => (item={u'dest': u'/usr/bin/iiab-make-kiwix-lib', u'src': u'iiab-make-kiwix-lib', u'mode': u'0755'})
ok: [127.0.0.1] => (item={u'dest': u'/usr/bin/iiab-make-kiwix-lib.py', u'src': u'iiab-make-kiwix-lib.py', u'mode': u'0755'})
ok: [127.0.0.1] => (item={u'dest': u'/usr/bin/iiab-make-apache-config.py', u'src': u'iiab-make-apache-config.py', u'mode': u'0755'})

TASK [kiwix : add kiwix to service list] ***************************************
ok: [127.0.0.1] => (item={u'option': u'name', u'value': u'kiwix-serve'})
ok: [127.0.0.1] => (item={u'option': u'description', u'value': u'"kiwix-serve is a web server for zim files"'})
ok: [127.0.0.1] => (item={u'option': u'kiwix_url', u'value': u'/kiwix'})
ok: [127.0.0.1] => (item={u'option': u'kiwix_path', u'value': u'/opt/iiab/kiwix'})
ok: [127.0.0.1] => (item={u'option': u'kiwix_port', u'value': 3000})
ok: [127.0.0.1] => (item={u'option': u'iiab_zim_path', u'value': u'/library/zims'})
ok: [127.0.0.1] => (item={u'option': u'kiwix_library_xml', u'value': u'/library/zims/library.xml'})
ok: [127.0.0.1] => (item={u'option': u'kiwix_content_path', u'value': u'/library/zims/content'})
ok: [127.0.0.1] => (item={u'option': u'enabled', u'value': True})

TASK [kiwix : Enable kiwix-serve service] **************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "Unable to restart service kiwix-serve: Job for kiwix-serve.service failed. See 'systemctl status kiwix-serve.service' and 'journalctl -xn' for details.\n"}
        to retry, use: --limit @/opt/iiab/iiab/iiab.retry

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=436  changed=115  unreachable=0    failed=1

tim-moody, 2017-06-13:
/opt/iiab/kiwix/bin exits but is empty

georgejhunt, 2017-06-13:
I had a similar problem, and discovered that flag no_network was true --
because heartbeat.txt at unleaskids was missing.
I notice that kiwix_first_pass is false. That does not seem right if the
bin/ is empty

tim-moody, 2017-06-13:
I reran on a fresh vm and did not have this problem. am wiping the laptop and rerunning

holta, 2017-07-07:
@tim-moody can this issue be closed?

ASIDE: When should be upgrade kiwix-serve to the latest, in your opinion?  (My understanding is that most of us are running a June 13th build, which may well be sufficient for IIAB 6.3 etc!)

tim-moody, 2017-07-07:
I have not seen this since the issue was created, so closing

closed 2017-07-07 by tim-moody