iiab/iiab #32: http://box/info docs/links missing

holta, 2017-06-13:
(1) These are Missing, even if URL's are capitalized as XCSE/xsce :

https://github.com/xsce/xsce/wiki/IIAB-6.2-Release-Notes
https://github.com/xsce/xsce/blob/release-6.2/ReleaseNotes6.1.md
https://github.com/xsce/xsce/blob/release-6.2/ReleaseNotes6.0.md

(2) http://box/info contains quite a lot of #anchor links like the following:
   http://box/info/IIAB-Platforms#disk-partitioning.html

The correct (working) link should be:
   http://box/info/IIAB-Platforms.html#disk-partitioning

(2) The new /usr/bin/iiab-refresh-docs (xs-refresh-wiki-docs for now, runs during online installs.o) should replace all references such as...

   http://FAQ.IIAB.IO
   http://schoolserver.org/FAQ
   http://wiki.laptop.org/go/IIAB/FAQ
   http://wiki.laptop.org/go/XS_Community_Edition/FAQ
With the following instead...
   http://box/info/html/FAQ

These other 2 files aren't referenced nearly as much yet?
   http://box/info/html/Security
   http://box/info/html/local_vars.yml

Anchor tags (deep links) to content *within* the FAQ should begin to work automagically!

3) Can/should we changed these to iiab-* ?

   xs-check-activities       xs-httpcache              xs-regenerate-activities
   xs-gen-iptables           xs-network-reset          xs-vpn
   xs-handle                 xs-refresh-wiki-docs

PS Builds off these 2 great work beefing up http://box/info already: #9 #31

holta, 2017-06-15:
Looks like @georgejhunt has already made tremendous progress here, FYI within PR #37

holta, 2017-06-20:
Just pasting in so more than just @georgejhunt see the evolving progress-

George,

1. Can you help fix these 1+3 links at the bottom of http://box/info ?

a) FAQ (Frequently Asked Questions) points to:
http://wiki.laptop.org/go/IIAB/FAQ
20min ago it correctly pointed to:
http://box/info/html/FAQ

b) Release Notes {6.2, 6.1, 6.0} links (bottom 3 links on main page box/info) incorrectly points to ONline docs.
20 min they correctly pointed to:
http://box/info/IIAB-6.2-Release-Notes.html
http://box/info/ReleaseNotes6.1.html
http://box/info/ReleaseNotes6.0.html

/usr/bin/xs-refresh-wiki-docs was not changed and something inside it is flaky, producing variable & incorrect output only 20 min later, despite nothing changing (no git pulls, no runansible's, nothing).  Even prior to the script intermittently (or worse consistently now) producing very stale output [incorrect links it's producing above] it had these 2 separate issues which can be very quick fixes / very quick wins:

2. Can you possibly convert online URL's like:
http://box/go/IIAB/FAQ#What_are_the_default_passwords.3F
(e.g. this is the top link within http://box/info/html/Security.html)

...to offline URL's like?
http://box/info/html/FAQ#What_are_the_default_passwords.3F

3. Can you possibly convert online URL's like:
https://github.com/iiab/iiab/wiki/IIAB-Installation#add-content
(like the 2nd link ["Add Content" instructions] inside FAQ #3 = http://10.8.0.14/info/html/FAQ#What_can_I_do_with_E-books_and_Internet-in-a-Box.3F )

...to offline URL's like:
http://10.8.0.14/info/IIAB-Installation.html#add-content

4. Please rename xs-refresh-wiki-docs to iiab-refresh-wiki-docs (and all others beginning with xs-* inside /usr/bin, this week or this month if at all possible!)

[Thanks George: this is a major step forward with OFFline documentation now usable, even with #anchor tags pointing to subcomponents of docs, when it's intermittently working of course!]

georgejhunt, 2017-06-20:
>    1. Can you possibly convert online URL's like:
>    FAQ#What_can_I_do_with_E-books_and_Internet-in-a-Box.3F
>    1. Please rename xs-refresh-wiki-docs to iiab-refresh-wiki-docs (and

Adam,
I just today ran an install on a ubuntu VM. And the links are correct.

And there must be something going on for your links to have changed in the
last 20 minutes, that has nothing to do with ansible.

Is there a browser cache getting the way? Is squid running? Are there more
than one wifi signal in your vicinity?

holta, 2017-06-20:
> Is there a browser cache getting the way?
No.  Do look at the file yourself http://10.8.0.14/info and re-run xs-refresh-wiki-docs to see this in action.

> Is squid running?
No.

> Are there more than one wifi signal in your vicinity?
WiFi is not be involved (Internet connection is via wireline Ethernet).

PS /usr/bin/xs-refresh-wiki-docs has not changed since 3 days ago:

root@box:~# which xs-refresh-wiki-docs
/usr/bin/xs-refresh-wiki-docs
root@box:~# ls -l /usr/bin/xs-refresh-wiki-docs
-rwxr-xr-x 1 root root 2193 jen 17 20:00 /usr/bin/xs-refresh-wiki-docs

georgejhunt, 2017-06-20:
perhaps we are looking at different machines:
iiab-admin@box:/opt/iiab/iiab/roles/httpd/templates$ ll
/usr/bin/iiab-refresh-wiki-docs
-rwxr-xr-x 1 root root 3358 jen 19 20:57 /usr/bin/iiab-refresh-wiki-docs*

7: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc
pfifo_fast state UNKNOWN group default qlen 100
    link/none
    inet 10.8.0.14 peer 10.8.0.13/32 scope global tun0
       valid_lft forever preferred_lft forever

No That's it, we're looking at different scripts. you were looking at xs---
and I was looking at iiab---

holta, 2017-06-20:
(1) Right, I had no idea I was supposed to run iiab-refresh-wiki-docs instead of xs-refresh-wiki-docs.  We just figured that out at the same time during the past 20min!

Can I delete /usr/bin/xs-refresh-wiki-docs now, once and for all?

Or will it just reappear with the next "git pull" ?

(2) George, can you convert online URL's like:
box/**go/IIAB**/FAQ#What_are_the_default_passwords.3F
...to OFFline URL's like?
box/**info/html**/FAQ#What_are_the_default_passwords.3F

( e.g. this is the 'passwords are secured' link at the top of:
http://wiki.laptop.org/go/IIAB/Security -offlined-to->
http://box/info/html/Security.html )

(3) Can you also convert online URL's like:
**s://github.com/iiab/iiab/wiki**/IIAB-Installation#add-content
...to OFFline URL's like:
**://box/info/IIAB-Installation.html#add-content

( e.g. this is thw '"Add Content" instructions' link that right near the top of
http://wiki.laptop.org/go/IIAB/FAQ#What_can_I_do_with_E-books_and_Internet-in-a-Box.3F -offline-to->
http://box/info/html/FAQ#What_can_I_do_with_E-books_and_Internet-in-a-Box.3F )

holta, 2017-06-20:
Clarif: implementing "sed" conversions (2) and (3) above will fix many links internal to the FAQ + Security + local_vars.html within info/html

e.g. the following also will have several links suddenly start working:

http://wiki.laptop.org/go/IIAB/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F
-offlined-to->
http://box/info/html/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F

holta, 2017-07-07:
Monday let's try to close this out (if smoke-testing looks goods, and the most basic functional testing) given 3 code commits have since been merged:

#49 lynx is required for offline docs (June 19)
#64 add two sed statements (June 27)
#77 suppress doc generation when requested (July 4)

holta, 2017-09-02:
These 3 "offline IT docs" links now work, for implementers/owners/operators, thanks to @georgejhunt!

http://box/info/html/Security.html
http://box/info/html/FAQ.html#What_can_I_do_with_E-books_and_Internet-in-a-Box.3F
http://box/info/html/FAQ.html#What_is_local_vars.yml_and_how_do_I_customize_it.3F

closed 2017-09-02 by holta