iiab/iiab #36: delete /library/dokuwiki-2014-09-29

holta, 2017-06-14:
Can we please delete dokuwiki-2014-09-29 ?

Two (apparently identical) copies of  the dokuwiki file tree within /library are plainly excessive~ no other package requires this!

PS Modifying https://github.com/iiab/iiab/blob/master/roles/dokuwiki/tasks/install.yml is likely the way to make this happen.

jvonau, 2017-06-15:
Think this is a result of multiple ansible runs, timestamps on the directories would be useful. The first pass renames the directory but a second pass extracts the tar file but the target directory already exists so it doesn't move it to the 'new' name. Would creating a symlink from /dokuwiki-2014-09-29 to /dokuwiki be OK? PR #39

holta, 2017-06-20:
#39 is already merged!

But see also #50 for another (related?) dokuwiki filename fix?

jvonau, 2017-06-20:
yea {{ dokuwiki_version }} went from a file name with a extension to an actual version string with an assumed file type tacked on later. Remember I did ask if that was acceptable and was for discussion. Should make testing a newer version(wow almost 3 years now) easier in the future by just moving the symlink to the newer directory.

jvonau, 2017-07-04:
@holta issue resolved?

holta, 2017-07-04:
I don't know!  As every machine I've touched for the past couple years has had this garbage directory "/library/dokuwiki-2014-09-29" so I'd like confirmation on Raspbian (if not other OS's) if indeed it's hopefully gone now :)

holta, 2017-07-07:
@georgejhunt is hoping to try to figure out why this happens on all OS's to some people but not others... (who merely get a symbolic link, rather than a duplicate directory)

holta, 2017-09-02:
Fixed.

closed 2017-09-02 by holta