iiab/iiab #46: The use of ansible_lsb may cause a failure on CentOS

jvonau, 2017-06-19:
The use of ansible_lsb can cause a failure on CentOS or Fedora if the redhat-lsb-core rpm is not installed(1). This optional rpm(2) is not generally installed by default(3) on server class installs but can be drawn into the mix by other packages that still list it as a dependency(4). Now with the move to systemd replacing initscripts this is more of a backwards compatibility package like net-tools providing the legacy ifconfig command which has been replaced with the newer ip addr usage. If one wishes to ensure this ansible module is available adding this rpm as an ansible dependency to scripts/ansible makes sense unless one's goal is to be a x11 free install.

1. https://github.com/v1v/ansible-role-chef-client/issues/5
2. http://mirror.centos.org/centos/7/os/x86_64/repodata/d918936f5019be3fb66e9981a28cb2a41477a2963d741d454f79377a22214f43-c7-x86_64-comps.xml
3. https://serverfault.com/questions/821874/how-to-install-a-minimal-version-of-lsb-release-on-centos
4. https://blogs.wcode.org/2015/01/fake-lpr-or-how-to-install-redhat-lsb-core-without-cups-and-x11/

tim-moody, 2017-06-20:
assuming that ansible_lsb uses it. on debian with no lsb_release command available, ansible_lsb still has the correct values. I'm pretty sure I checked this on centos as well, and the values are there even though the package is not.

jvonau, 2017-06-20:
Well nothing like not guessing and going to the source(1). It appears like it first looks for the lsb_release binary if found uses it, then falls back to parsing /etc/lsb-release if present but is not present on redhat based products. Why the change to iiab.yml going back to using ansible_distribution in aa107858736f9e1da8523e87e22844088b934d96 ?

1. https://github.com/ansible/ansible/blob/stable-2.2/lib/ansible/module_utils/facts.py

tim-moody, 2017-06-20:
an excellent question. I hadn't noticed the commit.

jvonau, 2017-06-20:
Just trying to future proof the code, FWIW Debain 9 is moving away from lsb(1).
1. https://www.debian.org/releases/stretch/amd64/release-notes/ch-information.en.html#reduced-lsb-support

georgejhunt, 2017-06-20:
>    ansible/module_utils/facts.py

I'm resonsible for
https://github.com/iiab/iiab/commit/aa107858736f9e1da8523e87e22844088b934d96,
because I was tryiing to do the massive xsce->iiab change, and ansible_lsb
was failing for me. I didn't want to spend the time to understand it at the
moment.

tim-moody, 2017-06-21:
In the cited Debian 9 reference debian is moving away from lsb in general but not from lsb_release which they mention as used internally. Ansible seems to be moving towards it as there is support in 2.2 but not in 1.9.6 and supplies a pseudo value if lsb_release is not found.

tim-moody, 2017-06-21:
The big question to me is whether we need to support two releases of an OS at the same time, especially Debian 8 and 9, and whether there are any differences to ansible variables.

tim-moody, 2017-06-21:
if so, I prefer {{ ansible_lsb.id }}-{{ ansible_lsb.major_release }}

georgejhunt, 2017-06-21:
I went back and verified on a centos machine that ansible_lsb is not
defined. So I think we are stuck doing set fact in 1-prep to do the
raspbian discovery.

Or maybe it is time to figure out if we can use the current release of
ansible (don't want to be stuck on 2.2 forever).

tim-moody, 2017-06-21:
please see my email. this is only true for 1.9.6

georgejhunt, 2017-06-21:
Ok. Now I understand what caused CentOS to burp. I was using {{ ansible_lsb
}} rather than {{ ansible_lsb.id }}.

Your email suggestions would appear to solve the jessie/stretch issue --
but involves duplication or symbolic links for what has worked up to now.
I'd rather have only one copy, and sym links

tim-moody, 2017-06-21:
I don't follow the symlinks comment.  what links? copy of what?

georgejhunt, 2017-06-21:
Right now vars/CentOS.yml works, and we can change its name from CentOS.yml
to CentOS-7.yml. I think that was your proposal.

When Centos 8 comes out, and still works with the playbook we currently are
using, do we duplicate the current file, or link to it with the filename
CentOS-8.yml?

tim-moody, 2017-06-21:
if they follow the time-honored tradition of changing a bunch of stuff we will probably want a separate CentOS-9.yml.

The amount of duplication will be reduced after we rationalize default vars.

tim-moody, 2017-06-21:
rather CentOS-8.yml

tim-moody, 2017-06-21:
not sure how Ubuntu fits the pattern as we sort of hope 16.4 thru 17.4 will all just work, but again I ask how many versions of each OS do we have to support?

jvonau, 2017-06-21:
"supplies a pseudo value if lsb_release is not found"
Using the cited link for ansible-2.2 if the binary is not found the fallback is to parse /etc/lsb-release which on a new virgin install out of the box may not exist on all distros and should not if the lsb type of packages are not installed. Now to help muddy the waters, lsb is installed in kalite for CentOS.

"if so, I prefer {{ ansible_lsb.id }}-{{ ansible_lsb.major_release }}"
how about something like
{{ ansible_distribution }}-{{ ansible_distribution_major_version }}

tim-moody, 2017-06-21:
the reason for the change is and was that on Raspian ansible_distribution says Debian, but lsb is says Raspian

tim-moody, 2017-06-21:
looks like you're right. I took a fresh centos with no lsb_release command and redhat-lsb not installed and then there is no ansible_lsb in

ansible -m setup -i ansible_hosts localhost --connection=local

tim-moody, 2017-06-21:
of course wget, git, and ansible are prerequisite packages, so we could have one more

jvonau, 2017-06-21:
Yea I agree lsb will get dragged in later anyway by kalite's playbook but not sure why if systemd is used. Might be a good time to test if removing lsb( used for CentOS6 at the time of introduction) from kalite breaks it and if not can be removed.

jvonau, 2017-06-21:
"(don't want to be stuck on 2.2 forever)"
Part of the problem moving to a later version of ansible is the [suppression of warning](https://github.com/XSCE/xsce/issues/858) masks what will be changing in the future. Now that the console has moved maybe it is time to rethink suppressing those warning as runansible runtags iiab-network are all cli based.

tim-moody, 2017-06-22:
I'm ready to capitulate. But riddle me this:
local facts on rpi shows the os as Debian
So I changed /etc/ansible/facts.d/local_facts.fact to
#!/bin/bash
OS=`grep ^ID= /etc/*release|cut -d= -f2`
xxx=`grep ^ID= /etc/*release|cut -d= -f2`
OS=${OS//\"/}

# enable this install config to be tracked
if [ ! -f /etc/iiab/uuid ]; then
  uuidgen > /etc/iiab/uuid
fi
UUID=`cat /etc/iiab/uuid`

source /etc/iiab/iiab.env
cd $XSCE_DIR

# get current version
BRANCH=`git rev-parse --abbrev-ref HEAD`
COMMIT=`git rev-parse --verify HEAD`

if [ -d /usr/lib64/php ]
then
  PHPLIB_DIR=/usr/lib64/php
else
  if [ -d /usr/lib/php5 ]; then
     PHPLIB_DIR=/usr/lib/php5
  else
     PHPLIB_DIR=/usr/lib/php
  fi
fi

if [ -f /proc/device-tree/mfg-data/MN ]
then
  XO_VERSION=`cat /proc/device-tree/mfg-data/MN`
else
  XO_VERSION="none"
fi

if [ -f /etc/iiab/iiab.env ]
then
  . /etc/iiab/iiab.env
  if [ -z $BASE_VERSION ]
  then
    BASE_VERSION="0"
  fi
  BASE="$BASE_VERSION"
else
  BASE="0"
fi

if [ -f /.preload ]
then
  PRELOAD="True"
else
  PRELOAD="False"
fi
ANSIBLE_VERSION=$(ansible --version|head -n 1|cut -f 2 -d " ")
cat <<EOF
{"phplib_dir"             : "$PHPLIB_DIR",
"iiab_branch"             : "$BRANCH",
"iiab_commit"             : "$COMMIT",
"iiab_uuid"               : "$UUID",
"xo_model"                : "$XO_VERSION",
"iiab_base_ver"           : "$BASE",
"ansble_version"          : "$ANSIBLE_VERSION",
"os"                      : "$OS",
"xxx"                     : "$xxx",
"iiab_preload"            : "$PRELOAD"}
EOF

and the output is

{"phplib_dir"             : "/usr/lib/php5",
"iiab_branch"             : "master",
"iiab_commit"             : "144bb156030faf89fad06837ac39597731e9c5da",
"iiab_uuid"               : "b4b59ee4-1e3e-4dcc-bc6e-25d063d10b3b",
"xo_model"                : "none",
"iiab_base_ver"           : "2",
"ansble_version"          : "2.2.0.0",
"os"                      : "Debian",
"xxx"                     : "raspbian",
"iiab_preload"            : "False"}

something is changing $OS or there's an environment variable or ansible variable or something

georgejhunt, 2017-06-22:
Sorry, I think I did it to you. See:
https://github.com/iiab/iiab/blob/master/roles/1-prep/templates/iiab.env.j2

tim-moody, 2017-06-22:
let's try to calculate things in one place using the same method. I am ready to repudiate lsb_release.id as I think Jerry is right about the future and it isn't there by default anyway. (I was misled by kalite.)

My current favorite algorithm is that of local facts: OS=grep ^ID= /etc/*release|cut -d= -f2. (not sure if the next line is required.) So I move to put that in iiab.env.j2 and not calculate in local facts, since we are already sourcing. (some prefer /etc/*elease I suppose because of case)

georgejhunt, 2017-06-22:
+1 One place.

So the source is in local facts, it's already in ansible. Then iiab.env.j2
using an ansible variable gets it into bash. And we still use iiab_env.py
to get the info into python?

tim-moody, 2017-06-22:
I think the source is iiab_env which moves into ansible by being included in local_facts.

jvonau, 2017-06-22:
I see runansible is using the same detection, why not just put OS: distroname in OS.yml? Everything is originating from there anyway and it looks like you have moved to that idea for other things. Next have iiab.env then use OS, add php_version and friends if needed..  Of course iiab.yml would need to be fixed to actually use Ubuntu and Rasbain yml files but a manual copy for testing of Fedora.yml to fedora.yml  plus below loads correctly... I kind of like it.. hitch is to get the local_facts.facts file in place before calling ansible for the first time which is a easy fix 

diff --git a/xsce.yml b/xsce.yml
index 69071f2..9fea069 100644
--- a/xsce.yml
+++ b/xsce.yml
@@ -4,7 +4,7 @@
 
   vars_files:
   - vars/default_vars.yml
-  - vars/{{ ansible_distribution }}.yml
+  - vars/{{ ansible_local["local_facts"]["os"] }}.yml
   - vars/local_vars.yml
   - /etc/xsce/config_vars.yml
 
@@ -18,3 +18,4 @@
       - { role: 7-edu-apps, tags: ['edu-apps'] }
       - { role: 8-mgmt-tools, tags: ['tools'] }
       - { role: 9-local-addons, tags: ['addons'] }

Is this workable before I put effort into this?

georgejhunt, 2017-06-22:
I like Tim's proposal to put it in iiab.env, be sourced into facts, and
used by iiab-env.py, when needed.

jvonau, 2017-06-22:
I think you are missing the point, in order to use the Raspbain.yml file that is sitting there ansible_distribution does not work. Now if you are going to rely on /etc/*elease you should really start at with that at square one. The first point you can do that is iiab.yml. which loads the OS.yml file needed. 
The variable returned from {{ ansible_local["local_facts"]["os"] }} is in lowercase but from what I can see returns fedora, centos, debain, raspbain, not sure about the rest. Just rename the OS.yml files to match the case and you should be good to go. On a fresh install runansible would have to put the local_facts.facts file in place before calling the playbook.

tim-moody, 2017-06-23:
I agree that this breaks the vars in iiab.yml strategy, but it was broken as soon as lsb_release was discredited, thought I may be wrong about the order of evaluation in thinking that vars in iiab.yml happens before local facts, easy to test.

If facts is after vars, I think we need a 0-vars that gets facts (maybe ansible does that automatically) and then uses include_vars tasks to get os.yml vars followed by local vars,  followed by config vars. Using include_vars means it could have logic, like handling different releases of an os for those that need it. It could then possibly compute vars, though dependencies need analyzing.  iiab.yml would only have defaults. Either way this might be more flexible.

This should do no work at all, but simply put variables in place (I'm looking at you uuid) so that it could be included in any subset of plays without side effects.

There is also the first run problem in that we put the local facts in play using ansible.

georgejhunt, 2017-06-23:
I was assuming that iiab.env would move out of
roles/1-prep/templates/iiab.env.j2 to root iiab, and be called by
runansible, runtags, and install-console.

yes it is true, local-facts.OS would be lower case, and maybe the file
names in vars would need to change.

jvonau, 2017-06-23:
Long with a rename of the OS file to what is returned by `grep ^ID= /etc/*release|cut -d= -f2` this fixes the "vars in iiab.yml strategy" [patch.diff.txt](https://github.com/iiab/iiab/files/1096647/patch.diff.txt)
uses the same detection as scripts/ansible, works for me.

jvonau, 2017-06-23:
Fact gathering occurs when the playbook is called. I think 0-vars has merit, I can see moving computed_vars.yml detected_network.yml moving there. iiab.yml could use some variables for distro versions if needed but not right now.

tim-moody, 2017-06-23:
to confirm I tried

 vars_files:
  - "{{ iiab_repo }}/vars/default_vars.yml"
  - "{{ iiab_repo }}/vars/{{ ansible_local.local_facts.os }}.yml"

in iiab-admin.yml and it found my os

so short term we should change iiab-env, facts, and the top level playbook and can think about more structural changes for the next release.

jvonau, 2017-06-23:
yea that looks cleaner, thanks for testing.

jvonau, 2017-06-23:
I need some context  "(I'm looking at you uuid)" Can you expand on that please in a new issue? That differs from what is on github, why the {{ iiab_repo }}?

jvonau, 2017-06-28:
#11 #13

jvonau, 2017-07-02:
What was decided in regards to using OS (PR #85 PR #91 ) or OS_ver (PR #83 ) as the OS.yml files naming? ie use debain.yml, centos.yml verse using debain8.yml, debain9.yml, centos7.yml. Either way works but a single OS file with some ifdiff like what is being done in squid could do away with [setting facts later in computed_vars.yml ](https://github.com/georgejhunt/iiab/blob/863df22184dbf12771681b9d590cf4ffb343e189/roles/1-prep/tasks/computed_vars.yml) ( PR #91 )
`{% if ansible_local.local_facts.os_ver  == "debian9" %}
proxy: squid
php_version: 7.0
postgresql_version: 9.6
mysql_service: mariadb
{% else %}
proxy: squid3
php_version: 5
postgresql_version: 9.4
mysql_service: mysql
{% endif %}
`

tim-moody, 2017-07-02:
my understanding is that vars files can not have logic. correct me if I am wrong.

georgejhunt, 2017-07-02:
I came the the same conclusion. But I don't know how to get a definitive
answer. I tried a few things that did not work.-- ie

import debian.yml
when: true

Which errored out

jvonau, 2017-07-03:
I'd be in favour of the os_ver, debain-X.yml, centos-X.yml, etc naming scheme as being the simplest to implement and providing a clear understanding of which variables/facts are being loaded for each os.

jvonau, 2017-07-03:
If PHPLIB_DIR could move from detected in facts to declaring phplib_dir in the OS.yml file, then everything should in one place. Bonus is local_facts.facts just does OS, XO detection and ansible recording.

jvonau, 2017-07-04:
Well if you choose to continue with os, debian.yml, centos.yml naming right now look at PR #58 as what you need today to resolve #11 where the whole raspbain.yml rabbit hole began.

tim-moody, 2017-07-04:
I agree in principle with putting os attributes in os.yml, but there may be efficiencies for things like PHPLIB_DIR that can be calculated being in local facts to cut down on the maintenance as the number of os.yml files increases by adding release.

jvonau, 2017-07-04:
Nothing is currently using [phplib_dir](https://github.com/iiab/iiab/search?utf8=%E2%9C%93&q=phplib_dir&type=), and php7(32-64bit) is not in the mix. Good time to cut it now and retool for os.yml.

jvonau, 2017-07-04:
If retaining the capitalized OS.yml names is wanted, could pass OS= through a 
`case $OS in
    centos)
    OS=CentOS
    ;;
    debian)
   OS=Debian
    ;;
.....
esac
` in local_facts.facts and save the rename

tim-moody, 2017-07-04:
has the added benefit that we can identify an os that is not supported rather than just having ansible crash when the os.yml file does not exist.

jvonau, 2017-07-05:
Well the rename is done, preventing the ansible crash addressed in PR #98

jvonau, 2017-07-05:
"my understanding is that vars files can not have logic. correct me if I am wrong."
Maybe, but complex, I could not get it to go either. Think setting facts after roles in iiab.yml might of worked but why with the use of os.yml as an abstraction layer. Just would of ended up being at the same place in the end entering into 1-prep that is the first play.

jvonau, 2017-07-08:
Resolved in master

closed 2017-07-08 by jvonau

jvonau, 2017-07-08:
@tim-moody "If facts is after vars, I think we need a 0-vars that gets facts (maybe ansible does that automatically)" Where you thinking about something like [this](https://github.com/jvonau/iiab/pull/7/files)?