iiab/iiab #55: nextcloud on debian 8.8

tim-moody, 2017-06-21:
ansible run succeeded.
on login was presented with untrusted domain error
edited /opt/nextcloud/config/config.php:
 'trusted_domains' =>
  array (
    0 => 'localhost',
    1 => '192.168.0.11'
  ),
now can login
but Admin/changeme not accepted

tim-moody, 2017-06-21:
I can login as root with the mysql root password

georgejhunt, 2017-06-21:
I will change Admin/changeme.  I started with Josh's PR, and was just
changing stuff to get it running.

Sometimes the frame of reference I carry, makes more work for me.

tim-moody, 2017-06-21:
tried to add Admin/changeme but nextcloud will not take it as too common

tim-moody, 2017-06-21:
with owncloud the autoconfig.php filled the config.php file (sort of) to add trusted domains, etc.
I don't see that happening here
https://docs.nextcloud.com/server/9/admin_manual/configuration_server/automatic_configuration.html 
claims it works

georgejhunt, 2017-06-21:
> automatic_configuration.html

A good discovery!

I reread this a couple of times.  It looks like we can generate an
autoconfig.php and plop it in ./nextcloud/config, and avoid "finish setup"
screen if we supply all required variables. Needs to be tried.

We might be able to bypass the password (too simple) check, if we can find
the syntax for adding a user.

tim-moody, 2017-06-21:
sudo ./occ config:system:set trusted_domains 2 --value=your.domain

makes me think we could set the wan to trusted at boot time to handle dhcp on wan

georgejhunt, 2017-06-21:
I used the command line to create user: Admin/1ChangeMe2, just to see how
restrictive the filter was.

I have not yet found a way around nextcloud's desire to encript the
password, and put it into mysql.

The ansible playbook created a db-admin-user nextcloud/nextcloudmysql --
which are the credentials supplied during install.

There seems to be a 60 second delay during the login process, which I have
not been able to understand.

jvonau, 2017-06-21:
Thought using the concept of "required_ip" as a network was off. Tim, just to clarify 192.168.0.11 is the iiab wan ip and not the client? If so then yes the member of the array, 1 in this case, could be set on the fly. Maybe something like: `sudo -u <user> <path>/occ config:system:set trusted_domains 1 --value=<wanip>` Now hook that up to the up the client dhcp program used and you should be set.

tim-moody, 2017-06-22:
not sure what "required ip" means, but yes, this is the server's wanip. doesn't make total sense to me either.

jvonau, 2017-06-22:
"required ip" = nextcloud_required_ip in defaults/main.yml and tasks/nextcloud_enabled.yml. I recall from prior work that this might of been a network range and not the server address. Moving forward I think the logic is that if the client uses a url that is not defined in trusted_domains the request if denied. Think you could test the theory by connecting up to the LAN, access the server with http://box/nextcloud which should work once "box" is added to trusted_domains but should fail for http://schoolserver/nextcloud until added to trusted_domains.

tim-moody, 2017-06-22:
just setting up a vm to try this

tim-moody, 2017-06-22:
with the current master box is untrusted as is schoolserver, due no doubt because the autoconfig functionality is not happening

tim-moody, 2017-07-06:
think this is fixed

closed 2017-07-06 by tim-moody

holta, 2017-09-02:
Ref #175