iiab/iiab #59: missing gawk can cause Wifi detection failure on first run

jvonau, 2017-06-23:
From: [http://iiab.io/t/iiab-sugardm-installation-former-topic-title-iiab-6-2-on-debian-9-installation-who-can-help/71](http://iiab.io/t/iiab-sugardm-installation-former-topic-title-iiab-6-2-on-debian-9-installation-who-can-help/71)

2017-06-13 21:47:51,201 p=2111 u=root | fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": "cat /proc/net/wireless | grep -v -e Inter -e face | gawk -F: '{print $1}'", "delta": "0:00:00.013347", "end": "2017-06-13 21:47:51.138643", "failed": true, "rc": 127, "start": "2017-06-13 21:47:51.125296", "stderr": "/bin/sh: 1: gawk: not found", "stdout": "", "stdout_lines": [], "warnings": []}
2017-06-13 21:47:51,202 p=2111 u=root | ...ignoring
2017-06-13 21:47:51,301 p=2111 u=root | TASK [1-prep : Set the discovered wireless, if found] 

While gawk is being installed later in packages.yml that is too late for this detection to work.

jvonau, 2017-07-20:
fixed with PR #137

holta, 2017-08-27:
@jvonau can we presumably resolve this now?

jvonau, 2017-08-27:
No, this causes the wifi not to be found during installation

jvonau, 2017-08-31:
roles/2-common/tasks/packages.yml install gawk but that is after 1-prep/detected_networks where I noted this error, related 7039b02.  While detected networks could be moved as is #137 that is a little intrusive.

closed 2017-09-05 by jvonau