iiab/iiab #69: local_tz can be left unset

jvonau, 2017-06-28:
The variable local_tz can be blank if TZ has not been exported to the environment first.

jvonau, 2017-06-28:
From a session with TK no TZ present:
        "ansible_env": {
            "ANSIBLE_LOG_PATH": "/opt/iiab/iiab/iiab-install.log", 
            "DISPLAY": ":0", 
            "HOME": "/root", 
            "LANG": "en_US.UTF-8", 
            "LANGUAGE": "en_US.UTF-8", 
            "LOGNAME": "root", 
            "LS_COLORS": "rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=01;05;37;41:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=01;36:*.au=01;36:*.flac=01;36:*.mid=01;36:*.midi=01;36:*.mka=01;36:*.mp3=01;36:*.mpc=01;36:*.ogg=01;36:*.ra=01;36:*.wav=01;36:*.axa=01;36:*.oga=01;36:*.spx=01;36:*.xspf=01;36:", 
            "MAIL": "/var/mail/root", 
            "PATH": "/sbin:/bin:/usr/sbin:/usr/bin", 
            "PWD": "/opt/iiab/iiab", 
            "SHELL": "/bin/bash", 
            "SHLVL": "4", 
            "SUDO_COMMAND": "/bin/bash ./load.txt", 
            "SUDO_GID": "1000", 
            "SUDO_UID": "1000", 
            "SUDO_USER": "olpc", 
            "TERM": "xterm", 
            "USER": "root", 
            "USERNAME": "olpc", 
            "XAUTHORITY": "/var/tmp/olpc-auth/.Xauthority", 
            "XDG_SEAT": "seat0", 
            "XDG_SESSION_ID": "c1", 
            "XDG_VTNR": "1", 
            "_": "/usr/bin/python"
        },

jvonau, 2017-06-29:
results in /etc/php.ini timezone issues and elgg farts..
[iiab-install-TK.log.txt](https://github.com/iiab/iiab/files/1112285/iiab-install-TK.log.txt)

jvonau, 2017-07-02:
[Where does my global php.ini file live?](https://stackoverflow.com/questions/17850787/where-is-my-php-ini-file-linux-centos) Note the different paths for the php.ini files. Distros might be using some other location to define where the master php.ini file lives. To ensure global changes wanted are incorporated across all distros OS.yml could have 'php_config_file: /full/path/to/php.ini' matching with the output of  '`php -i |grep 'Configuration File'`'(Note the 'Path' of). 

--- a/roles/httpd/tasks/main.yml
+++ b/roles/httpd/tasks/main.yml
-      - { src: 'php.ini.j2' , dest: '/etc/php.ini' , mode: '0644' }
+      - { src: 'php.ini.j2' , dest: '{{ php_config_file }}' , mode: '0644' }

jvonau, 2017-07-04:
php_version: 7.0 in Ubuntu.yml
php-config on redhat based os 
 --extension-dir     [/usr/lib64/php/modules]
php: /usr/bin/php /usr/lib64/php /etc/php.ini /etc/php.d /usr/include/php /usr/share/php /usr/share/man/man1/php.1.gz
From log
"ansible_userspace_architecture": "x86_64",
ansible_local": {
            "local_facts": {
                "ansble_version": "2.2.0.0", 
                "iiab_base_ver": "0", 
                "iiab_branch": "master", 
                "iiab_commit": "4462652deb20150c08a07d57bc9c4075a41afa9e", 
                "iiab_preload": "False", 
                "iiab_uuid": "8c4f7b2c-2fff-4cc8-b244-e7b1b580c5ee", 
                "os": "Ubuntu", 
                "phplib_dir": "/usr/lib/php",                
                "xo_model": "none"
            },

@holta TK is travelling. Something is not right, I need a `whereis php` on a 64bit Ubuntu box when you get a chance and paste the output here please.

holta, 2017-07-04:
@jvonau:

```
root@box:~# whereis php
php: /usr/bin/php /usr/bin/php7.0 /usr/lib/php /etc/php /etc/php.ini /usr/include/php /usr/share/php7.0-opcache /usr/share/php7.0-gd /usr/share/php7.0-xml /usr/share/php7.0-odbc /usr/share/php7.0-xmlrpc /usr/share/php7.0-curl /usr/share/php7.0-mbstring /usr/share/php7.0-common /usr/share/php /usr/share/php7.0-ldap /usr/share/php7.0-mysql /usr/share/php7.0-intl /usr/share/php7.0-json /usr/share/php7.0-pgsql /usr/share/php7.0-readline /usr/share/php7.0-mcrypt /usr/share/php7.0-imap /usr/share/php7.0-zip /usr/share/man/man1/php.1.gz

root@box:~# uname -a
Linux box.lan 4.8.0-58-generic #63~16.04.1-Ubuntu SMP Mon Jun 26 18:08:51 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

holta, 2017-07-04:
No idea why the above says Ubuntu 16.04.1 -- below it insists it's 16.04.2, as the original installer claimed:

```
root@box:~# more /etc/issue
Ubuntu 16.04.2 LTS \n \l

root@box:~# more /etc/issue.net
Ubuntu 16.04.2 LTS
```

jvonau, 2017-07-05:
Ok need to see a stock unmodified(before runansible) /etc/php.ini from Ubuntu just to compare with what we are providing that is from fedora originally. For completeness `whereis php` on raspbian and 64bit debian would be useful also. There may also be path differences, could I also get `ls -la /lib` `ls -la /lib64` for the debian family, those are symlinks on redhat pointing to /usr/lib and /usr/lib64.  I think the extension_dir= is being set wrong and just want to cover all bases to help decide the correct course of action.

holta, 2017-07-05:
@jvonau I don't have access to any Debian machines.  But the latest Raspbian Lite on RPi3 shows:

```
root@box:~# whereis php
php: /usr/bin/php /etc/php.ini /usr/share/php /usr/share/man/man1/php.1.gz

root@box:~# ls -la /lib
total 388
drwxr-xr-x 17 root root  4096 Jul  1 17:09 .
drwxr-xr-x 23 root root  4096 Jul  1 17:54 ..
drwxr-xr-x  5 root root 12288 Jul  1 17:16 arm-linux-gnueabihf
drwxr-xr-x  2 root root  4096 Jul  1 16:59 bridge-utils
lrwxrwxrwx  1 root root    21 Jun 21 04:49 cpp -> /etc/alternatives/cpp
drwxr-xr-x  3 root root  4096 Jun 21 04:43 crda
drwxr-xr-x  3 root root  4096 Jun 21 04:52 dhcpcd
drwxr-xr-x 12 root root  4096 Jun 21 04:52 firmware
drwxr-xr-x  2 root root  4096 Jun 21 04:30 ifupdown
drwxr-xr-x  2 root root  4096 Jun 21 04:28 init
-rwxr-xr-x  1 root root 76936 Nov 26  2014 klibc-YL2Pal4e_FwRI58JJ6S97Xf241g.so
lrwxrwxrwx  1 root root    30 Jun 19 15:52 ld-linux-armhf.so.3 -> arm-linux-gnueabihf/ld-2.19.so
lrwxrwxrwx  1 root root    24 Jun 21 04:38 ld-linux.so.3 -> /lib/ld-linux-armhf.so.3
lrwxrwxrwx  1 root root    17 May 25  2014 libip4tc.so.0 -> libip4tc.so.0.1.0
-rw-r--r--  1 root root 22264 May 25  2014 libip4tc.so.0.1.0
lrwxrwxrwx  1 root root    17 May 25  2014 libip6tc.so.0 -> libip6tc.so.0.1.0
-rw-r--r--  1 root root 22284 May 25  2014 libip6tc.so.0.1.0
lrwxrwxrwx  1 root root    15 May 25  2014 libipq.so.0 -> libipq.so.0.0.0
-rw-r--r--  1 root root  9668 May 25  2014 libipq.so.0.0.0
lrwxrwxrwx  1 root root    16 May 25  2014 libiptc.so.0 -> libiptc.so.0.0.0
-rw-r--r--  1 root root  5308 May 25  2014 libiptc.so.0.0.0
lrwxrwxrwx  1 root root    20 Jul 26  2014 libnih-dbus.so.1 -> libnih-dbus.so.1.0.0
-rw-r--r--  1 root root 34108 Jul 26  2014 libnih-dbus.so.1.0.0
lrwxrwxrwx  1 root root    15 Jul 26  2014 libnih.so.1 -> libnih.so.1.0.0
-rw-r--r--  1 root root 87368 Jul 26  2014 libnih.so.1.0.0
lrwxrwxrwx  1 root root    20 May 25  2014 libxtables.so.10 -> libxtables.so.10.0.0
-rw-r--r--  1 root root 42788 May 25  2014 libxtables.so.10.0.0
drwxr-xr-x  3 root root  4096 Jun 21 04:27 lsb
drwxr-xr-x  2 root root  4096 Jun 21 04:27 modprobe.d
drwxr-xr-x  4 root root  4096 Jul  4 12:40 modules
drwxr-xr-x  2 root root  4096 Jun 21 04:28 startpar
drwxr-xr-x  8 root root  4096 Jun 21 04:28 systemd
drwxr-xr-x 15 root root  4096 Sep 17  2014 terminfo
drwxr-xr-x  4 root root  4096 Jul  1 17:24 udev
drwxr-xr-x  2 root root  4096 Jun 21 04:30 xtables

root@box:~# ls -la /lib64
ls: cannot access /lib64: No such file or directory

root@box:~# uname -a
Linux box.lan 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux

root@box:~# more /etc/issue
Raspbian GNU/Linux 8 \n \l

root@box:~# more /etc/issue.net
Raspbian GNU/Linux 8
```

jvonau, 2017-07-05:
Good enough for now. Yup, no /lib64 on a 32bit userspace rpi, expected but needed info on 64bit machines. Better grab the output of '`php -i |grep 'Configuration File'`' just to be sure which ini file is being loaded. `ls -la /usr/lib/* | grep php ` should find php and on debian with a version number.

holta, 2017-07-05:
FWIW @jvonau on Raspbian:

```
root@box:~# php -i | grep 'Configuration File'
Configuration File (php.ini) Path => /etc/php5/cli
Loaded Configuration File => /etc/php5/cli/php.ini

root@box:~# ls -la /usr/lib/* | grep php
/usr/lib/php5:

root@box:~# ls -la /usr/lib | grep php
drwxr-xr-x  5 root root    4096 Jul  1 17:49 php5
```

jvonau, 2017-07-05:
Great, whereis is showing /etc/php.ini while /etc/php5/cli/php.ini is loaded, can't recall atm if /etc/php.ini is read and respected or superseded by the difference in loaded ini files. Feel free to chime in if you know. Useful info anyway because we are looking for php/modules. Ubuntu is php 7 based, the config file which defines the library path might will be different again. Can you repeat the above on the Ubuntu please. Really need virgin ini files from across the distros to compare.

jvonau, 2017-07-05:
`php -i | grep 'Configuration File'` `php -i | grep extension_dir` from all distros. Debian-9 moved to php7 so I'll bet that there will be a usr/lib/php7 for that.

holta, 2017-07-05:
@jvonau , do you need additional HDD space to spin up your own VM(s)?  VM's are very compact FYI, if you need this.

Meantime, from Ubuntu 16.04.2:

```
root@box:~# ls -la /lib
total 228
drwxr-xr-x 25 root root  4096 Jun 19 15:37 .
drwxr-xr-x 26 root root  4096 Jul  1 01:54 ..
drwxr-xr-x  2 root root  4096 Jun 17 20:14 apparmor
drwxr-xr-x  2 root root  4096 Jun 17 20:57 bridge-utils
drwxr-xr-x  2 root root  4096 Feb 15 15:30 brltty
lrwxrwxrwx  1 root root    21 Jun 17 19:39 cpp -> /etc/alternatives/cpp
drwxr-xr-x  3 root root  4096 Feb 15 15:26 crda
drwxr-xr-x  3 root root  4096 Jun 17 21:06 dhcpcd
drwxr-xr-x 77 root root 36864 Jul  1 01:53 firmware
drwxr-xr-x  2 root root  4096 Feb 15 15:25 hdparm
drwxr-xr-x  2 root root 12288 Jul  4 13:30 i386-linux-gnu
drwxr-xr-x  2 root root  4096 Feb 15 15:20 ifupdown
drwxr-xr-x  2 root root  4096 Feb 15 15:25 init
-rwxr-xr-x  1 root root 70952 Jan 26 13:18 klibc-k3La8MUnuzHQ0_kG8hokcGAC0PA.so
lrwxrwxrwx  1 root root    25 Jun 16 16:36 ld-linux.so.2 -> i386-linux-gnu/ld-2.23.so
drwxr-xr-x  2 root root  4096 Feb 15 15:25 linux-sound-base
drwxr-xr-x  3 root root  4096 Feb 15 15:21 lsb
drwxr-xr-x  2 root root  4096 Jul  1 01:53 modprobe.d
drwxr-xr-x  6 root root  4096 Jul  1 01:53 modules
drwxr-xr-x  3 root root  4096 Feb 15 15:25 recovery-mode
drwxr-xr-x  2 root root  4096 Jun 17 20:15 resolvconf
drwxr-xr-x  8 root root  4096 Jun 17 20:12 systemd
drwxr-xr-x  2 root root  4096 Feb 15 15:25 sysvinit
drwxr-xr-x 15 root root  4096 Feb 19  2016 terminfo
drwxr-xr-x  4 root root  4096 Jul  1 01:52 udev
drwxr-xr-x  2 root root  4096 Feb 15 15:25 ufw
drwxr-xr-x  3 root root 16384 Jul  4 13:30 x86_64-linux-gnu
drwxr-xr-x  2 root root  4096 Feb 15 15:25 xtables

root@box:~# ls -la /lib64
total 8
drwxr-xr-x  2 root root 4096 Jun 19 15:37 .
drwxr-xr-x 26 root root 4096 Jul  1 01:54 ..
lrwxrwxrwx  1 root root   32 Jun 16 16:57 ld-linux-x86-64.so.2 -> /lib/x86_64-linux-gnu/ld-2.23.so

root@box:~# php -i | grep 'Configuration File'
Configuration File (php.ini) Path => /etc/php/7.0/cli
Loaded Configuration File => /etc/php/7.0/cli/php.ini

root@box:~# ls -la /usr/lib/* | grep php
/usr/lib/php:
-rwxr-xr-x   1 root root   659 May 11 19:12 php7.0-fpm-checkconf
-rwxr-xr-x   1 root root   309 May 11 19:12 php7.0-fpm-reopenlogs
-rw-r--r--   1 root root  4786 Mar  1  2016 php-helper
-rw-r--r--   1 root root  9531 Mar  1  2016 php-maintscript-helper
-rw-r--r--   1 root root   101 May 11 19:12 php7.0-fpm.conf

root@box:~# ls -la /usr/lib | grep php
drwxr-xr-x   4 root root    4096 Jun 17 20:58 php
```

holta, 2017-07-05:
Ubuntu:

```
root@box:~# php -i | grep extension_dir
extension_dir => /usr/lib/php/20151012 => /usr/lib/php/20151012
```

Raspbian:

```
root@box:~#  php -i | grep extension_dir
extension_dir => /usr/lib/php5/20131226 => /usr/lib/php5/20131226
sqlite3.extension_dir => no value => no value
```

jvonau, 2017-07-05:
Ok in the above Ubuntu is that virgin machine?(all most used box there.. lol)  That is with php 7 using /etc/php.ini, think stop supplying that 4 year old ini file and move to sliding in a custom ini file with specs needed for timezone and if needed extension_dir would be a good move. With `php -i | grep 'Scan'` will give a config path that goes into os.yml `{{phpconfdir}}` that can be used to supply the custom ini template into, ie /etc/php5/apache2/conf.d/ for debian, Now `{{phplibdir}}` moves to os.yml also as I'll bet we have /usr/lib/php7 with debian-9. Just need to know if there are any /lib64 /usr/lib64 issues from the debian family to know what to do.

holta, 2017-07-05:
Unfort I don't know the legal definition of "virgin machine" !  I can say Ubuntu 16.04.2 LTS was installed 2 weeks ago (onto this very low-end NUC) and then IIAB 6.3 master was installed on top of that.  I have no idea if that is that "virgin enough" ?  But if it helps:

```
root@box:~# php -i | grep 'Scan'
Scan this dir for additional .ini files => /etc/php/7.0/cli/conf.d
```

jvonau, 2017-07-05:
"virgin machine" = installed, updated, before runansible, before `curl`ing the load script. I'm looking for the out of the box setting of php and what the php.ini file looks like, so php would need to be hand installed.  I don't think Debian suffers from this as `Configuration File (php.ini) Path` from above points to /etc/php5/cli and with that logic /etc/php/7.0/cli should be fine also but we have debian & ubuntu whereis php showing that /etc/php.ini might be play. I'll interpret that to mean the /etc/php.ini template could have changed some settings to some degree but need the virgin settings to confirm that thought. Well enough for today.

holta, 2017-07-05:
Ok.  Seems there is no PHP installed on virgin Ubuntu 16.04.2 machines, prior to installation of IIAB (see below).  @jvonau are you needing someone to run "apt install php" on such a machine?

root@box:~# more /etc/issue
Ubuntu 16.04.2 LTS \n \l

root@box:~# uname -a
Linux MED2 4.8.0-58-generic #63~16.04.1-Ubuntu SMP Mon Jun 26 18:08:51 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

root@box:~# whereis php
php:

root@box:~# ls -la /lib
total 224
drwxr-xr-x 23 root root  4096 Jun 20 02:58 .
drwxr-xr-x 24 root root  4096 Jun 30 00:44 ..
drwxr-xr-x  2 root root  4096 May 30 08:27 apparmor
drwxr-xr-x  2 root root  4096 Feb 15 15:30 brltty
lrwxrwxrwx  1 root root    21 May 30 07:56 cpp -> /etc/alternatives/cpp
drwxr-xr-x  3 root root  4096 Feb 15 15:26 crda
drwxr-xr-x 75 root root 36864 Jul  5 11:54 firmware
drwxr-xr-x  2 root root  4096 Feb 15 15:25 hdparm
drwxr-xr-x  2 root root 12288 Jul  3 17:27 i386-linux-gnu
drwxr-xr-x  2 root root  4096 Feb 15 15:20 ifupdown
drwxr-xr-x  2 root root  4096 Feb 15 15:25 init
-rwxr-xr-x  1 root root 70952 Jan 26 13:18 klibc-k3La8MUnuzHQ0_kG8hokcGAC0PA.so
lrwxrwxrwx  1 root root    25 Jun 16 16:36 ld-linux.so.2 -> i386-linux-gnu/ld-2.23.so
drwxr-xr-x  2 root root  4096 Feb 15 15:25 linux-sound-base
drwxr-xr-x  3 root root  4096 Feb 15 15:21 lsb
drwxr-xr-x  2 root root  4096 Jul  5 11:54 modprobe.d
drwxr-xr-x  5 root root  4096 Jul  5 11:54 modules
drwxr-xr-x  3 root root  4096 Feb 15 15:25 recovery-mode
drwxr-xr-x  2 root root  4096 May 30 08:28 resolvconf
drwxr-xr-x  8 root root  4096 May 30 08:24 systemd
drwxr-xr-x  2 root root  4096 Feb 15 15:25 sysvinit
drwxr-xr-x 15 root root  4096 Feb 19  2016 terminfo
drwxr-xr-x  4 root root  4096 Jul  1 02:48 udev
drwxr-xr-x  2 root root  4096 Feb 15 15:25 ufw
drwxr-xr-x  3 root root 20480 Jul  3 17:27 x86_64-linux-gnu
drwxr-xr-x  2 root root  4096 Feb 15 15:25 xtables
root@MED2:~# ls -la /lib64
total 8
drwxr-xr-x  2 root root 4096 Jun 20 02:58 .
drwxr-xr-x 24 root root 4096 Jun 30 00:44 ..
lrwxrwxrwx  1 root root   32 Jun 16 16:57 ld-linux-x86-64.so.2 -> /lib/x86_64-linux-gnu/ld-2.23.so

root@box:~# php -i | grep 'Configuration File'
The program 'php' can be found in the following packages:
 * php7.0-cli
 * hhvm
Try: apt install <selected package>

root@box:~# ls -la /usr/lib/* | grep php

root@box:~# ls -la /usr/lib | grep php

root@box:~# php -i | grep extension_dir
The program 'php' can be found in the following packages:
 * php7.0-cli
 * hhvm
Try: apt install <selected package>

root@box:~# locate php
/etc/apparmor.d/abstractions/php5
/lib/modules/4.8.0-54-generic/kernel/drivers/pci/hotplug/acpiphp_ibm.ko
/lib/modules/4.8.0-56-generic/kernel/drivers/pci/hotplug/acpiphp_ibm.ko
/lib/modules/4.8.0-58-generic/kernel/drivers/pci/hotplug/acpiphp_ibm.ko
/usr/share/app-install/desktop/gphpedit:gphpedit.desktop
/usr/share/app-install/desktop/slbackup-php:slbackup-php.desktop
/usr/share/app-install/icons/gphpedit.png
/usr/share/doc/apg/php.tar.gz
/usr/share/gedit/plugins/snippets/php.xml
/usr/share/gtksourceview-3.0/language-specs/php.lang
/usr/share/icons/Humanity/mimes/16/gnome-mime-application-x-php.svg
/usr/share/icons/Humanity/mimes/22/application-x-php.svg
/usr/share/icons/Humanity/mimes/22/gnome-mime-application-x-php.svg
/usr/share/icons/Humanity/mimes/24/application-x-php.svg
/usr/share/icons/Humanity/mimes/24/gnome-mime-application-x-php.svg
/usr/share/icons/Humanity/mimes/48/application-x-php.svg
/usr/share/icons/Humanity/mimes/48/gnome-mime-application-x-php.svg
/usr/share/icons/unity-icon-theme/places/svg/service-phpdocumentation.svg
/usr/share/lintian/checks/phppear.desc
/usr/share/lintian/checks/phppear.pm
/usr/share/lintian/data/files/php-libraries
/usr/share/mime/application/x-php.xml
/usr/share/nano/php.nanorc
/usr/share/yelp-xsl/js/jquery.syntax.brush.php-script.js
/usr/share/yelp-xsl/js/jquery.syntax.brush.php.js
/var/cache/dictionaries-common/sqspell.php
/var/lib/app-info/icons/ubuntu-xenial-universe/64x64/gphpedit_gphpedit.png

jvonau, 2017-07-05:
> Ok. Seems there is no PHP installed on virgin Ubuntu 16.04.2 machines,
> prior to installation of IIAB (see below). @jvonau
> <https://github.com/jvonau> are you needing someone to run "apt install
> php" on such a machine?

​Yes please, install php7.0-cli there might be others but start with that.
Next poke at it for a bit with what is in the issue...
1 is there a /etc/php.ini file? if so attach that file to the issue. or
make a note of that.
2 next find the php.ini file that php -i | grep 'Configuration File'
returns and upload that...
3 the result of php -i | grep extension_dir  as a note. or better yet  php
-i > php.txt as an attachment.
Thanks.

holta, 2017-07-05:
After running "apt install php7.0-cli" on a clean Ubuntu 16.04.2, see below & the 2 requested attachments at bottom...

```
root@box:/# updatedb; locate php.ini
/etc/php/7.0/cli/php.ini
/usr/lib/php/7.0/php.ini-development
/usr/lib/php/7.0/php.ini-production
/usr/lib/php/7.0/php.ini-production.cli

root@box:/# php -i | grep 'Configuration File'
Configuration File (php.ini) Path => /etc/php/7.0/cli
Loaded Configuration File => /etc/php/7.0/cli/php.ini

root@box:/# php -i | grep extension_dir
extension_dir => /usr/lib/php/20151012 => /usr/lib/php/20151012
```
[php.ini.txt](https://github.com/iiab/iiab/files/1125486/php.ini.txt)
[php-minus-i-output.txt](https://github.com/iiab/iiab/files/1125485/php-minus-i-output.txt)

jvonau, 2017-07-05:
Shoot, should of asked for a `whereis php` also, that should not have /etc/php.ini in the output as it's not on the filesystem.

holta, 2017-07-06:
```
root@box:~# whereis php
php: /usr/bin/php7.0 /usr/bin/php /usr/lib/php /etc/php /usr/share/php7.0-json /usr/share/php7.0-readline /usr/share/php7.0-opcache /usr/share/php7.0-common /usr/share/man/man1/php.1.gz
```

jvonau, 2017-07-06:
Based on the last one I'll conclude that /etc/php.ini **is** being respected once on the filesystem. Rather than supplying the whole file thus clobbering what maybe pre-existing administrator defined setting think it would be safer to supply a file into the "scan" directory just like how zmq was handled before being split out.

holta, 2017-08-27:
@jvonau can we resolve this and/or bring it towards resolution with any associated PR's?

jvonau, 2017-08-27:
the PR #102  is noted above.

jvonau, 2017-09-09:
https://github.com/iiab/iiab-admin-console/issues/16 https://github.com/iiab/iiab-admin-console/pull/17/files

jvonau, 2017-09-09:
#198 Raspbian-9 was released 2017-08-16 prior to the opening of this discussion. This becomes more important as php has been bumped to be version 7 based, as debian-9 and ubuntu-16. In #46 
> Nothing is currently using phplib_dir, and php7(32-64bit) is not in the mix. Good time to cut it now and retool for os.yml.

holta, 2017-10-13:
@jvonau is researching whether we can clean up our ~5-year-old /etc/php.ini (1229 lines) which might help speed up Nextcloud logins among other things?

There are 3+ php.ini's on our Raspbian setup:
```
root@box:~# locate php.ini
/etc/php.ini
/etc/php/7.0/apache2/php.ini
/etc/php/7.0/cli/php.ini
/opt/iiab/iiab/roles/httpd/templates/php.ini.j2    [ becomes /etc/php.ini ]
/usr/lib/php/7.0/php.ini-development
/usr/lib/php/7.0/php.ini-production
/usr/lib/php/7.0/php.ini-production.cli
```
There are 4+ php.ini's on our Ubuntu setup:
```
root@box:~# locate php.ini
/etc/php.ini
/etc/php/7.0/apache2/php.ini
/etc/php/7.0/cli/php.ini
/etc/php/7.0/fpm/php.ini
/opt/iiab/iiab/roles/httpd/templates/php.ini.j2    [ becomes /etc/php.ini ]
/usr/lib/php/7.0/php.ini-development
/usr/lib/php/7.0/php.ini-production
/usr/lib/php/7.0/php.ini-production.cli

root@box:~# ls /etc/php/7.0/mods-available
calendar.ini  iconv.ini     mysqlnd.ini    shmop.ini      xmlreader.ini
ctype.ini     imagick.ini   odbc.ini       simplexml.ini  xmlrpc.ini
curl.ini      imap.ini      opcache.ini    sockets.ini    xmlwriter.ini
dom.ini       intl.ini      pdo.ini        sysvmsg.ini    xsl.ini
exif.ini      json.ini      pdo_mysql.ini  sysvsem.ini    zip.ini
fileinfo.ini  ldap.ini      pdo_odbc.ini   sysvshm.ini
ftp.ini       mbstring.ini  phar.ini       tokenizer.ini
gd.ini        mcrypt.ini    posix.ini      wddx.ini
gettext.ini   mysqli.ini    readline.ini   xml.ini
```

holta, 2018-09-14:
@jvonau should this ticket be closer a year later?

holta, 2018-09-20:
php.ini is also being discussed in the context of WordPress and Moodle norms for schools/etc here: #1147

holta, 2018-09-25:
FYI /etc/php.ini is no longer in use since yesterday's PR #1166 removed it from IIAB 6.7/master.

See bullet list item 3 within https://github.com/iiab/iiab/pull/1165 for a bit more explanation.

(@jvonau please re-open this tkt if it's still needed!)

closed 2018-09-25 by holta