iiab/iiab #71: Needless routines on Rpi

jvonau, 2017-06-28:
In 92-rtc-i2c.rules the program being called is hwclock which is part of the util-linux package. The chrony and ntp packages are intended to be server packages but function as clients also(1). Both, more or less, do the same thing but are mutually exclusive to each other as servers. The ntpdate package is a client only package but you have to add ntp-pool files, while systemd provides timedatectl which uses chrony under the hood.(2) I'll bet timedatectl is already present in raspbian and other distros but requires configuring before timedatectl will work correctly. 

Record rtc time in UTC:  timedatectl set-local-rtc 0   This sets the third line of /etc/adjtime to UTC
Set the local timezone:  timedatectl set-timezone <local TZ>    This sets up /etc/localtime to be a symlink to /usr/share/zoneinfo/<local TZ> Need if /etc/localtime does not exist. 
Enable updating upon reboot: systemctl enable systemd-timedated
For immediate updating: systemctl start systemd-timedated

1. https://chrony.tuxfamily.org/comparison.html
2. https://manpages.debian.org/jessie/systemd/timedatectl.1.en.html

jvonau, 2017-07-02:
iiab_TZ can set the timezone in roles/2-common/tasks/main.yml

holta, 2017-08-27:
@jvonau @tim-moody is this issue coming together for IIAB 6.4 in coming weeks?

(FWIW budget-sensitive a.k.a. poor folks are using RTC's less and less, on the RPi Zero hardware especially...not that we want to abandon support for RTC's of course...even if their use is very rare)

jvonau, 2017-08-27:
I'm loosing interest in fix the mess.

holta, 2017-08-27:
> I'm loosing interest in fix the mess.

I hear you!

@tim-moody & @georgejhunt please close and/or advise next steps.

jvonau, 2017-09-22:
#271 #280

tim-moody, 2017-09-22:
which PR has the change to 92-rtc-i2c.rules?

jvonau, 2017-09-22:
@tim-moody None yet, needs discussion but there are 2 stanza that do the same thing for 92-rtc-i2c.rules

holta, 2017-10-25:
I'll defer this to our [IIAB 6.6](https://github.com/iiab/iiab/milestone/3) Milestone, if there are no objections?

(Or does someone want to close / resolve / put this to rest for IIAB 6.5, as Feature Freeze approaches in 5 days?)

holta, 2018-07-18:
@jvonau am moving this to our [IIAB 6.7](https://github.com/iiab/iiab/milestone/4) milestone, assuming it cannot happen of [IIAB 6.6](https://github.com/iiab/iiab/wiki/IIAB-6.6-Release-Notes)'s July 31 target.

Or worst case, should this be closed?

closed 2019-07-26 by jvonau