iiab/iiab #78: Downgrading python setuptools breaks CentOS install

jvonau, 2017-06-29:
Andreas Gros 	
Jun 18
The next issue I'm running into is that the downgrade of setuptools doesn't work.
Once the uninstallation is done, pip throws an error that pkg_resources isn't present anymore:
File "/bin/pip", line 5, in <module>
from pkg_resources import load_entry_point

ImportError: No module named pkg_resources

https://github.com/iiab/iiab/blob/master/roles/kalite/tasks/install.yml

jvonau, 2017-06-29:
possible fix #79

holta, 2017-08-27:
@jvonau @tim-moody is this something important/converging for IIAB 6.4 in coming weeks?

jvonau, 2017-08-27:
Unfixed I'll go with CentOS is broken as is, and I would not even attempt one without #167

jvonau, 2017-09-05:
With no posted proof (iiab-install.log) of a successful CentOS install one has to rely on "smoketested" in #121 but I think that is for only the change, not a fresh install. On Jun 21 PR #46 
> looks like you're right. I took a fresh centos with no lsb_release command and redhat-lsb not installed and then there is no ansible_lsb in

At that point running a CentOS install would of confirmed or dismissed the presence of this reported pip error from PR #20 commit b3616f59645d639eb290b0c1abe25aebeaac23a6 as that is part of the 6.3 release.

jvonau, 2017-09-18:
PR #259 removes the code in question

closed 2017-09-18 by jvonau