iiab/iiab #80: updating via git breaks networking on after first install

jvonau, 2017-06-29:
2017-06-29 16:08:44,022 p=16960 u=root |  ok: [127.0.0.1]
2017-06-29 16:08:44,052 p=16960 u=root |  TASK [network : Grab clean copy of iiab-gen-iptables] **************************
2017-06-29 16:08:44,482 p=16960 u=root |  ok: [127.0.0.1] => (item={0: u'gateway/iiab-gen-iptables', 1: u'/usr/bin/iiab-gen-iptables'})
2017-06-29 16:08:44,865 p=16960 u=root |  ok: [127.0.0.1] => (item={0: u'gateway/check-LAN', 1: u'/usr/bin/check-LAN'})
2017-06-29 16:08:44,890 p=16960 u=root |  TASK [network : Create link so that old invocation will work] ******************
2017-06-29 16:08:45,111 p=16960 u=root |  fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "gid": 0, "group": "root", "mode": "0755", "msg": "refusing to convert between file and link for /usr/bin/xs-gen-iptables", "owner": "root", "path": "/usr/bin/xs-gen-iptables", "size": 3406, "state": "file", "uid": 0}
2017-06-29 16:08:45,114 p=16960 u=root |  	to retry, use: --limit @/opt/iiab/iiab/iiab-network.retry

remove the file before creating the symlink

jvonau, 2017-07-02:
On older installs users need to remove /usr/bin/xs-gen-iptables /usr/bin/xs-handle https://github.com/iiab/iiab/pull/68/commits/17a0d97b058136c22e199cd6d9fa947c0c0801bf

holta, 2017-07-02:
Exactly what happened to me earlier today on a 10-day-old install if 6.3 master on Ubuntu.

/usr/bin/xs-gen-iptables and /usr/bin/xs-handle need to be manually removed (or automatically perhaps?)

jvonau, 2017-07-10:
Not that upgrades are supported but here is a gotcha

TASK [moodle : Start postgresql-iiab] **************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "Unable to start service postgresql-iiab: Job for postgresql-iiab.service failed. See \"systemctl status postgresql-iiab.service\" and \"journalctl -xe\" for details.\n"}
systemctl | grep postgresql
● postgresql-iiab.service                                                                                    loaded failed failed    PostgreSQL database server
  postgresql-xs.service                                                                                      loaded active running   PostgreSQL database server

holta, 2017-08-27:
@jvonau can we close this issue?

jvonau, 2017-08-27:
No still un-fixed

jvonau, 2017-09-01:
fixed in #144

closed 2017-09-01 by jvonau