iiab/iiab #86: http://box/usb causes unnec popups

holta, 2017-07-01:
Here is a typical popup on Ubuntu, when inserting a USB stick:

```
Unable to mount 32 GB Volume
Device /dev/sdb1 is already mounted at `/media/usb0'.
```

But the same occurs on other OS's, e.g. Raspbian Pixel and presumably all others?

Note: the USB stick is mounted successfully, and the content appears successfully off of http://box/usb (this is only a question of reducing the confusing clutter that we're currently placing in front of teachers -- would be tremendous to make this useless/erroneous popup go away, if not now, then in future!)

jvonau, 2017-07-01:
The reason is the window manager is trying to mount the usbkey on behalf of the user again (at /run/media/'user'/'device id') but usbmount has already mounted it at /media/usb0

georgejhunt, 2017-07-07:
This is an image setup issue. Googling suggests that you can bring up the file-manager->edit->preferences->click in left column -- Volume Management->uncheck "Mount removable media automatically when they are inserted"

jvonau, 2017-11-12:
closeable?

holta, 2017-11-12:
How about I move this to IIAB 6.6 or 7.0 &mdash; it's not a high priority item but might be very nice to have to make educators lives for more education-focused :)

holta, 2019-07-26:
> you can bring up the file-manager->edit->preferences->click in left column -- Volume Management->uncheck "Mount removable media automatically when they are inserted"

Documented @ http://FAQ.IIAB.IO = http://wiki.laptop.org/go/IIAB/FAQ#Can_teachers_display_their_own_content.3F

closed 2019-07-26 by holta

holta, 2022-05-29:
Please see the **new** http://FAQ.IIAB.IO doc, in these 2 areas:

- ["4 Can teachers display their own content?"](https://wiki.iiab.io/go/FAQ#Can_teachers_display_their_own_content%3F)
- ["41 Can I permanently attach an external USB drive, to add more content?"](https://wiki.iiab.io/go/FAQ#Can_I_permanently_attach_an_external_USB_drive,_to_add_more_content%3F)