iiab/iiab #88: FAQ/Docs need 6.3 groundtruth (e.g. KA Lite ports 8006, 8007, 8008)

holta, 2017-07-01:
Example:

Haitians and others need 2 (or 3) KA Lite languages on 6.3, with videos/lessons direct-linked from http://box

They do this on ports 8006 for French and 8008 for English, as explained at the bottom of FAQ #22:
http://wiki.laptop.org/go/IIAB/FAQ#KA_Lite_Administration:_What_tips_.26_tricks_exist.3F

But they need precise instructions how to maintain this, so they can ./runansible on occasion, remaining full+active members of the IIAB community, so these ports are encoded into iiab-gen-iptables in an agreed-upon / recoverable way.

tim-moody, 2017-07-02:
iptables allows a range, which I use on the demo machine. Maybe we should just allow 8001-8009 when services externally accessible flag (or whatever it is) is true.

holta, 2017-07-02:
Very Interesting Idea.  The flag being:

services_externally_visible: True

...e.g. within:
http://download.iiab.io/6.3/rpi/local_vars.yml

...which triggers:
https://github.com/iiab/iiab/blob/master/roles/network/templates/gateway/iiab-gen-iptables

jvonau, 2017-07-02:
I see that services_externally_visible is only needed in gateway mode now, LAN has always been wide open, and now appliance is also wide open under the current thinking with the use of $MODE. The whole point of services_externally_visible was so the box was **not** wide open. Prior [related issue](https://github.com/XSCE/xsce/issues/914)

jvonau, 2017-07-02:
[ProxyPass](https://github.com/iiab/iiab/search?utf8=%E2%9C%93&q=proxy&type=) might work here. Setup /kalite pointing to localhost:8008, kalite-fr pointing to 8006. If proxypass works with http://box/kalite there is no need to open ports.

tim-moody, 2017-07-02:
services_externally_visible was indeed invented for appliance mode, when it went from completely open to completely closed. when did it go back to open? I don't think that's true.

jvonau, 2017-07-02:
Not sure when, hard to tell with the move, you can figure out that. Click on Adam's link above, the code there in full colour.

jvonau, 2017-07-02:
Think the MODE= code is broken anyway with the use of backticks around the grep string, those  backticks denote to execute what is within the them.
`grep iiab_network_mode_applied  /etc/iiab/iiab.ini | gawk '{print $3}'`
bash: Gateway: command not found"
You can't see the backticks on this page as the wiki uses the same keyboard symbol to denote code but it is a copy and paste.

Removing that keyboard symbol on either end of the string yields the desired result. 
grep iiab_network_mode_applied  /etc/iiab/iiab.ini | gawk '{print $3}'
Gateway

jvonau, 2017-07-02:
From https://github.com/iiab/iiab/wiki/IIAB-Installation
"Appliance - No firewall, no dhpcd, no DNS, just a contributor to an already existing network. "
Might want to fix up the references to /opt/schoolserver on that page.

holta, 2017-07-02:
> "Appliance - No firewall, no dhpcd, no DNS, just a contributor to an already existing network."

@jvonau: does the above need to be changed?

> Might want to fix up the references to /opt/schoolserver on that page.

Agreed.  I'll do this in coming days.  While I'd intentionally left the references to /opt/schoolserver over the past month (in support of IIAB/XSCE 6.2 legacy users) the time has now come to move on, with IIAB 6.3's release imminent in coming weeks (if not in coming days possibly?)

jvonau, 2017-07-02:
@holta: No, that is the behaviour I observed, but I would double check with Tim as he disagrees.

holta, 2017-07-07:
Let's try to finalize during Monday's call what services_externally_visible should do during Monday's call.  I'm happy with Tim's suggestion that this open up "campus" WAN-side access to 8001-8009.  Or 8006-8008 could also work, per https://github.com/iiab/iiab/wiki/IIAB-Networking#list-of-open-ports--services

For the imminent IIAB 6.3 release, I've begun documenting this in @ http://FAQ.iiab.io

([FAQ#22 = "KA Lite Administration: What tips & tricks exist?"](http://wiki.laptop.org/go/IIAB/FAQ#KA_Lite_Administration:_What_tips_.26_tricks_exist.3F) now much improved, but not quite there yet, pending this wider decision!)

closed 2017-09-02 by holta