iiab/iiab #89: IDEA: for web loading

jvonau, 2017-07-02:
For the future consider using a http://download.iiab.io/6.3/x86/load.txt file. Including logic for os detection(/etc/*release) and automatically install as needed per distro, based on http://download.iiab.io/6.2/x86/, results with a unified starting point for all distros.

tim-moody, 2017-07-02:
this was the intent of iiab-factory/curl-me, which is a wip and was intended to be iiab.me/install(.txt if necessary)

also rather than lengthy overrides in local_vars, default_vars and os.yml files should be reconciled with current practices.

jvonau, 2017-07-02:
Reconciled with what or who's current practice?

jvonau, 2017-07-02:
Well default_vars can go away altogether, roles/playbook/defaults/main.yml should have all variables anyway, if missing could be added, and provide what default_vars does. Try `cat roles/*/defaults/main.yml` to check what are the defaults currently. You can keep default_vars around as a master reference but you don't have to load it. Now os.yml holds distro specific configuration paths and service names and is not intended for users to mess with. With the use of copying of distro.localvars to local_vars there is a sane starting point for the distro for what is or is not supported by the distro in question. It seems like once the os detection is squared away distro.localvars and os.yml could be merged together. The supplied local_vars could then only have installed supported service_enabled(elgg, kalite) settings and user configurable settings(SSID) present in the file by default. What is not installed by default could also be added but I don't suggest having that present in local_vars as that makes no longer supported software(pathagar schooltool) available to the user to install. Experienced users and developers could add the service_install and service_enabled to local_vars should they wish to play with the disabled playbooks. Hope I'm conveying my thoughts clearly enough.

tim-moody, 2017-07-02:
Default vars should hold common defaults. os should hold exceptions and local vars deployment exceptions

jvonau, 2017-07-04:
"Default vars should hold common defaults" 
That is in place now by virtue of roles/plays/defaults/main.yml 
`cat roles/*/defaults/* | sort -u | grep _install`
Those are loaded before anything in iiab.yml
Now default_vars are turning some off `cat vars/default_vars.yml | grep _install`

"os should hold exceptions"
I agree, os.yml is currently holding soft coding of variables, os flags. I'll note that cups_install: True does appear though, should that be there?

"local vars deployment exceptions"
The os.localvars files that become local_vars, those should really have all the user changeable variables such as ssid, default_language, iiab_home_url, present for ready use by the end user as local_vars is a one time setup. One could just supply a copy of default_vars to use as was first intended but I see an [issue](https://github.com/iiab/iiab/compare/master...jvonau:case?expand=1) with an easy fix, to the better location IMHO. Just create a PR from the link above if you would like to use it for the finer grain control that might be need later. Should using default_vars as local_vars becomes a thing again, then axe default_vars in iiab.yml and adjust runansible to the copy.

closed 2017-07-04 by jvonau