iiab/iiab #111: nextcloud not accessible from vpn

tim-moody, 2017-07-07:
/etc/apache2/sites-available/nextcloud.conf has

Require ip 127.0.0.1 172.18.96.1/255.255.224.0 192.168.0.0/255.255.255.0

to which must be added 10.8.0.0/255.255.255.0 (or we could ease the restriction)

(nextcloud config.php has '*' so it is already wide open)

holta, 2017-07-07:
Does this apply only over VPN?

Or is "campus" use of Nextcloud also blocked on the WAN side of an IIAB 6.3 ?

tim-moody, 2017-07-07:
campus works, vpn not

jvonau, 2017-07-08:
What is the ip/netmask of the "campus" network that was tested? If the "campus" network is the same as what is defined in Require ip already it will work. However if the "campus" network falls outside of what is defined in Require ip then you can't login. ie 192.168.122.0/24 would fail. I suspect that Tim's network is already using 192.168.0.0/24

tim-moody, 2017-07-08:
unless of course the required ip is softcoded to that of server.

jvonau, 2017-07-08:
Perfect, in a dhcp environment where the portable server is jacked into different WANs on a regular bases what would be the procedure to refresh this "Require ip" setting?

holta, 2018-07-18:
@tim-moody I still feel this is important, but will move it to our [IIAB 6.7 Milestone](https://github.com/iiab/iiab/milestone/4) if it's not likely to converge now in coming weeks ?

@jvonau: do you think this is plausibly solvable?  Or not realistic?

holta, 2018-07-19:
@jvonau suggests this restriction is likely in Nextcloud or Apache config files &mdash; sounds very solvable.

holta, 2018-07-27:
Fixed by PR #967, Thanks @georgejhunt !

closed 2018-07-27 by holta