iiab/iiab #112: Upgrade path?

jvonau, 2017-07-08:
How are future upgrades going to be handled? Lets say going from 6.3 -> 6.4
The acid test for a clean upgrade path is git checkout -b <new_release> and a git pull origin <new_release> 
Starting from a fresh clone:
git clone https://github.com/iiab/iiab -b 6.3-rc2 --depth 1
[root@box iiab]# cd iiab
[root@box iiab]# git branch
* 6.3-rc2
[root@box iiab]# git checkout -b 6.4
Switched to a new branch '6.4'

The end user should be able to switch from the current branch to any other release branch or master without issues.
Using master as an example
[root@box iiab]# git pull origin master
remote: Counting objects: 1366, done.
remote: Compressing objects: 100% (618/618), done.
remote: Total 1366 (delta 773), reused 1214 (delta 640), pack-reused 0
Receiving objects: 100% (1366/1366), 385.57 KiB | 0 bytes/s, done.
Resolving deltas: 100% (773/773), completed with 167 local objects.
From https://github.com/iiab/iiab
 * branch            master     -> FETCH_HEAD
Auto-merging vars/default_vars.yml
CONFLICT (add/add): Merge conflict in vars/default_vars.yml
Auto-merging runansible
CONFLICT (add/add): Merge conflict in runansible
Auto-merging roles/xovis/tasks/main.yml
CONFLICT (add/add): Merge conflict in roles/xovis/tasks/main.yml
Auto-merging roles/wordpress/tasks/install.yml
CONFLICT (add/add): Merge conflict in roles/wordpress/tasks/install.yml
Auto-merging roles/teamviewer/tasks/install.yml
CONFLICT (add/add): Merge conflict in roles/teamviewer/tasks/install.yml
Auto-merging roles/sugarizer/tasks/main.yml
CONFLICT (add/add): Merge conflict in roles/sugarizer/tasks/main.yml
Auto-merging roles/sugar-stats/tasks/statistics-consolidation.yml
CONFLICT (add/add): Merge conflict in roles/sugar-stats/tasks/statistics-consolidation.yml
Auto-merging roles/schooltool/tasks/main.yml
CONFLICT (add/add): Merge conflict in roles/schooltool/tasks/main.yml
Auto-merging roles/postgresql/tasks/main.yml
CONFLICT (add/add): Merge conflict in roles/postgresql/tasks/main.yml
Auto-merging roles/phpmyadmin/tasks/main.yml
CONFLICT (add/add): Merge conflict in roles/phpmyadmin/tasks/main.yml
Auto-merging roles/pathagar/tasks/main.yml
CONFLICT (add/add): Merge conflict in roles/pathagar/tasks/main.yml
Auto-merging roles/owncloud/tasks/main.yml
CONFLICT (add/add): Merge conflict in roles/owncloud/tasks/main.yml
Auto-merging roles/owncloud/tasks/F18.yml
CONFLICT (add/add): Merge conflict in roles/owncloud/tasks/F18.yml
Auto-merging roles/osm/tasks/main.yml
CONFLICT (add/add): Merge conflict in roles/osm/tasks/main.yml
Auto-merging roles/openvpn/tasks/main.yml
CONFLICT (add/add): Merge conflict in roles/openvpn/tasks/main.yml
Auto-merging roles/nextcloud/tasks/nextcloud_enabled.yml
CONFLICT (add/add): Merge conflict in roles/nextcloud/tasks/nextcloud_enabled.yml
Auto-merging roles/nextcloud/tasks/main.yml
CONFLICT (add/add): Merge conflict in roles/nextcloud/tasks/main.yml
Auto-merging roles/nextcloud/tasks/F18.yml
CONFLICT (add/add): Merge conflict in roles/nextcloud/tasks/F18.yml
Auto-merging roles/network/templates/squid/squid.sysconfig
CONFLICT (add/add): Merge conflict in roles/network/templates/squid/squid.sysconfig
Auto-merging roles/network/templates/named/named.j2
CONFLICT (add/add): Merge conflict in roles/network/templates/named/named.j2
Auto-merging roles/network/templates/named/named
CONFLICT (add/add): Merge conflict in roles/network/templates/named/named
Auto-merging roles/network/templates/dhcp/dhcpd-env.j2
CONFLICT (add/add): Merge conflict in roles/network/templates/dhcp/dhcpd-env.j2
Auto-merging roles/network/tasks/squid.yml
CONFLICT (add/add): Merge conflict in roles/network/tasks/squid.yml
Auto-merging roles/network/tasks/rpi_debian.yml
CONFLICT (add/add): Merge conflict in roles/network/tasks/rpi_debian.yml
Auto-merging roles/network/tasks/restart.yml
CONFLICT (add/add): Merge conflict in roles/network/tasks/restart.yml
[... 78 lines cut ...]
Auto-merging roles/1-prep/templates/iiab.env.j2
CONFLICT (add/add): Merge conflict in roles/1-prep/templates/iiab.env.j2
Auto-merging roles/1-prep/tasks/prep.yml
CONFLICT (add/add): Merge conflict in roles/1-prep/tasks/prep.yml
Auto-merging roles/1-prep/tasks/main.yml
CONFLICT (add/add): Merge conflict in roles/1-prep/tasks/main.yml
Auto-merging roles/1-prep/tasks/iiab_ini.yml
CONFLICT (add/add): Merge conflict in roles/1-prep/tasks/iiab_ini.yml
Auto-merging roles/1-prep/tasks/detected_network.yml
CONFLICT (add/add): Merge conflict in roles/1-prep/tasks/detected_network.yml
Auto-merging roles/1-prep/tasks/computed_vars.yml
CONFLICT (add/add): Merge conflict in roles/1-prep/tasks/computed_vars.yml
Auto-merging roles/1-prep/defaults/main.yml
CONFLICT (add/add): Merge conflict in roles/1-prep/defaults/main.yml
Auto-merging iiab.yml
CONFLICT (add/add): Merge conflict in iiab.yml
Auto-merging iiab-network.yml
CONFLICT (add/add): Merge conflict in iiab-network.yml
Auto-merging iiab-from-console.yml
CONFLICT (add/add): Merge conflict in iiab-from-console.yml
Auto-merging iiab-base.yml
CONFLICT (add/add): Merge conflict in iiab-base.yml
Auto-merging ansible.cfg
CONFLICT (add/add): Merge conflict in ansible.cfg
Automatic merge failed; fix conflicts and then commit the result

holta, 2017-07-08:
There is basically nearly-zero demand for upgrades, as the offline world (essentially!) always asks for new content + new features to be bundles onto 1 new SD card or similar.

*BUT* we should be open to an upgrade process if it hopefully becomes genuinely hassle-free at some point in future... as offline nations/regions acquire partial connectivity increasingly, and so become tempted by partial upgrades of content especially~

jvonau, 2017-07-08:
"upgrade process if it hopefully becomes genuinely hassle-free"
The hassle-free part comes from using comes from using git correctly.

closed 2017-07-08 by jvonau