iiab/iiab #116: Documentation of runtags & speed up increases

jvonau, 2017-07-09:
This routine has been hidden in developer land for too long and should be available for use freely by the enduser and be fully supported. Think anby needs docs on how to use runtags.

To go with that speed up theme how about some comments on [this idea](https://github.com/jvonau/iiab/pull/7/files)

holta, 2017-07-10:
Anything that improved documentation AND speed gets my vote (if runtags is safe to promote?)

LMK where in http://wiki.laptop.org/go/IIAB/FAQ or similar ./runtags should best be documented.

jvonau, 2017-09-25:
>  And ansible feels "somewhat" secand nature in the meantime doing all these runansible and runtags. The only thing I had do disable for this success has been nextcloud. Yet thanks to runtags (for repeating runs) this has been like easy play. Here's the output I got with nextcloud enabled before. Did not look like the timeout has been on my end, yet I had no clue how to troubleshoot this so I disabled it. Now I'm experimenting with moodle. Moodle 3 looks good! (Even though some additions are needed to make it work with this pre-installation (regarding a zip-module and some else). I'm into this for now.
Cheers!

holta, 2017-09-25:
Above comment is from [Anby](http://iiab.io/u/anby), who he posted to our forums here:

http://iiab.io/t/iiab-6-3-rc2-lite-on-rpi-zero-w-with-2017-06-21-raspbian-jessie-lite/80/3

holta, 2017-10-25:
@jvonau does PR #303 speeds up ./runtags in the same basic way?

If so should we resolve this #116 (moving documentation mandate to #303?)

holta, 2017-10-25:
@jvonau plz re-open as nec.

closed 2017-10-25 by holta