iiab/iiab #119: End users has no opportunity to edit local_vars before runansible

jvonau, 2017-07-09:
The doing it from scratch routine does not allow end-user to edit local_vars prior to the install. The end-user has no way to override what is pre-selected to be installed. The common place to catch this would be in runansible, patch attached. The curl scripts would bypass this check as the local_vars files are present, but really should offer the user a chance to edit before install like curl-me has the ability to.
[pause.txt](https://github.com/iiab/iiab/files/1134157/pause.txt)

Other question is why are these curl scripts not under git version control somewhere? You can run them direct from github see my iiab repo for an example. Should you wish to retain a shorter url for the user to enter apache redirects are your friend, but I don't see why with copy and pasting the string is so easy.

holta, 2017-07-10:
Pausing in the midst of an hour-long script is a bad idea, which sends newbies running for the hills.

We need to bend over backwards so newvomers face Zero Hassles, and get to a minimal install ASAP.

If they are offered the opportunity to review/edit local_vars.yml *In Advance* of their hour-long install, that might however work ?

holta, 2017-07-10:
Many newbie installer folk will prefer to choose from among (a couple) prefab/community-published/community-reviewed local_vars.yml files.

(Ideally finalizing that selection of their favorite lite-vs-whatev local_vars.yml right at the command-line, so it truly remains a 1-line install if at all poss!?)

jvonau, 2017-07-10:
In the current scheme the curl script is supplying or could supply the local_vars file and this check would be automatically skipped, so no change for that workflow. However seasoned users may skip reading the docs, and just proceed to git clone and expect the local_vars file to already be present as it was in the past. Yes a selection of pre-fabed local_vars files already present in the git repo would be a clue that the work flow changed. That is in place with the os.localvars files so why the wget to grab a different one when the file could be right there in the git repo and just needs a rename that could be scripted in the curl script? Any real reason why these files are not under git control somewhere where they might be viewed in advance with all the other files used and changes tracked in one place?

holta, 2017-07-10:
As a documentation person I personally far prefer a solid wiki for document version control, unlike the broken (or at least VERY primitive ;) wiki offered by GitHub.

As such I currently use wiki.laptop.org (and archive.org etc!) to backup essential stuff in:

http://download.iiab.io/6.4/rpi/

Anyway people who manage document-like artifacts need very different tools from coders generally.  In the end there are significantly stronger tools to then manage newbie-human-facing documentation-like stuff elsewhere on the web.

We'll most surely evolve this, as we find better ways to serve BOTH kinds of contributors (coders & documentation people, who have quite different needs...and functional/living docs make the intersection of these 2 worlds even more interesting as we're finding!)

Anyway that's my personal opinion: I see absolutely no reason everything should be placed into GitHub...but admittedly everything does look like a nail when one has a git hammer ;)

jvonau, 2017-07-10:
Perhaps a readme in vars/ that has a list of urls for the web based curl scripts would be enough of a reminder of what is really being used for the automated routines.

closed 2017-07-10 by jvonau