iiab/iiab #132: offline testing

jvonau, 2017-07-16:
I ran across this error while testing with no internet cable present, should be reproducible with `./runtags prep` or a run of ansible under the console, addressed in PR #131

TASK [1-prep : Find active gateway config based on macaddress] ****************************************
skipping: [127.0.0.1]

TASK [1-prep : Set has ifcfg gw based on on macaddress if found] **************************************
fatal: [127.0.0.1]: FAILED! => {"failed": true, "msg": "the field 'args' has an invalid value, which appears to include a variable that is undefined. The error was: 'dict object' has no attribute 'stdout'\n\nThe error appears to have been in '/opt/schoolserver/iiab/roles/1-prep/tasks/detected_network.yml': line 102, column 3, but may\nbe elsewhere in the file depending on the exact syntax problem.\n\nThe offending line appears to be:\n\n\n- name: Set has ifcfg gw based on on macaddress if found\n  ^ here\n"}
	to retry, use: --limit @/opt/schoolserver/iiab/iiab.retry

Can somebody else pull the internet plug and test as above to confirm or dismiss this issue.

holta, 2017-08-27:
Is #66 important/relevant?

(Deprecation warnings with ansible 2.3+)

jvonau, 2017-08-27:
Think the warning have been silenced.

closed 2017-08-27 by holta