iiab/iiab #145: sudden loss of power locks elgg/mysql

jvonau, 2017-07-20:
from /var/log/mariadb/mariadb.log

170718 05:40:07 mysqld_safe Starting mysqld daemon with databases from /var/lib/mysql
170718  5:40:07 [Note] /usr/libexec/mysqld (mysqld 10.0.23-MariaDB) starting as process 1345 ...
170718  5:40:07 [Note] InnoDB: Using mutexes to ref count buffer pool pages
170718  5:40:07 [Note] InnoDB: The InnoDB memory heap is disabled
170718  5:40:07 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
170718  5:40:07 [Note] InnoDB: Memory barrier is not used
170718  5:40:07 [Note] InnoDB: Compressed tables use zlib 1.2.8
170718  5:40:07 [Note] InnoDB: Using Linux native AIO
170718  5:40:07 [Note] InnoDB: Using CPU crc32 instructions
170718  5:40:07 [Note] InnoDB: Initializing buffer pool, size = 128.0M
170718  5:40:07 [Note] InnoDB: Completed initialization of buffer pool
170718  5:40:07 [Note] InnoDB: Highest supported file format is Barracuda.
170718  5:40:07 [Note] InnoDB: The log sequence numbers 2175716 and 2175716 in ibdata files do not match the log sequence number 2802427 in the ib_logfiles!
170718  5:40:07 [Note] InnoDB: Database was not shutdown normally!
170718  5:40:07 [Note] InnoDB: Starting crash recovery.
170718  5:40:07 [Note] InnoDB: Reading tablespace information from the .ibd files...
170718  5:40:08 [Note] InnoDB: Restoring possible half-written data pages 
170718  5:40:08 [Note] InnoDB: from the doublewrite buffer...
170718  5:40:08 [Note] InnoDB: 128 rollback segment(s) are active.
170718  5:40:08 [Note] InnoDB: Waiting for purge to start
170718  5:40:08 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.26-76.0 started; log sequence number 2802427
170718  5:40:08 [Note] Plugin 'FEEDBACK' is disabled.
170718  5:40:08 [Note] Server socket created on IP: '::'.
170718  5:40:08 [ERROR] mysqld: Table './mysql/user' is marked as crashed and should be repaired
170718  5:40:08 [Warning] Checking table:   './mysql/user'
170718  5:40:08 [ERROR] mysql.user: 1 client is using or hasn't closed the table properly
170718  5:40:08 [ERROR] mysqld: Table './mysql/db' is marked as crashed and should be repaired
170718  5:40:08 [Warning] Checking table:   './mysql/db'
170718  5:40:08 [ERROR] mysql.db: 1 client is using or hasn't closed the table properly
170718  5:40:08 [Note] /usr/libexec/mysqld: ready for connections.
Version: '10.0.23-MariaDB'  socket: '/var/lib/mysql/mysql.sock'  port: 3306  MariaDB Server
170720  8:15:32 [ERROR] mysqld: Table './elggdb/elgg_users_sessions' is marked as crashed and should be repaired
170720  8:15:32 [Warning] Checking table:   './elggdb/elgg_users_sessions'
170720  8:25:54 [ERROR] mysqld: Table './elggdb/elgg_users_entity' is marked as crashed and should be repaired
170720  8:25:54 [Warning] Checking table:   './elggdb/elgg_users_entity'
170720  8:25:54 [ERROR] mysqld: Table './elggdb/elgg_system_log' is marked as crashed and should be repaired

and after restarting mariadb

[root@box iiab]# systemctl restart mariadb
[root@box iiab]# tail -f /var/log/mariadb/*
170720  8:37:14 [Note] InnoDB: Initializing buffer pool, size = 128.0M
170720  8:37:14 [Note] InnoDB: Completed initialization of buffer pool
170720  8:37:14 [Note] InnoDB: Highest supported file format is Barracuda.
170720  8:37:14 [Note] InnoDB: 128 rollback segment(s) are active.
170720  8:37:14 [Note] InnoDB: Waiting for purge to start
170720  8:37:14 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.26-76.0 started; log sequence number 2802437
170720  8:37:14 [Note] Plugin 'FEEDBACK' is disabled.
170720  8:37:14 [Note] Server socket created on IP: '::'.
170720  8:37:14 [Note] /usr/libexec/mysqld: ready for connections.
Version: '10.0.23-MariaDB'  socket: '/var/lib/mysql/mysql.sock'  port: 3306  MariaDB Server
170720  8:37:41 [ERROR] mysqld: Table './elggdb/elgg_system_log' is marked as crashed and should be repaired

holta, 2017-08-27:
@jvonau is this the same as https://github.com/xsce/xsce/issues/879 ?

jvonau, 2017-08-27:
No this one is under mysql while the linked issue is mongodb, might have the same underline cause thou.

holta, 2017-09-01:
@jvonau do you remember what OS this occurred on, when you pasted it in July 20?

jvonau, 2017-09-01:
My old fedora-22 xsce-6.2 install upgraded to iiab master. That should not
matter, as the DB was working fine before the power got cut.

holta, 2017-09-01:
Let's keep an eye on this as @georgejhunt just upgraded elgg from 1.12.12 to 1.12.16 (PR #179)

Or tried...seems neither 1.12.12 nor 1.12.16 are working right now, ah well :-)

closed 2017-09-18 by jvonau