iiab/iiab #147: power_off.php does not appear in /library/www/html/services

holta, 2017-07-20:
Also the directory itself ("services") does not appear.  Confirmed on a clean install of IIAB 6.3 when following the instructions at http://FAQ.iiab.io number 16 here: http://wiki.laptop.org/go/IIAB/FAQ#Is_a_.22Rapid_Power_Off.22_button_possible_for_low-electricity_environments.3F

i.e. "allow_apache_sudo: True" was correctly placed into local_vars.yml prior to ./runansible

PS as a temporary workaround, one can do:

```
mkdir /library/www/html/services
cp /opt/iiab/iiab/roles/httpd/files/html/services/power_off.php /library/www/html/services
```

georgejhunt, 2017-07-20:
> IIAB/FAQ#Is_a_.22Rapid_Power_Off.22_button_possible_for_
> low-electricity_environments.3F

We need to understand why the last line in quote below is not effective
from: https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/fl.yml
- name: Create directory for common packages
file: path={{ item }}
mode=0755
owner=root
group=root
state=directory
with_items:
- "{{ doc_root }}/common/css"
- "{{ doc_root }}/common/js"
- "{{ doc_root }}/common/fonts"
- "{{ doc_root }}/common/html"
- "{{ doc_root }}/common/images"
- "{{ doc_root }}/common/assets"
- "{{ doc_root }}/common/services"

georgejhunt, 2017-07-20:
I suspect it should be - "{{ doc_root }}/services"

holta, 2017-07-20:
> I suspect it should be - "{{ doc_root }}/services"

http://box/services/power_off.php is what we've documented @ http://FAQ.iiab.io
http://box/common/services/power_off.php is possible if there's a need to alter this.

(For now power_off.php does not appear in either directory during installation of IIAB 6.3)

tim-moody, 2017-07-20:
looks like https://github.com/tim-moody/iiab/blob/6.3-pwroff/roles/httpd/tasks/html.yml never got the copy of services like the other common files, so that is why power_off.php is not there.

the intended url is /common/services

holta, 2017-07-20:
FYI http://box/services/power_off.php is what I shipped to Guatemala & Peru (earlier today).

If http://box/common/services/power_off.php is definitely the official+working method going forward, someone just reconfirm (and I'll change the documentation).

tim-moody, 2017-07-20:
actually master looks right, so maybe this is the back port to 6.3 that's the problem

holta, 2017-08-27:
Let's declare http://box/common/services/power_off.php the official POWEROFF button going forward.  It's an awful long URL but few will notice and it'll do :-)

closed 2017-08-27 by holta

holta, 2017-09-02:
Ref #57