iiab/iiab #149: "osm_install: False" in local_vars.yml fails on master too

holta, 2017-07-21:
Context: @jvonau helped fix "**python_path**" in https://github.com/iiab/iiab/commit/e426b10ec394a92132af552a8246f2c467c12f07 but that wasn't quite enough, as  ./runansible (on a clean IIAB 6.4 master) fails as follows:

TASK [osm : Copy the files] ****************************************************
fatal: [127.0.0.1]: FAILED! => {"failed": true, "msg": "'**osm_path**' is undefined"}
       to retry, use: --limit @/opt/iiab/iiab/iiab.retry

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=281  changed=187  unreachable=0    failed=1

jvonau, 2017-07-21:
either of the mods in PR #150 should work

jvonau, 2017-07-21:
Post the full trace of osm, I can't reproduce this, should be skipping all the tasks
TASK [osm : Install IIAB required packages] ************************************
skipping: [127.0.0.1] => (item=gcc) 
skipping: [127.0.0.1] => (item=python-dev) 
skipping: [127.0.0.1] => (item=liblzma-dev) 
skipping: [127.0.0.1] => (item=libapache2-mod-xsendfile) 
skipping: [127.0.0.1] => (item=libapache2-mod-wsgi) 

TASK [osm : fix the pip bug (incompatible with requests)] **********************
skipping: [127.0.0.1]

TASK [osm : Install IIAB required packages] ************************************
skipping: [127.0.0.1] => (item=gcc) 
skipping: [127.0.0.1] => (item=python-pip) 
skipping: [127.0.0.1] => (item=python-devel) 
skipping: [127.0.0.1] => (item=xz-devel) 
skipping: [127.0.0.1] => (item=mod_xsendfile) 
skipping: [127.0.0.1] => (item=mod_wsgi) 

TASK [osm : Install latest MarkupSafe] *****************************************
skipping: [127.0.0.1]

TASK [osm : Install latest pytz] ***********************************************
skipping: [127.0.0.1]

TASK [osm : Install IIAB with pip] *********************************************
skipping: [127.0.0.1]

TASK [osm : remove Whoosh with pip] ********************************************
skipping: [127.0.0.1]

TASK [osm : Install Whoosh 2.6 with pip] ***************************************
skipping: [127.0.0.1]

TASK [osm : Set osm_path] ******************************************************
skipping: [127.0.0.1]

TASK [osm : Copy IIAB config file] *********************************************
skipping: [127.0.0.1]

TASK [osm : Create a link from sites-enabled to sites-available] ***************
skipping: [127.0.0.1]

TASK [osm : Remove the link from sites-enabled to sites-available] *************
skipping: [127.0.0.1]

TASK [osm : Create link to cgi] ************************************************
skipping: [127.0.0.1]

TASK [osm : Create link to cgi] ************************************************
skipping: [127.0.0.1]

TASK [osm : Restart httpd service] *********************************************
skipping: [127.0.0.1]

TASK [osm : Create the knowledge data set folders] *****************************
skipping: [127.0.0.1]

TASK [osm : Copy the files] ****************************************************
skipping: [127.0.0.1]

TASK [osm : add osm to service list] *******************************************
skipping: [127.0.0.1] => (item={u'option': u'name', u'value': u'Internet-in-a-Box'}) 
skipping: [127.0.0.1] => (item={u'option': u'description', u'value': u'"The Internet-in-a-Box is a small, inexpensive device which provides essential Internet resources without any Internet connection. It provides a local copy of half a terabyte of the world\u2019s Free information."'}) 
skipping: [127.0.0.1] => (item={u'option': u'enabled', u'value': False}) 
skipping: [127.0.0.1] => (item={u'option': u'path', u'value': u'/osm'})

holta, 2017-07-21:
Here's the result of running "./runtags osm" with "osm_install: False" in local_vars.yml on Raspbian Lite:

```
TASK [osm : Install IIAB required packages] ************************************
skipping: [127.0.0.1] => (item=[])

TASK [osm : fix the pip bug (incompatible with requests)] **********************
skipping: [127.0.0.1]

TASK [osm : Install IIAB required packages] ************************************
skipping: [127.0.0.1] => (item=[])

TASK [osm : Install latest MarkupSafe] *****************************************
skipping: [127.0.0.1]

TASK [osm : Install latest pytz] ***********************************************
skipping: [127.0.0.1]

TASK [osm : Install IIAB with pip] *********************************************
skipping: [127.0.0.1]

TASK [osm : remove Whoosh with pip] ********************************************
skipping: [127.0.0.1]

TASK [osm : Install Whoosh 2.6 with pip] ***************************************
skipping: [127.0.0.1]

TASK [osm : Set osm_path] ******************************************************
skipping: [127.0.0.1]

TASK [osm : Copy IIAB config file] *********************************************
skipping: [127.0.0.1]

TASK [osm : Create a link from sites-enabled to sites-available] ***************
skipping: [127.0.0.1]

TASK [osm : Remove the link from sites-enabled to sites-available] *************
skipping: [127.0.0.1]

TASK [osm : Create link to cgi] ************************************************
skipping: [127.0.0.1]

TASK [osm : Create link to cgi] ************************************************
skipping: [127.0.0.1]

TASK [osm : Restart httpd service] *********************************************
skipping: [127.0.0.1]

TASK [osm : Create the knowledge data set folders] *****************************
skipping: [127.0.0.1]

TASK [osm : Copy the files] ****************************************************
fatal: [127.0.0.1]: FAILED! => {"failed": true, "msg": "'osm_path' is undefined"}
        to retry, use: --limit @/opt/iiab/iiab/iiab.retry

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=61   changed=17   unreachable=0    failed=1
```

holta, 2017-07-21:
Clarif: I've not tried to apply proposed fix #150 on the RPi3 that generated the above output, as this is exactly what "borked" my other RPi3 -- when I followed these "command line instructions" specified near the bottom of https://github.com/iiab/iiab/pull/150 :

```
git checkout -b jvonau-osm-path master
git pull https://github.com/jvonau/iiab.git osm-path

```
(So I'm hesitant to "break" another RPi3 SD image in the same way...each of which completed runansible successfully using the "osm_install: True" workaround..._or is there a better way to smoke-test this or similar proposed fixes?)_

jvonau, 2017-07-21:
Sorry I don't see the  "command line instructions" in 150... I use what I know works, see the email about adding a stanza. I might of broke my own rule about git pull origin master before branching and been one commit behind master in the PR,  that might of tossed the git pull of whack.(no common git hash to be found, rebased/pushed) Maybe for the development cycle the depth of the initial git clone in load.txt can go a bit deeper than 1, maybe 10 just to ease testing woes?

jvonau, 2017-07-22:
FWIW, I'm thinking the template module should be skipped but is not is a bug in ansible 2.2.0. none of PR 150 is really need on 2.3.1 except for the oversight for disabling osm on redhat.

holta, 2017-09-02:
@jvonau do you happen to know if we can close this?

Or do we need 1 more test with either below?
http://download.iiab.io/6.4/rpi/load-min.txt
http://download.iiab.io/6.4/rpi/load-min-vpn.txt

holta, 2017-09-18:
@jvonau believes this is largely solved by #150

closed 2017-09-18 by holta