iiab/iiab #158: Running Ansible on prisine Ubuntu LTS fails.

arky, 2017-08-23:
Trying to install IIAB following the build from the scratch instructions. The machine is pristine Ubuntu LTS, however the install fails while running the ansible section with the following error.  

#####
Error
######
Running local playbooks!                                                                                                                               
                                                                                                                                                       
PLAY [all] *********************************************************************                                                                       
                                                                                                                                                       
TASK [setup] *******************************************************************                                                                       
ok: [127.0.0.1]                                                                                                                                        
                                                                                                                                                       
TASK [1-prep : Set flag for fedora 18] *****************************************                                                                       
skipping: [127.0.0.1]                                                                                                                                  
                                                                                                                                                       
TASK [1-prep : abort if the OS is not supported] *******************************                                                                       
ok: [127.0.0.1] => {                                                                                                                                   
    "failed": false,                                                                                                                                   
    "failed_when_result": false,                                                                                                                       
    "msg": "ubuntu Operating System is not supported"
}

jvonau, 2017-08-23:
I've seen this in other logs with Ubuntu but the install continued(think it's a bogus warning). Are there more lines present? Can you attach the log file to this issue please.

arky, 2017-08-23:
@jvonau Thanks, You are right. it was the missing 'apparmor' ?

######################
Error 
######################
TASK [1-prep : get the value into a variable] **********************************                                                              [752/887]
ok: [127.0.0.1]                                                                                                                           
                                                                                                                   
TASK [1-prep : Install chrony package] *****************************************                                            
changed: [127.0.0.1] => (item=[u'chrony'])                                                                                                
                                                                                                                                        
TASK [1-prep : Update chrony config file] **************************************                                      
changed: [127.0.0.1]                                                                                               
                                                                                                                   
TASK [1-prep : Disable apparmor -- on by default in ubuntu] ********************                                                                       
fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "no service or tool found for: apparmor"}                                     
        to retry, use: --limit @/opt/iiab/iiab/iiab.retry                                                                                  
                                                                                                                                            
PLAY RECAP *********************************************************************                                                        
127.0.0.1                  : ok=11   changed=6    unreachable=0    failed=1    
##################################################################

jvonau, 2017-08-23:
That is a real error, please reopen this issue.

arky, 2017-08-23:
Done!

jvonau, 2017-08-23:
@holta @georgejhunt From the [ML fpaste](https://paste.fedoraproject.org/paste/7c0~KknqZUB-FlaG52hRPg), it appears that apparmor is installed and was disabled noted by TASK changed.

2017-08-20 18:43:28,791 p=26381 u=root |  TASK [1-prep : Install chrony package] *****************************************
2017-08-20 18:44:01,581 p=26381 u=root |  changed: [127.0.0.1] => (item=[u'chrony'])
2017-08-20 18:44:01,663 p=26381 u=root |  TASK [1-prep : Update chrony config file] **************************************
2017-08-20 18:44:03,622 p=26381 u=root |  changed: [127.0.0.1]
2017-08-20 18:44:03,703 p=26381 u=root |  TASK [1-prep : Disable apparmor -- on by default in ubuntu] ********************
2017-08-20 18:44:06,298 p=26381 u=root |  changed: [127.0.0.1]
2017-08-20 18:44:06,380 p=26381 u=root |  TASK [1-prep : Disable selinux on next boot] ***********************************
2017-08-20 18:44:06,492 p=26381 u=root |  skipping: [127.0.0.1]

However this issue demo's that apparmor may not always be installed.

jvonau, 2017-08-23:
@arky just to clarify are you testing release-6.3 or master branch?

arky, 2017-08-23:
I was testing with Master branch.

holta, 2017-08-24:
@arky: Any progress you can report?  Feel free to join our live Community/Team Call 10:30AM NYC Time if it helps!

Agenda: http://minutes.iiab.io

holta, 2017-08-24:
@arky: @jvonau also meant to ask you what exact version of Ubuntu are you running?

e.g. can you recheck if it's 16.04.3 or similar?

tim-moody, 2017-08-24:
the code in question iiab/scripts/local_facts.fact

the first question is why is the message "msg": "ubuntu Operating System is not supported" without any version, so we need to know what version of ubuntu you're using. I would have expected ubuntu-16 to be the value of OS_VER.

holta, 2017-08-24:
@arky: in case you cannot join the call, please do share more about your work in Cambodia if poss!

We have many other "OLPC Friends" there in Cambodia, doing similar work with schools~

jvonau, 2017-08-24:
- name: abort if the OS is not supported
  debug: msg="{{ ansible_local.local_facts.os }} Operating System is not supported"
failed_when: ansible_local.local_facts.os_ver == "OS_not_supported"

This is a misleading debug message, should never make it past iiab.yml without having an os-ver.yml file present.

arky, 2017-08-24:
Thanks all for trying to help me with this problem. Unfortunately I am unable to reproduce the problem when I started over. Am new to Ansible so it is definitely something I did wrong. 

 PS: I will try to join the community call. Thanks for the heads up.

closed 2017-08-24 by arky

holta, 2017-08-24:
@arky: plz email me if poss (holt @ laptop.org) to get audio call info.

jvonau, 2017-08-25:
PR #161 allowed the install to progress, but I suspect apparmor is still active as dhcpd is affected and perhaps named as in #67

jvonau, 2017-08-26:
---------- Forwarded message ----------
Date: Wed, Aug 23, 2017 at 2:29 PM
Subject: Re: Installation of IIAB on ubuntu

Thanks a lot George and Adam for the help. Using an Ethernet connection worked like a charm and the installation was successful.  I however could not see the IIAB wireless SSID on any of my other computers.Hence was not able to  connect to  http://box,  http://box.lan, http://schoolserver.lan, http://localhost or http://172.18.96.1  from the browser.  I Could not figure out how to turn on or make the server AP accessible to other computers.

I then  noticed my wireless interface card is an intel wifi link 5100 and the drivers for it are "iwlwifi", which according to this website http://linuxwireless.org/en/users/Drivers/   should support AP mode. However when i ran "iw list"  and  looked in the 'Supported interface mode*s' section for "*AP",  i did not see "AP" listed. The only mode available was "Monitor".

At this point i got a small external WIFI router, disabled security and dhcp and renamed it to "internet in a box" for easy visibility. I manually put the an ip of 172.18.96.101 subnet 255.255.224.0 on the router and connected it to the server with a usb to ethernet adapter.  At this point i knew for sure the external router and IIAB server were in the same network. However, i was still not able to connect to the school server from a browser. Its only after connecting to another laptop using an Ethernet cable and manually setting the ip address of the laptop to 172.18.96.99 that i was able to access the server.

So i will keep playing with my WIFI interface card to see if it can work at  all, but i will appreciate any help with setting the server up with an external wireless router so that the clients can access the server wirelessly.

tim-moody, 2017-08-26:
the expected ip addr of an iiab lan client in 172.18.100.20x

the external lan side router standard ip is 172.18.126.10x

make sure that the device to which the external router is connected is in br0

network mode should be gateway or lan controller

jvonau, 2017-08-26:
the key part is 
> However, i was still not able to connect to the school server from a browser. Its only after connecting to another laptop using an Ethernet cable and manually setting the ip address of the laptop to 172.18.96.99 that i was able to access the server

So br0/wired_interface enp0s25 is available but not handing out dhcp leases.

and from the email thread:
> My last error on ./runansible was still
  unable to start service: job for dhcpd.service fails because the control process exited with error code.

jvonau, 2017-09-07:
[July 29 JD reported the same apparmor issue ](https://groups.google.com/forum/#!topic/xsce-devel/OIW9_m89Juo)