iiab/iiab #165: raspbian-9.xml

tim-moody, 2017-08-26:
should this have is_debian_9: True?

holta, 2017-08-26:
@jvonau do you have an opinion?

Is Raspbian Stretch similar enough to deserve "is_debian_9: True" ?

jvonau, 2017-08-26:
yes, that is PR #138

holta, 2017-09-02:
@tim-moody @jvonau can we close this, or does it need discussion in Monday's community/team call?  (http://minutes.iiab.io)

tim-moody, 2017-09-02:
not an enhancement. it is either a bug or not necessary, but it has not yet been implemented in master.
can't say if it is in a PR

jvonau, 2017-09-03:
I've reworked #138 to address just the missing dependencies on raspbian-9 so I will now go with no not needed.

holta, 2017-09-13:
@tim-moody we need a verdict here to move forward: can you help us reach a conclusion now?

tim-moody, 2017-09-13:
yes. raspbian9 is debian 9

closed 2017-09-13 by holta