iiab/iiab #166: current master fails on raspbian stretch

tim-moody, 2017-08-26:
TASK [network : reset the eth0 interface] **************************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["ifdown", "etho"], "delta": "0:00:00.012559", "end": "2017-08-26 19:32:19.178094", "failed": true, "rc": 1, "start": "2017-08-26 19:32:19.165535", "stderr": "ifdown: unknown interface etho", "stdout": "", "stdout_lines": [], "warnings": []}
...ignoring

TASK [network : restart the networking service] ********************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "Unable to restart service networking: Job for networking.service failed because the control process exited with error code.\nSee \"systemctl status networking.service\" and \"journalctl -xe\" for details.\n"}
        to retry, use: --limit @/opt/iiab/iiab/iiab.retry

N.B. cmd": ["ifdown", "etho"], unlikely was ever smoke tested.

jvonau, 2017-08-26:
I caught this when I reviewed the [XSCE PR](https://github.com/XSCE/xsce/pull/913) but it was never corrected.

jvonau, 2017-08-26:
typo fix #168

tim-moody, 2017-08-26:
looking further, I don't think the etho typo is not the root cause as there is an  ignore_errors: True clause and this never gave a problem before. I notice that the error that killed it is the next one where networking failed to start with the message ifup: failed to bring up eth0 (no typo)

the rpi3 I'm running on does not have an eth0, but rather an enxb827eb6f70ac device as the gateway. I don't understand that, but it is recently purchased and/or stretch changed the naming.

jvonau, 2017-08-26:
> the rpi3 I'm running on does not have an eth0, but rather an enxb827eb6f70ac device as the gateway. I don't understand that, but it is recently purchased and/or stretch changed the naming.
PR #160 should of forced that to be eth0, are you on the latest master?

jvonau, 2017-08-26:
> looking further, I don't think the etho typo is not the root cause as there is an ignore_errors: True clause and this never gave a problem before. I notice that the error that killed it is the next one where networking failed to start with the message ifup: failed to bring up eth0 (no typo)

"ifdown:  was suppose to release "eth0" but didn't so that blocks trying use that interface when bringing up the network in the next stanza.

jvonau, 2017-08-26:
I'll see if I can softcode the variables in rpi_debian.yml so either way works.

jvonau, 2017-08-27:
softcoding part of #169, see the comments there also.

tim-moody, 2017-08-27:
Further experiments:

Rpi3 A                 with jessie have eth0  with stretch have eth0
Rpi3 B (new)       with jessie have eth0  with stretch have enxb827eb6f70ac, but after reboot have eth0

and if I rerun ansible it now succeeds ?!

tim-moody, 2017-08-27:
this was without applying #168 or #169

jvonau, 2017-08-27:
There needs a reboot for this piece of code to work: 5fdbe61aab9c9a459f576cfb9a4b377c2ff6b376 with the need to reboot forced to take place sooner with PR #160

jvonau, 2017-08-27:
Now test to see if these work:
iiab_lan_enabled: False
iiab_wan_enabled: False

jvonau, 2017-08-27:
>  ifup: failed to bring up eth0 
> stretch have enxb827eb6f70ac, but after reboot have eth0
> and if I rerun ansible it now succeeds ?!

Yea, it now matches the hardcoding of eth0 in rpi_debian.yml which was not found when the device was known as enxb827eb6f70ac

jvonau, 2017-08-27:
IIAB on Ubuntu email thread has the manual --> dhcp from #156 change now causing dhcpcd to fail in debian.yml. Think you can confirm that the generated /etc/network/interface template works as a gateway on rasbian-9 with 'dhcp' and it is shared. So lets not start 'dhcpcd' on ubuntu? or revert?

jvonau, 2017-08-27:
IIAB on Ubuntu email thread:
> .  "not running dhcpcd because /etc/network/interfaces defines some interfaces that will use DHCP client"

The manual --> dhcp from #156 I believe is causing dhcpcd to fail in debian.yml, and stopping at that point. Think you can confirm that the generated /etc/network/interface template works as a gateway on rasbian-9 with 'dhcp' and it is shared, so disable dhcpcd on non-rpi? or revert?

jvonau, 2017-08-28:
I'm thinking ubuntu-16, debian-9, raspbian-9 more or less behave the same in respect to how the network can be configured, based on the reaction of the systems to dhcp in the currently supplied interfaces file and would be the preferred method. I'm thinking raspbian-8 needing the manual method is the exception, with that note there is a wip branch as part of #171.

tim-moody, 2017-08-28:
perhaps we should abandon jessie.

jvonau, 2017-08-28:
With a bit of work and testing I think #171 will be able to support all flavours in short order.

holta, 2017-08-28:
I agree with Tim that supporting Jessie+derivatives is not worth it going forward.

(To avoid burning out our small group especially, we are not IBM!)

jvonau, 2017-08-30:
related #7

jvonau, 2017-08-31:
#178 replacing #168 #169

jvonau, 2017-08-31:
[JK-dhcpcd.txt](https://github.com/iiab/iiab/files/1265977/JK-dhcpcd.txt) selected interesting results of systemctl list-unit-files from JK's rpi set with a static ip

jvonau, 2017-08-31:
systemd's https://www.freedesktop.org/wiki/Software/systemd/PredictableNetworkInterfaceNames/
Nice list of [network managers](https://wiki.archlinux.org/index.php/Network_configuration#Change_device_name) and other useful info.

jvonau, 2017-09-01:
PR #178 should be testable

holta, 2017-09-02:
Ref #198 (rpi ex virgine: dhcpcd research)

holta, 2017-09-13:
@tim-moody I do not see "networking.service failed" during ./runansible so am closing this issue (please re-open if helpful & nec!)

closed 2017-09-13 by holta