iiab/iiab #175: Nextcloud install incomplete due PHP issues (Raspbian Stretch only? Debian 9 too?)

holta, 2017-08-30:
Does anyone know if this same incomplete installation occurs on Debian 9?

In any case, I've confirmed this across 2 different installations of http://download.iiab.io/6.4/rpi/load-vpn.txt on Raspbian Stretch Lite from 2017-08-16.

Here is what appears in browser when visiting http://box/nextcloud or http://192.168.0.x/nextcloud :

    PHP module zip not installed.
    Please ask your server administrator to install the module.

    PHP module mb multibyte not installed.
    Please ask your server administrator to install the module.

    PHP modules have been installed, but they are still listed as missing?
    Please ask your server administrator to restart the web server.

jvonau, 2017-08-30:
It looks to be a [name change issue](https://github.com/iiab/iiab/search?utf8=%E2%9C%93&q=php-zip&type=) is going on with php-zip while roles/nextcloud/tasks/main.yml is using the not using the {{ php_version }} maybe php-zip should move to the next stanza? Not sure about the other one, missing dependency maybe?

holta, 2017-09-01:
Is #69 on Ubuntu related and/or helpful?

(Certainly we experienced very much the same kind of Nextcloud PHP errors on Ubuntu 16.04 in late June!)

jvonau, 2017-09-01:
Might be the cause if all the dependencies are satisfied

holta, 2017-09-03:
Close this issue if PR #202 works ("Elgg upgrade 1.12.16 LTS to 2.3.3; Nextcloud - add php's zip mbstring")

holta, 2017-09-13:
@floydianslips can you confirm http://box.lan/nextcloud works?

(since #202 was pushed 8 days ago)

holta, 2017-09-13:
http://box.lan/nextcloud works when installed the latest IIAB 6.4 master from http://download.iiab.io/6.4/rpi/load-vpn.txt onto Raspbian Stretch Lite 2017-09-07, thanks for everyone's help !

closed 2017-09-13 by holta

diplodata, 2018-06-23:
So what's the solution??