iiab/iiab #176: RPi installs all require "tune2fs -m 1 /dev/mmcblk02" (disk reserve: ~5% to 1%)

holta, 2017-08-30:
Where does "tune2fs -m 1 /dev/mmcblk02" belong in our RPi scripts, so as to ensure *all* IIAB RPi installs start out with this new default, long before any [minimize scripts](https://github.com/iiab/iiab-factory/blob/master/box/rpi/min-sd) are run?  (For many reasons, min-sd is often not run at all!)

This tune2fs command can easily be added to http://download.iiab.io/6.4/rpi/load.txt and similar scripts if necessary, but a more general/upstream solution would be better!

Context: users of MicroSD cards inside Raspberry Pi-like computers in developing countries cannot afford to throw away 5% of their "disk" space, particularly when this means Gigabytes never used, and worse when this 5% offers a false sense of security to large numbers of people that have no Linux skills to deal with the situation when they do hit 5%.

So we are changing from ~5% to 1% reserve space on RPi for now, using "tune2fs -m 1 /dev/mmcblk02" to try this new default.  A bit more backstory here: http://lists.laptop.org/pipermail/server-devel/2017-August/008193.html

_This does not answer the underlying question of how to semi-automagically purge SD cards' stale ("LRU") Content Packs (and bloated log files if nec, etc) which is a UX-driven challenge we urgently need to face as we approach 2018+onwards !_

tim-moody, 2017-08-30:
please see https://github.com/iiab/iiab-factory/blob/master/box/rpi/ch-reserve.sh

holta, 2017-08-30:
Should something in this repo (iiab/iiab) make a call to iiab/iiab-factory's ch-reserve.sh to guarantee this is run every time?

Or is there a better way to structure this?  (Hopefully @georgejhunt has ideas!)

holta, 2017-09-01:
Per rough consensus during our Thur call ([minutes.iiab.io](http://minutes.iiab.io)) I've added "tune2fs -m 1 /dev/mmcblk02" to [load.txt](http://download.iiab.io/6.4/rpi/load.txt) (and likewise similar scripts in [download.iiab.io/6.4/rpi](http://download.iiab.io/6.4/rpi/) for smaller/vpn-enhanced installations of IIAB 6.4!)

RELATED: Huge thanks also to @tim-moody for [github.com/iiab/iiab-factory/blob/master/box/generic/make-content-ready.sh](https://github.com/iiab/iiab-factory/blob/master/box/generic/make-content-ready.sh) whose essential 5 lines I've also added, so new implementers now have a _far better Kiwix & KA Lite experience almost immediately after load.txt completes_ (but please do reboot as soon as load.txt completes!)

In future load.txt (etc) should be published within [github.com/iiab/iiab-factory/box](github.com/iiab/iiab-factory/box) and hopefully modularized too?

closed 2017-09-01 by holta