iiab/iiab #182: unable to get static ip on debian 9

tim-moody, 2017-08-31:
in config vars
gui_static_wan_gateway: "10.100.0.1"
gui_desired_network_role: "Appliance"
gui_static_wan_ip: "10.100.0.170"
gui_static_wan_nameserver: "10.100.0.1"
gui_static_wan_netmask: "255.255.255.0"

resultant address: 10.100.0.96

tim-moody, 2017-08-31:
root@iiab-demo:~# cat /etc/network/interfaces
# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

source /etc/network/interfaces.d/*

# The loopback network interface
auto lo
iface lo inet loopback

# iiab_network_mode is Appliance

#################   APPLIANCE #########################
auto eth0
iface eth0 inet manual

tim-moody, 2017-08-31:
in xsce 6.2 time frame this worked

tim-moody, 2017-08-31:
and the iface name has switched to eth0

tim-moody, 2017-08-31:
operator error. static check box was not checked.

tim-moody, 2017-08-31:
still, after checking static, save and install conf option, reboot, it doesn't seem right to have

2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP gr                                                                                                              oup default qlen 1000
    link/ether b8:ae:ed:71:a6:92 brd ff:ff:ff:ff:ff:ff
    inet 10.100.0.170/24 brd 10.100.0.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet 10.100.0.96/24 brd 10.100.0.255 scope global secondary eth0
       valid_lft forever preferred_lft forever

jvonau, 2017-08-31:
looks like you have both static and dhcp suppled addresses.

jvonau, 2017-08-31:
I would try 'static' where 'manual' is used. or # out the whole line.

tim-moody, 2017-08-31:
I didn't mention that after checking static /etc/network/interfaces has

# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

source /etc/network/interfaces.d/*

# The loopback network interface
auto lo
iface lo inet loopback

# iiab_network_mode is Appliance

#################   APPLIANCE #########################
auto eth0
 # gui_static_wan_ip is set
iface eth0 inet static
    address 10.100.0.170
    netmask 255.255.255.0
    gateway 10.100.0.1
    dns-nameservers 10.100.0.1
    dns-search iiab.me

jvonau, 2017-08-31:
This sucks, think one has to define the static stuff in /etc/dhcpcd.conf so dhcpcd will ignore the interface. Turning off dhcpcd is not an option as that will break hotpluging so cell tethering would go out the window without it I would guess(to be proven).

tim-moody, 2017-08-31:
I just don't see what triggers dhcpcd to get an ip addr

root@iiab-demo:/opt/iiab/iiab# systemctl status dhcpcd
● dhcpcd.service - LSB: IPv4 DHCP client with IPv4LL support
   Loaded: loaded (/etc/init.d/dhcpcd; generated; vendor preset: enabled)
   Active: active (running) since Thu 2017-08-31 16:38:31 EDT; 3h 12min ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/dhcpcd.service
           └─558 /sbin/dhcpcd

Aug 31 16:38:34 iiab-demo.iiab.me dhcpcd[558]: DUID 00:01:00:01:21:39:b6:f5:b8:ae:ed:71:a6:92
Aug 31 16:38:34 iiab-demo.iiab.me dhcpcd[558]: eth0: IAID ed:71:a6:92
Aug 31 16:38:34 iiab-demo.iiab.me dhcpcd[558]: eth0: soliciting an IPv6 router
Aug 31 16:38:35 iiab-demo.iiab.me dhcpcd[558]: eth0: rebinding lease of 10.100.0.96
Aug 31 16:38:35 iiab-demo.iiab.me dhcpcd[558]: eth0: probing address 10.100.0.96/24
Aug 31 16:38:39 iiab-demo.iiab.me dhcpcd[558]: eth0: leased 10.100.0.96 for 86400 seconds
Aug 31 16:38:39 iiab-demo.iiab.me dhcpcd[558]: eth0: adding route to 10.100.0.0/24
Aug 31 16:38:39 iiab-demo.iiab.me dhcpcd[558]: eth0: adding default route via 10.100.0.1
Aug 31 16:38:40 iiab-demo.iiab.me dhcpcd[558]: eth0: removing route to 10.100.0.0/24
Aug 31 16:38:40 iiab-demo.iiab.me dhcpcd[558]: eth0: removing default route via 10.100.0.1
root@iiab-demo:/opt/iiab/iiab# cat /etc/dhcpcd.conf
# A sample configuration for dhcpcd.
# See dhcpcd.conf(5) for details.

# Allow users of this group to interact with dhcpcd via the control socket.
#controlgroup wheel

# Inform the DHCP server of our hostname for DDNS.
hostname

# Use the hardware address of the interface for the Client ID.
#clientid
# or
# Use the same DUID + IAID as set in DHCPv6 for DHCPv4 ClientID as per RFC4361.
# Some non-RFC compliant DHCP servers do not reply with this set.
# In this case, comment out duid and enable clientid above.
duid

# Persist interface configuration when dhcpcd exits.
persistent

# Rapid commit support.
# Safe to enable by default because it requires the equivalent option set
# on the server to actually work.
option rapid_commit

# A list of options to request from the DHCP server.
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
# Most distributions have NTP support.
option ntp_servers
# Respect the network MTU. This is applied to DHCP routes.
option interface_mtu

# A ServerID is required by RFC2131.
require dhcp_server_identifier

# Generate Stable Private IPv6 Addresses instead of hardware based ones
slaac private

jvonau, 2017-09-01:
> 6:38:35 iiab-demo.iiab.me dhcpcd[558]: eth0: rebinding lease of 10.100.0.96

dhcpcd by default will respond on all interfaces unless told otherwise, plus we are not stopping the service before restarting the network(even before for debian.yml), so as soon as the interface becomes available dhcpcd reacts. see PR #178/files

tim-moody, 2017-09-01:
this seems relevant
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=140252
especially

Since the move to Jessie, we've had to change the way we give our Raspberry Pi's a static IP address . Rather than setting it in /etc/network/interfaces, it has to be done in /etc/dhcpcd.conf. There are examples of simple configurations of dhcpcd.conf going around that look like this:

jvonau, 2017-09-01:
That is exactly what I'm talking about and I've never seen the forum.

tim-moody, 2017-09-01:
and this

https://raspberrypi.stackexchange.com/questions/39785/dhcpcd-vs-etc-network-interfaces

jvonau, 2017-09-01:
It would nice to know the same base info as in the email thread 171 before resolvconf and dhcpcd are altered by the playbook.

jvonau, 2017-09-01:
PR #178 should be testable

holta, 2017-09-08:
While non-ideal, @tim-moody clarified during this morning's call (http://minutes.iiab.io) that there is a workaround on Debian 9 (or possibly all Linuxes) here, since 2 IP addresses result: the static one can be used while the other/dynamic one can be ignored.

jvonau, 2017-11-07:
#490

jvonau, 2017-11-12:
file a new issue if you see this again

holta, 2017-11-12:
@tim-moody please re-open this ticket or file a new one as appropriate, Thanks!

closed 2017-11-12 by holta