iiab/iiab #188: Elgg 1.12.12 & 1.12.16 don't show anything in browser (500 Internal Server Error)

holta, 2017-09-01:
@jvonau asks if a "schoolserver" reference that he spotted in Elgg's sql dump/log might be causing this?

Full Details in PR #179 (upgrade elgg from 1.12.12 to 1.12.16) and issue #145 (sudden loss of power locks elgg/mysql).

jvonau, 2017-09-01:
Since php is involved adding #69. My 1.12.12 install was working as the correct extension_dir  is defined in /etc/php.ini for my platform. Quick test rename /etc/php.ini and restart apache2.

tim-moody, 2017-09-02:
http://iiab.me/elgg/ brings up page, though css seems missing

holta, 2017-09-02:
Close this issue if PR #202 works (Elgg upgrade from 1.12.16 LTS to 2.3.3)

holta, 2017-09-04:
#202 merged; awaits final testing so we can close this.

closed 2017-09-04 by holta

jvonau, 2017-09-23:
It's a dns issue, if you add the server ip address to /etc/hosts it works.

> GET 
http://10.8.0.10/elgg/ [HTTP/1.1 200 OK 317ms]
GET 
http://box/elgg/cache/1504389501/default/font-awesome/css/font-awesome.css [HTTP/1.1 200 OK 253ms]
GET 
http://box/elgg/cache/1504389501/default/elgg.css [HTTP/1.1 200 OK 311ms]
GET 
http://box/elgg/cache/1504389501/default/lightbox/elgg-colorbox-theme/colorbox.css [HTTP/1.1 304 Not Modified 216ms]
GET 
http://box/elgg/cache/1504389501/default/jquery.js [HTTP/1.1 304 Not Modified 968ms]
GET 
http://box/elgg/cache/1504389501/default/jquery-ui.js [HTTP/1.1 304 Not Modified 1239ms]
GET 
http://box/elgg/cache/1504389501/default/elgg/require_config.js [HTTP/1.1 304 Not Modified 1395ms]
GET 
http://box/elgg/cache/1504389501/default/require.js [HTTP/1.1 304 Not Modified 328ms]
GET 
http://box/elgg/cache/1504389501/default/elgg.js [HTTP/1.1 304 Not Modified 361ms]
Expected 'none', URL, or filter function but found 'progid'.  Error in parsing value for 'filter'.  Declaration dropped. colorbox.css:65:12
Unknown property '-moz-osx-font-smoothing'.  Declaration dropped. font-awesome.css:4:660
Error in parsing value for 'background-clip'.  Declaration dropped. elgg.css:948:19
Error in parsing value for 'background-clip'.  Declaration dropped. elgg.css:958:19
Password fields present on an insecure (http://) page. This is a security risk that allows user login credentials to be stolen.[Learn More] (unknown)
Password fields present in a form with an insecure (http://) form action. This is a security risk that allows user login credentials to be stolen.[Learn More] (unknown)
GET 
http://box/elgg/cache/1504389501/default/languages/en.js [HTTP/1.1 304 Not Modified 98ms]
GET 
http://box/elgg/cache/1504389501/default/elgg/init.js [HTTP/1.1 304 Not Modified 112ms]
GET 
http://box/elgg/cache/1504389501/default/elgg/ready.js [HTTP/1.1 304 Not Modified 110ms]
GET 
http://box/elgg/cache/1504389501/default/elgg/Plugin.js [HTTP/1.1 304 Not Modified 94ms]