iiab/iiab #191: Our Ansible messaging is unnecessarily scaring away new contributors

holta, 2017-09-02:
3 examples, that longtime Mozilla community manager @arky reminds us have very real consequences -- let's start by cleaning up false alarms & misinformation that sends people running -- towards the authentic longer-term goal that Arky outlines: (actionable Ansible messaging, quickly understandable by free software/open source veterans, with verbose flag etc if nec?)

1. Right on top “ubuntu Operating System is not supported” and “raspbian Operating System is not supported" causes people to (A) give up and/or (B) not believe anything they read from that point onwards :/  #158
2. Our installation of Ansible 2.2.0 sets off loud alarm bells in new contributors when it asks to upgrade pip from “8.1.1 to 9.0.1”.  Can “pip install ansible==2.2 --disable-pip-version-check” be added to scripts/ansible?  PR #181
3. Bright Red messaging within “ignored” sections of Ansible runs represent a flood of false alarms, that cause new contributors to Stop Believing Anything They Read.  These can easily be suppressed.  @jvonau has a way to make this happen.

holta, 2017-09-03:
Aside to bring Readable+Topical Messaging to new contributors of all kind:

In #204 (unused/obsolete stuff) @georgejhunt brings in Release Notes 6.0 and 6.1 from our older xsce/xsce repo, so they do not get list, remaining a part of offline IT/implementer docs @ http://box/info

jvonau, 2017-09-05:
Point 3 would of entailed using an installing/installed marker to flag to suppress warnings expected on a first run where we know things are going to be missing, I'm going use nextcloud as the example where one is used in /opt/nextcloud/config/config.php. There was the frame work I put into place to support that but it has been removed(#85 base_ver) except for the the installing boolean. I'm going to say that it's going to be the same old song and dance in one gets proposed so I'm not interested now.

holta, 2017-09-05:
Thanks @jvonau

Few software engineers (alive or dead) ENJOY writing (nevermind maintaining!) clean documentation during their spare time...

Nevermind crisply legible explanations enabling contributors around the planet, whose 1st language is *rarely* English &mdash; a requirement we were not raised to take seriously &mdash; and yet now is firmly upon us...

Nor do many software developers yet have experience with illustrations, https://obsproject.com screen capture walk-thrus, or http://simpleshow.com cartoons that the world HAS come to expect !

_Hence the reason we all need to push ourselves forward together on this Epic Front -_

holta, 2017-09-07:
@arky provided/provoked priceless de-obfuscation ideas during our community/team call today (http://minutes.iiab.io) for us to make Ansible far more humanely+quickly understandable for new developers especially~

_Let's try our hardest to implement (several) of these as part of our [IIAB 6.4 Release](https://github.com/iiab/iiab/milestone/1) sometime in coming weeks._

jvonau, 2017-09-19:
Perhaps the author of "ubuntu Operating System is not supported" should revert or do I really need to open a PR for that?

holta, 2017-09-19:
All new users despise the deceptive/erroneous message "[ACTUAL OS] Operating System is not supported"

I've asked @georgejhunt to review #274 (remove "OS not supported" broken conditional).

holta, 2017-11-01:
Thanks @jvonau  & Thanks All!

This work took 2 months to complete and was not easy.

But is now finally converging gracefully towards an APPROACHABLE IIAB 6.5 with critical final touches like:

PR #466 "make Ansible's 9 install stages understandable"
PR #477 "./runansible -> iiab-install avoids repetition of Stages 1-9 when rerun"

closed 2017-11-01 by holta