iiab/iiab #192: dokuwiki pollutes /library with dokuwiki-170219b & dokuwiki-2017-02-19b

holta, 2017-09-02:
Can either one or the other hopefully be removed?

```
root@box:/library# ls -ld dokuwiki*
lrwxrwxrwx 1 root     root   25 Sep  2 11:51 dokuwiki -> /library/dokuwiki-170219b
drwxr-xr-x 2 www-data root 4096 Sep  2 11:51 dokuwiki-170219b
drwxrwxr-x 8 root     root 4096 Mar  7 20:02 dokuwiki-2017-02-19b

root@box:/library# du -hsc dokuwiki*
0       dokuwiki
4.0K    dokuwiki-170219b
29M     dokuwiki-2017-02-19b

```

holta, 2017-09-02:
Likely the underlying story as #36 (can we delete /library/dokuwiki-2014-09-29 ?)

georgejhunt, 2017-09-02:
yes

holta, 2017-09-02:
Is the empty directory ["dokuwiki-170219b"](https://github.com/iiab/iiab/pull/184/files) truly needed for some reason?

Ref PR #184 (get DokuWiki name correct)
Ref PR #136 (bump DokuWiki to current stable version)

jvonau, 2017-09-03:
info used in #208

holta, 2018-09-15:
Still not the prettiest thing in the world, but FYI at least it now shows: (on Raspbian)
```
root@box:/library# ls -ld doku*
lrwxrwxrwx 1 root     root         29 Aug 31 05:15 dokuwiki -> /library/dokuwiki-2018-04-22a
drwxr-xr-x 8 www-data iiab-admin 4096 May  3 02:36 dokuwiki-2018-04-22a
```
...whereas on Debian 9.5 and Ubuntu 18.04 it shows slightly different permissions:
```
root@box:/library# ls -ld doku*
lrwxrwxrwx 1 root     root    29 Sep 14 22:19 dokuwiki -> /library/dokuwiki-2018-04-22a
drwxr-xr-x 8 www-data wheel 4096 May  3 02:36 dokuwiki-2018-04-22a
```

closed 2018-09-15 by holta