iiab/iiab #193: Ansible fails all subsequent runs on Raspbian Stretch: Sugarizer "npm : Depends: node-gyp (>= 0.10.9)"

holta, 2017-09-02:
./runansible ran to completion using http://download.iiab.io/6.4/rpi/load-vpn.txt on Raspbian Stretch Lite on RPi3 (master, about 12 hours ago...after runansible immediate reboot, I ran "cd /opt/iiab/iiab; ./runansible" continuing the load-vpn.txt script from there).

The main page http://box/sugarizer loaded corrected during spot testing.

But then all 3 subsequent runs of "./runansible" or "./runtags sugarizer" failed as follows:

TASK [sugarizer : Actually get it installed] ***********************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["apt", "install", "-y", "npm"], "delta": "0:00:02.159470", "end": "2017-09-02 12:28:13.043157", "failed": true, "rc": 100, "start": "2017-09-02 12:28:10.883687", "stderr": "\nWARNING: apt does not have a stable CLI interface. Use with caution in scripts.\n\nE: Unable to correct problems, you have held broken packages.", "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nSome packages could not be installed. This may mean that you have\nrequested an impossible situation or if you are using the unstable\ndistribution that some required packages have not yet been created\nor been moved out of Incoming.\nThe following information may help to resolve the situation:\n\nThe following packages have unmet dependencies:\n npm : Depends: node-gyp (>= 0.10.9) but it is not going to be installed", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "Some packages could not be installed. This may mean that you have", "requested an impossible situation or if you are using the unstable", "distribution that some required packages have not yet been created", "or been moved out of Incoming.", "The following information may help to resolve the situation:", "", "The following packages have unmet dependencies:", " npm : Depends: node-gyp (>= 0.10.9) but it is not going to be installed"], "warnings": []}
        to retry, use: --limit @/opt/iiab/iiab/iiab.retry

holta, 2017-09-02:
@georgejhunt LMK if you're voting to bypass this issue in IIAB 6.4, if #196 makes it too hard to get Sugarizer working at this time?

(Presumably deferring until [milestone 6.5](https://github.com/iiab/iiab/milestone/2) if so.)

georgejhunt, 2017-09-02:
yes, that's my vote. It looks to me that this is a dependency issue that
will go away if we let stretch settle. There are already tons of citations
about this issue.  We're not the only ones to have experienced it. And it
is a npm issue, not just a sugarizer.

npm is the node packager, and has a lot of visibility

holta, 2017-09-02:
Done: this Sugarizer/npm issue is moved from [milestone 6.4](https://github.com/iiab/iiab/milestone/1) to [milestone 6.5](https://github.com/iiab/iiab/milestone/2) for now.

FYI I don't understand why http://box/sugarizer worked last night during the fresh/master install Raspbian Stretch @jvonau helped me with (and still does work, even if runansible no long works!)

So let's just keep an eye on this evolving situation @llaske might be on top of as well?!

jvonau, 2017-09-02:
Does this error occur now on a new fresh install? If so disable for now if not need to check if installation is really needed before trying to install twice.

holta, 2017-09-02:
> Does this error occur now on a new fresh install?

Yes.  Unless something sudden/dramatic changed over the past 12 to 18 hours.

The following in [local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars_lite.yml) was not enough to get ./runansible to complete:
```
sugarizer_install: True
sugarizer_enabled: False
```
This WAS enough to get ./runansible to complete:
```
sugarizer_install: False
sugarizer_enabled: False
```
_FWIW Sugarizer still loads, when browsing to http://box/sugarizer/ !_

jvonau, 2017-09-02:
I'll go with upstream broke it, disable in the web localvars load scripts
and locally, waiting a day or so would be the best before trying again
locally first.

holta, 2017-09-02:
@jvonau I'll take that as a vote to move Sugarizer (#193) back from [milestone 6.5](https://github.com/iiab/iiab/milestone/2) to [milestone 6.4](https://github.com/iiab/iiab/milestone/1), if we this gets back on track in coming days &mdash; let's try to resolve this live Monday or Thursday @ http://minutes.iiab.io

holta, 2017-09-03:
I moved this ticket (Sugarizer #193) back to Milestone 6.4 for now, as Sugarizer definitely works on Raspbian Stretch, as I've re-verified on the very most current master, and also earlier this week.

If npm is a side effect causing us/all ongoing headaches, let try our best to extinguish that in coming weeks, as expelling a (clearly working) Sugarizer from Internet-in-a-Box for bureaucratic reasons would be Tragic Indeed :/

holta, 2017-09-04:
Ref #216 (npmreinstall)

jvonau, 2017-09-04:
If you can't do a runansible twice then iiab-from-console.yml would be broken also. The above error would of popped up if qusi- functional testing using disregard_network: True was used at some point in the validation. This error would of killed the console out in the field with no internet.

holta, 2017-09-05:
@georgejhunt believe Sugarizer is likely/hopefully fixable for IIAB 6.4 &mdash; thanks to @jvonau's suggestion to bypass npm and install Sugarizer's prereqs directly?  Or some such approach!

holta, 2017-09-08:
Fixed with PR #216 thanks to @georgejhunt & @jvonau

holta, 2017-09-28:
@tim-moody faced this same error today & others face it every week/month:
```
" npm : Depends: node-gyp (>= 0.10.9) but it is not going to be installed"
```
ASIDE: Sugarizer's other notorious bug (that randomly pollutes your drive with 1, 2 or 3 giant 256MB files during install of MongoDB) is arguably much worse:
https://github.com/iiab/iiab/issues/254 (what's causing 801MB /var/lib/mongodb ?)

jvonau, 2017-09-28:
#304

holta, 2017-09-28:
Let's try #304 (Install nodejs correctly) soon to see if this helps!

holta, 2017-10-02:
This issue is worse than we earlier understood &mdash; somehow #216 did not solve it (or a subsequent regression broke #216?) and but here's a new pattern we missed a month ago &mdash; not only do subsequent runs of "./runtags sugarizer" all fails, but also:

_Sugarizer fails to install on the very 1st run of "./runtags sugarizer" even if it was never installed before_.

Specifically if the small-sized local_vars.yml is used during installing, including:
```
sugarizer_install: False
sugarizer_enabled: False
```
And then changed as follow:
```
sugarizer_install: True
sugarizer_enabled: True
```
Sugarizer's install (with either "./runtags sugarizer" or "./runansible") will then fail as follows:

> TASK [sugarizer : Actually get it installed] ***********************************
> [WARNING]: Consider using apt module rather than running apt-get
>
> fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["apt-get", "install", "-y", "npm"], "delta": "0:00:01.965267", "end": "2017-10-02 08:02:27.828316", "failed": true, "msg": "non-zero return code", "rc": 100, "start": "2017-10-02 08:02:25.863049", "stderr": "E: Unable to correct problems, you have held broken packages.", "stderr_lines": ["E: Unable to correct problems, you have held broken packages."], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nSome packages could not be installed. This may mean that you have\nrequested an impossible situation or if you are using the unstable\ndistribution that some required packages have not yet been created\nor been moved out of Incoming.\nThe following information may help to resolve the situation:\n\nThe following packages have unmet dependencies:\n npm : Depends: node-gyp (>= 0.10.9) but it is not going to be installed", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "Some packages could not be installed. This may mean that you have", "requested an impossible situation or if you are using the unstable", "distribution that some required packages have not yet been created", "or been moved out of Incoming.", "The following information may help to resolve the situation:", "", "The following packages have unmet dependencies:", " npm : Depends: node-gyp (>= 0.10.9) but it is not going to be installed"]}

Also, all subsequent attempts to install Sugarizer will fail in the same way :/  This has been confirmed on 2 different RPi3's running IIAB 6.4 RC6+ on Raspbian Lite from 2017-09-07.

Recap: the only (known) way to get Sugarizer to install, is to use "./runansible" during its very first run (e.g. clean-installing on a new OS) with the following in local_vars.yml:
```
sugarizer_install: True
sugarizer_enabled: True
```

holta, 2017-10-02:
PS this is quite serious, but we can defer to [IIAB 6.5](https://github.com/iiab/iiab/milestone/2) if it's too hard to solve this week.

jvonau, 2017-10-02:
notes in Ref #240

holta, 2017-10-02:
Clarification on the wider pattern &mdash; "./runtags sugarizer" seems to always fail (even on very different machines) with the following in local_vars.yml:
```
sugarizer_install: True
sugarizer_enabled: False
```
Thankfully "./runtags sugarizer" does complete with the following: (gratuitous?)
```
sugarizer_install: False
sugarizer_enabled: False
```
_Finally, note that Sugarizer's basic functionality (http://box/sugarizer) works despite all the above!  REGARDLESS if the original "./runansible" completed with the following:
```
sugarizer_install: True
sugarizer_enabled: True
```

holta, 2017-10-12:
Just to reconfirm this happened on Raspbian Lite 2017-09-07 again today:

2017-10-11 17:00:18,623 p=16180 u=root |  fatal: [127.0.0.1]: FAILED! => {"chan\
ged": true, "cmd": ["apt-get", "install", "-y", "npm"], "delta": "0:00:02.16534\
5", "end": "2017-10-11 17:00:15.255348", "failed": true, "msg": "non-zero retur\
n code", "rc": 100, "start": "2017-10-11 17:00:13.090003", "stderr": "E: Unable\
 to correct problems, you have held broken packages.", "stderr_lines": ["E: Una\
ble to correct problems, you have held broken packages."], "stdout": "Reading p\
ackage lists...\nBuilding dependency tree...\nReading state information...\nSom\
e packages could not be installed. This may mean that you have\nrequested an im\
possible situation or if you are using the unstable\ndistribution that some req\
uired packages have not yet been created\nor been moved out of Incoming.\nThe f\
ollowing information may help to resolve the situation:\n\nThe following packag\
es have unmet dependencies:\n npm : Depends: node-gyp (>= 0.10.9) but it is not\
 going to be installed", "stdout_lines": ["Reading package lists...", "Building\
 dependency tree...", "Reading state information...", "Some packages could not \
be installed. This may mean that you have", "requested an impossible situation \
or if you are using the unstable", "distribution that some required packages ha\
ve not yet been created", "or been moved out of Incoming.", "The following info\
rmation may help to resolve the situation:", "", "The following packages have u\
nmet dependencies:", " npm : Depends: node-gyp (>= 0.10.9) but it is not going \
to be installed"]}

Hard to tell if this occurs on every 5th install, or every 10th install, but it's in that range :-(

holta, 2017-10-12:
Correction/Clarification: the above Sugarizer install is repeatable, apparently every time one executes ./runansible.

This occured upon changing local_vars.yml to medium-sized, after a successful install of min-sized local_vars.yml originally using http://download.iiab.io/6.5/rpi/load-min.txt

holta, 2017-10-12:
This is more serious than I originally understood: it is not just ".runtags sugarizer" that fails, but all subsequent runs of "./ansible" yikes :/

So the verbiage that many of us agreed to place in [local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml) ~10 days ago (http://minutes.iiab.io) might not be sufficient in the end?
```
# sugarizer_enabled is currently IGNORED as basic Sugarizer works w/o Journal!
# https://github.com/iiab/iiab/issues/193 Subsequent "./runtags sugarizer" fail
# https://github.com/iiab/iiab/issues/240 Sugarizer 0.8 to 0.9 ongoing issues
```

holta, 2017-10-13:
@jvonau is the above sub-issue (Sugarizer causing "./runansible" to fail every time, after upgrading a machine from MIN-sized to MED-sized local_vars.yml) likely fixed by your PR #304 "Install nodejs correctly" ?

Even if the larger #193 issue if not full solved possibly?

holta, 2017-10-13:
As #193 here has grown into an umbrella ticket to keep Sugarizer issues organized, here's a list of the 14  Sugarizer tickets, just so important ideas don't get lost in future: (to solve and/or mitigate!)

https://github.com/XSCE/xsce/issues/879 "on power failure mongodb is corrupt"
#193 Ansible fails all subsequent runs on Raspbian Stretch: Sugarizer "npm : Depends: node-gyp (>= 0.10.9)"
PR #216 Npmreinstall
#240 Sugarizer 0.8 to 0.9 <<< also a good summary since Sept 11, 2017
PR #245 Sugar (no changes)
PR #246 Sugar (bump 0.8 to 0.9)
#254 what's causing 801MB /var/lib/mongodb ? <<< also a good summary since Sept 16, 2017
PR #260 MongoDB: 2 improvements to reduce DB bloat
PR #262 try to shrink mongodb
PR #263 Mongo conf
PR #299 smallfiles=true added to mongod.conf
PR #304 Install nodejs correctly
#337 3 red warnings when running "systemctl" (and on boot too?)
PR #359 disable systemd sugarizer.service

Thanks to @georgejhunt @jvonau and @arky who've made this all happen, allowing Sugarizer to work in IIAB 6.4 and 6.5/master as the OLPC/Sugar community greatly desires (thankfully single-user mode is fully sufficient for now...and hopepully Sugarizer will stop [blocking Ansible](https://github.com/iiab/iiab/issues/193#issuecomment-335990225) soon...)

holta, 2017-10-14:
Resolving for now.  @jvonau feel free to re-open as appropriate.

closed 2017-10-14 by holta