iiab/iiab #195: Upgrade usbmount 0.0.14 to 0.0.22...for Debian & Raspbian especially (others too?)

holta, 2017-09-02:
@tim-moody spoke in favor of upgrading to usbmount 0.0.22 during Aug 31 call.

Related: "drop download of usbmount for debian-9" PR #180 appears to have been reverted.

So hopefully @georgejhunt can clarify as someone (not sure who, no worries!) wrote "in actual fact usbmount does not exist in default stretch sources" into 2017-9-4 call agenda/minutes @ http://minutes.iiab.io

georgejhunt, 2017-09-02:
it failed the smoke test on a rpi stretch

georgejhunt, 2017-09-02:
No, I get lost a lot, because I'm doing too much in parallel.
 0.0.22 usbmount fails to install on a  debian 9 VM, but installs on raspbian-9.

root@box:~# cat /etc/*release
PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
NAME="Debian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
root@box:~# apt-get install usbmount
Reading package lists... Done
Building dependency tree       
Reading state information... Done
usbmount is already the newest version (0.0.14.1).
root@box:~#

holta, 2017-09-02:
@tim-moody or anyone: please help finalize which usbmount should be installed on which OS!

Close this issue if usbmount is working on at least a couple OS's?

Per FAQ.IIAB.IO's ["Can teachers display their own content?"](http://wiki.laptop.org/go/IIAB/FAQ#Can_teachers_display_their_own_content.3F)

tim-moody, 2017-09-02:
pretty sure it installed on x64 debian 9.1

georgejhunt, 2017-09-03:
I think we're in good shape on this one.  Our playbook installs 0.0.14 from
our download site, and then if 0.0.22 is available in the repo, it is
upgraded.

Or at least that is what I assume happened, since the version on the pi3 is
0.0.22

jvonau, 2017-09-03:
Think the raspbian-9 would be using the original is_debuntu path and not the is_debian_9 path gets .22 out of the box while debian-9 gets .14 with no upgrade? Now usbmount is not upgraded in the next section because we use present in that stanza. Check if there is an update available on the debian-9 machine if there is just remove the exception in the code. If there is no update available grab the .22 deb from the cache of raspbian-9 and upload that to the DL location for debian-9 to use?

holta, 2017-09-13:
Ref #165 (Should vars/raspbian-9.yml have "is_debian_9: True" ?)

jvonau, 2017-09-14:
Curt Thompson 
> 4. USB sticks are no longer auto-mounted.
Naturally this means that Teacher Content is also no longer auto-mounted to http://box/usb (the USB stick I tested is FAT32-formatted with Haitian content placed into directory "usb", which certainly used to work in the past!)

Might need the same udev mod that CentOS & Fedora require now that a later version of systemd is involved. Would be nice to see if #146 can or can't work. but the pieces are already there to do an upload from a browser, for use with olcp's micro-format based updater so there is some merit poking at it to see if it's dead.

holta, 2017-09-19:
@georgejhunt is interested in helping to solve this.

So I'm pasted in below what @tim-moody clarified to me over IM (live chat) on 2017-09-13:

> on my demo machine, debian 9.1 with usbmount 0.0.14 a usb stick does not automount

> actually maybe it is trying to mount
```
Sep 13 19:59:22 [actual FQDN] kernel: usb 3-1: new SuperSpeed USB device number 4 using xhci_hcd
Sep 13 19:59:22 [actual FQDN] kernel: usb 3-1: New USB device found, idVendor=0951, idProduct=1666
Sep 13 19:59:22 [actual FQDN] kernel: usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 13 19:59:22 [actual FQDN] kernel: usb 3-1: Product: DataTraveler 3.0
Sep 13 19:59:22 [actual FQDN] kernel: usb 3-1: Manufacturer: Kingston
Sep 13 19:59:22 [actual FQDN] kernel: usb 3-1: SerialNumber: 60A44C413C91FE30AB430B29
Sep 13 19:59:27 [actual FQDN] kernel: usb 3-1: Set SEL for device-initiated U2 failed.
Sep 13 19:59:27 [actual FQDN] kernel: usb-storage 3-1:1.0: USB Mass Storage device detected
Sep 13 19:59:27 [actual FQDN] kernel: scsi host3: usb-storage 3-1:1.0
Sep 13 19:59:27 [actual FQDN] kernel: usb 3-1: USB disconnect, device number 4
Sep 13 19:59:28 [actual FQDN] kernel: usb 3-1: new SuperSpeed USB device number 5 using xhci_hcd
Sep 13 19:59:28 [actual FQDN] kernel: usb 3-1: New USB device found, idVendor=0951, idProduct=1666
Sep 13 19:59:28 [actual FQDN] kernel: usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 13 19:59:28 [actual FQDN] kernel: usb 3-1: Product: DataTraveler 3.0
Sep 13 19:59:28 [actual FQDN] kernel: usb 3-1: Manufacturer: Kingston
Sep 13 19:59:28 [actual FQDN] kernel: usb 3-1: SerialNumber: 60A44C413C91FE30AB430B29
Sep 13 19:59:33 [actual FQDN] kernel: usb 3-1: Set SEL for device-initiated U2 failed.
Sep 13 19:59:33 [actual FQDN] kernel: usb-storage 3-1:1.0: USB Mass Storage device detected
Sep 13 19:59:33 [actual FQDN] kernel: scsi host3: usb-storage 3-1:1.0
Sep 13 19:59:33 [actual FQDN] kernel: usb 3-1: USB disconnect, device number 5
Sep 13 19:59:33 [actual FQDN] kernel: usb 3-1: new SuperSpeed USB device number 6 using xhci_hcd
Sep 13 19:59:33 [actual FQDN] kernel: usb 3-1: New USB device found, idVendor=0951, idProduct=1666
ETC...
```
> it does automount on Ubuntu 16.04 with usbmount 0.0.22

holta, 2017-09-19:
See #268 ("cheap hack to ensure the latest usbmount in installed...")

jvonau, 2017-09-19:
Be interesting if activity server was installed and a usb key with a directory named 'xs-activity-server' is present and inserted. The logging should show the key get mounted then unmounted(cause there is nothing to do) but that would point to using the same udev workaround that is presently used with CentOS.

georgejhunt, 2017-09-19:
Some progress:

1. Both Jessie and stretch load usbmount 0.0.22
2. Functions ok in jessie, fails in stretch
3. In stretch, when I put a "touch /root/fromusbmount" in /etc/usbmount/mount.d/70-usb-library, /root is untouched, until I restart udev (udev is started in initrd, and seems to be trying to write to a read-only disk)
4. The 70-usb-library is running as root
5. During its execution, the usb stick is mounted and readable (so I believe usbmount is functioning correctly).
6. But after usbmount script has completed, the /media/usb0 mounted stick disappears. (survives in jessie)

### data on udev in jessie:
Multi-Arch: foreign
Source: systemd
Version: 215-17+deb8u7
Depends: libacl1 (>= 2.2.51-8), libblkid1 (>= 2.19.1), libc6 (>= 2.17), libkmod2 (>= 5~), libselinux1 (>= 2.0.65), libudev1 (= 215-17+deb8u7), adduser, lsb-base (>= 3.0-6), util-linux (>= 2.16), procps
Pre-Depends: debconf (>= 1.4.69) | debconf-2.0
Breaks: consolekit (<< 0.4.6-1), kmod (<< 14), plymouth (<< 0.9.0-7), systemd (<
### data on udev in stretch:
Architecture: armhf
Multi-Arch: foreign
Source: systemd
Version: 232-25+deb9u1
Replaces: bash-completion (<< 1:2.1), systemd (<< 232-20)
Depends: libacl1 (>= 2.2.51-8), libblkid1 (>= 2.19.1), libc6 (>= 2.17), libgcc1 (>= 1:3.5), libkmod2 (>= 5~), libselinux1 (>= 2.1.9), adduser, libudev1 (= 232-25+deb9u1), lsb-base (>= 3.0-6), util-linux (>= 2.27.1), procps
Pre-Depends: dpkg (>= 1.17.14)
Breaks: bash-completion (<< 1:2.1), consolekit (<< 0.4.6-1), ifplugd (<< 0.28-19.1~), ifupdown (<< 0.8.5~), joystick (<< 1:1.4.9-1~), kmod (<< 14), plymouth (<< 0.9.0-7), systemd (<< 232-20), usb-modeswitch (<< 2.3.0+repack0-1~)

So I think we should file a bug, and wait for upstream to fix it.

jvonau, 2017-09-20:
Have a look at udev_needs_patch in roles/1-prep/tasks/main.yml

georgejhunt, 2017-09-20:
Bingo Jerry, you fixed it

holta, 2017-09-25:
@georgejhunt & I have both tested this on Raspbian Stretch Lite!

holta, 2017-10-25:
@jvonau & All can I resolve this now?

jvonau, 2017-11-12:
close this

closed 2017-11-12 by holta