iiab/iiab #197: Permit installation of IIAB over WiFi instead of Ethernet

holta, 2017-09-02:
Ref #194 (Pi wifi gw) outlines a future possibility that might perhaps allow people to build an IIAB system even when they lack access to an Ethernet cable providing Internet.

Instead of using Ethernet, they would download and install IIAB software over WiFi (even though WiFi connections/authentication/time-limits/bandwidth-limits mean this will never be quite as reliable as Ethernet, for so many reasons!)

jvonau, 2017-09-03:
addressed with #178 except for configuration of the actual connection.

jvonau, 2017-09-09:
[headless wifi boot](https://www.raspberrypi.org/forums/viewtopic.php?t=191252)
Re: Raspbian Stretch Headless Setup Procedure
Thu Sep 07, 2017 9:06 pm

holta, 2017-10-16:
@tim-moody suggested "hotspot" functionality is possible at very the same time as you're downloading IIAB+ZIMs/etc to create an RPi3 server, e.g. for Peace Corps Volunteers to begin testing immediately, without the need to toggle from "online provisioning/wifi mode" to "offline hotspot mode".

@jvonau suggested during our call we first need to find out if RPi3's firmware supports that ?

While this would be amazing: I'm also nervous this could become a year-long project that might/could distract from the install-over-WiFi MVP (minimum viable product) done we're seeking to prove as part of [IIAB 6.5](https://github.com/iiab/iiab/milestone/2).  Trying to enable simple AP mode on XO-1.5 became an almost year-long project earlier, when we valiantly tried that in 2015-2016 (also for very good reasons, but still today our R&D budget remains limited ;)

Either way - schools/clinics in the developing world essentially all tell us they need a toggle to turn OFF upstream Internet as - so we'll need something like this at the command-line (or Admin Console later?) that rapidly changes between these 2 modes.  Reasons Being that developing world schools/clinics:
1. cannot possibly afford the costs of sharing their upstream connectivity to their IIAB users
2. sometimes Internet like Television is inflicted thoughtlessly on young children as a video pacifier/ digital babysitting, chock full of pointless-or-worse advertising (with nary a thought to nurturing childrens' own best interests!!)
3. school/clinics/etc in the developing world simply cannot afford the higher security/upkeep implications associated with "online" devices (as humble institutions trying to do their best, under extraordinarily limited means...)

jvonau, 2017-10-16:
Lets shoot for installing over wifi, which should now work with 6.4 as we don't try to overwrite /etc/network/interfaces any longer but is untested/unconfirmed, may need a bit of tweaking going forward in master. I recall @georgejhunt mentioned that he *was* able to achieve AP+wifi_client using his bash scripting on a rpi, but I can't see that landing in time for 6.5, although I think the ground work for trying to write a non-hostapd wifi config file can be introduced in 6.5 with that in mind to ease the transition to the different layout in the near future. Part of the shift would be the need to omit hostapd startup, maybe a rethink on always using a bridge, and include iptables manipulation outside of ansible while sharing a common source. I have started to look at removing /etc/sysconfig/iiab* files, becoming part of /etc/iiab/iiab.env as a step towards making that easier. I'll put up some RFC PR for a visual look and some feedback. Timing of the setting of the machine's hostname/domainname during the install and the ability to resolve is one subject that would come up for debate.

holta, 2017-10-17:
Is it Ok that "apt update" tries to upgrade these 2 packages, after IIAB is installed on Raspbian?

hostapd/stable 2:2.4-1+deb9u1 armhf [upgradable from: 2:2.4-1]
wpasupplicant/stable 2:2.4-1+deb9u1 armhf [upgradable from: 2:2.4-1]

jvonau, 2017-10-17:
Hopefully no surprises come with the updates. Recently released I would
figure, brand new installs should get those newer packages installed by
default going forward.

holta, 2017-10-19:
@jvonau suggests this might work today (!) if someone can test this as follows:
1. install Raspbian With Desktop
2. connect to their home's WiFi, entering SSID/password
3. ./runansible etc per an install/load script similar to: http://download.iiab.io/6.5/rpi/

jvonau, 2017-11-12:
basic install over wifi works

holta, 2017-12-04:
I tried to install the latest IIAB 6.5 / master onto an RPi3 using a ~1mbit/sec [*] WiFi connection.

The result is that many hours later (after many restarts of ./iiab-install) all network connectivity was lost, right here exactly:
```
TASK [network : Restart dhcpcd] ************************************************
```
A reboot does not help regrettably ("ping 8.8.8.8" yield "connect: Network is unreachable") even while /etc/wpa_supplicant/wpa_supplicant.conf remains valid (yet does not connect to the home's WiFi anymore, sadly).  So this install is sadly at a standstill.

FWIW after Stage 9 completed, the earlier "[network : Restart dhcpcd]" froze in its tracks, never returning to the prompt (RPi3 has no Internet connectivity since that ssh session froze) with the "Restart dhcpd" message above.

_An attached screen/keyboard is the only way to diagnose at this point: whaI should I try?_

[*] ASIDE: downloads of KA Lite, Kiwix, Sugarizer (etc) earlier all failed due to timeouts, so I temporarily changed Ansible's get_url parameter to "timeout: 1000" (about 17 minutes) in as many of these playbooks as possible.  Which solved most of these Ansible failures (but not KA Lite's, so for now I've set kalite_install: False within /opt/iiab/iiab/vars/local_vars.yml).

holta, 2017-12-04:
Good News: ./iiab-install on the above RPi3 did in fact finally complete, as confirmed by /opt/iiab/iiab/iiab-install.log &mdash; but this had been obscured by the ssh connection dropping during/after "[network : Restart dhcpcd]" for whatever reason.

Bad News: for the moment it is necessary to type "iiab-hotspot-off" and then reboot, to restore your RPi3's WiFi connection to the Internet, in order to complete the install etc.  This default accidentally got reversed.

SEE "reboot_to_AP: False" in https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L72-L75 which is accidentally being ignored for the moment.

@jvonau figured this out: which also explains why my ssh connection into the RPi3 was dropped right after Stage 9, which would have caused WiFi-based implementer folk to be left confused (as I was very much confused, as to why my ssh connection suddenly ended with no explanation, and was not clearly recoverable even after multiple reboots...until Jerry explained!)

holta, 2017-12-23:
Let's try our best to get #616 "WiFi-installed IIAB does not connect using an Ethernet cable" working for IIAB 6.5's final release, on RPi at minimum (and possibly others, depending on community verdict?!)

holta, 2018-01-12:
Amazingly after 3 months work, this now seem increasingly solid!  And almost ready for IIAB 6.5's release, using the commands {iiab-hotspot-on, iiab-hotspot-off} and a big thank you to all who developed the current "promiscuous mode" workaround for Raspbian's WiFi bug (10SEC disease) at #638 etc.

i.e. reinforced by PRs #649 #651 #653 and many others!

(So we cab re-open this umbrella PR later as nec...e.g. to track simultaneous upstream/downstream WiFi if PRs #430, #431 & #543 evolve?  But am closing this tkt for now, as we've finally delivered the core of what we promised for [IIAB 6.5](https://github.com/iiab/iiab/wiki/IIAB-6.5-Release-Notes))

closed 2018-01-12 by holta