iiab/iiab #198: rpi ex virgine

tim-moody, 2017-09-02:
raspbian-9-lite of 8/16:

•	Added wipi and ethernet dongles
•	root@raspberrypi:~# **ip a**
•	1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
•	    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
•	    inet 127.0.0.1/8 scope host lo
•	       valid_lft forever preferred_lft forever
•	    inet6 ::1/128 scope host
•	       valid_lft forever preferred_lft forever
•	2: enxb827eb45ad20: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
•	    link/ether b8:27:eb:45:ad:20 brd ff:ff:ff:ff:ff:ff
•	    inet 192.168.0.38/24 brd 192.168.0.255 scope global enxb827eb45ad20
•	       valid_lft forever preferred_lft forever
•	    inet6 fe80::2ec2:b48f:8d06:57b5/64 scope link
•	       valid_lft forever preferred_lft forever
•	3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
•	    link/ether b8:27:eb:10:f8:75 brd ff:ff:ff:ff:ff:ff
•	4: enx3c8cf8f64bfd: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
•	    link/ether 3c:8c:f8:f6:4b:fd brd ff:ff:ff:ff:ff:ff
•	5: wlx40a5ef055c87: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
•	    link/ether 40:a5:ef:05:5c:87 brd ff:ff:ff:ff:ff:ff
•	root@raspberrypi:~#** cat /etc/network/interfaces**
•	# interfaces(5) file used by ifup(8) and ifdown(8)
•	
•	# Please note that this file is written to be used with dhcpcd
•	# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'
•	
•	# Include files from /etc/network/interfaces.d:
•	source-directory /etc/network/interfaces.d
•	cat /etc/dhcpcd.conf
•	# A sample configuration for dhcpcd.
•	# See dhcpcd.conf(5) for details.
•	
•	# Allow users of this group to interact with dhcpcd via the control socket.
•	#controlgroup wheel
•	
•	# Inform the DHCP server of our hostname for DDNS.
•	hostname
•	
•	# Use the hardware address of the interface for the Client ID.
•	clientid
•	# or
•	# Use the same DUID + IAID as set in DHCPv6 for DHCPv4 ClientID as per RFC4361.
•	# Some non-RFC compliant DHCP servers do not reply with this set.
•	# In this case, comment out duid and enable clientid above.
•	#duid
•	
•	# Persist interface configuration when dhcpcd exits.
•	persistent
•	
•	# Rapid commit support.
•	# Safe to enable by default because it requires the equivalent option set
•	# on the server to actually work.
•	option rapid_commit
•	
•	# A list of options to request from the DHCP server.
•	option domain_name_servers, domain_name, domain_search, host_name
•	option classless_static_routes
•	# Most distributions have NTP support.
•	option ntp_servers
•	# Respect the network MTU. This is applied to DHCP routes.
•	option interface_mtu
•	
•	# A ServerID is required by RFC2131.
•	require dhcp_server_identifier
•	
•	# Generate Stable Private IPv6 Addresses instead of hardware based ones
•	slaac private
•	
•	# Example static IP configuration:
•	#interface eth0
•	#static ip_address=192.168.0.10/24
•	#static routers=192.168.0.1
•	#static domain_name_servers=192.168.0.1 8.8.8.8
•	
•	# It is possible to fall back to a static IP if DHCP fails:
•	# define static profile
•	#profile static_eth0
•	#static ip_address=192.168.1.23/24
•	#static routers=192.168.1.1
•	#static domain_name_servers=192.168.1.1
•	
•	# fallback to static profile on eth0
•	#interface eth0
•	#fallback static_eth0
•	**cat /etc/resolv.conf**
•	# Generated by resolvconf
•	nameserver 192.168.0.1
•	systemctl status networking
•	● networking.service - Raise network interfaces
•	   Loaded: loaded (/lib/systemd/system/networking.service; enabled; vendor preset: enabled)
•	   Active: active (exited) since Wed 2017-08-16 01:22:13 UTC; 2 weeks 3 days ago
•	     Docs: man:interfaces(5)
•	 Main PID: 288 (code=exited, status=0/SUCCESS)
•	   CGroup: /system.slice/networking.service
•	
•	Aug 16 01:22:12 raspberrypi systemd[1]: Starting Raise network interfaces...
•	Aug 16 01:22:13 raspberrypi systemd[1]: Started Raise network interfaces.
•	systemctl status dhcpcd
•	● dhcpcd.service - dhcpcd on all interfaces
•	   Loaded: loaded (/lib/systemd/system/dhcpcd.service; enabled; vendor preset: enabled)
•	  Drop-In: /etc/systemd/system/dhcpcd.service.d
•	           └─wait.conf
•	   Active: active (running) since Wed 2017-08-16 01:23:01 UTC; 2 weeks 3 days ago
•	 Main PID: 545 (dhcpcd)
•	   CGroup: /system.slice/dhcpcd.service
•	           ├─444 wpa_supplicant -B -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0
•	           ├─545 /sbin/dhcpcd -q -w
•	           └─773 wpa_supplicant -B -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlx40a5ef055c87
•	
•	Sep 02 19:43:07 raspberrypi dhcpcd[545]: enx3c8cf8f64bfd: waiting for carrier
•	Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlan1: if_up: No such device
•	Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlan1: waiting for carrier
•	Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlan1: removing interface
•	Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlx40a5ef055c87: waiting for carrier
•	Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlx40a5ef055c87: carrier acquired
•	Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlx40a5ef055c87: adding address fe80::a222:a53b:bd85:e029
•	Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlx40a5ef055c87: IAID ef:05:5c:87
•	Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlx40a5ef055c87: carrier lost
•	Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlx40a5ef055c87: deleting address fe80::a222:a53b:bd85:e029
•	Unit dhclient.service could not be found.
•	ls: cannot access '/etc/sysconfig/network': No such file or directory
•	ls: cannot access '/etc/sysconfig/network-scripts/': No such file or directory
•

jvonau, 2017-09-02:
Thanks confirms that resolvconf is being used and should not be removed, and that the wifi is being handled by dhcpcd which is controlling wpa_supplican on the rpi.

jvonau, 2017-09-02:
> Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlan1: if_up: No such device
Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlan1: waiting for carrier
Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlan1: removing interface
Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlx40a5ef055c87: waiting for carrier
Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlx40a5ef055c87: carrier acquired
Sep 02 19:44:14 raspberrypi dhcpcd[545]: wlx40a5ef055c87: adding address fe80::a222:a53b:bd85:e02

wlan1 is the same device as the 'wlx..new name' think the next change on the virgin test would be to add the net.ifnames=0 to /boot/cmdline.txt just to see what the change are to enx....

tim-moody, 2017-09-02:
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:45:ad:20 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.38/24 brd 192.168.0.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::2ec2:b48f:8d06:57b5/64 scope link
       valid_lft forever preferred_lft forever
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:10:f8:75 brd ff:ff:ff:ff:ff:ff
4: wlan1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 40:a5:ef:05:5c:87 brd ff:ff:ff:ff:ff:ff
5: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 3c:8c:f8:f6:4b:fd brd ff:ff:ff:ff:ff:ff
```

tim-moody, 2017-09-02:
new names do not persist if net.ifnames=0 removed from cmdline.txt

tim-moody, 2017-09-02:
removed ethernet cable from builtin adapter and attached to usb dongle

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: enxb827eb45ad20: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:45:ad:20 brd ff:ff:ff:ff:ff:ff
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:10:f8:75 brd ff:ff:ff:ff:ff:ff
4: wlx40a5ef055c87: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 40:a5:ef:05:5c:87 brd ff:ff:ff:ff:ff:ff
5: enx3c8cf8f64bfd: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 3c:8c:f8:f6:4b:fd brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.41/24 brd 192.168.0.255 scope global enx3c8cf8f64bfd
       valid_lft forever preferred_lft forever
    inet6 fe80::9183:26fc:ec05:1026/64 scope link
       valid_lft forever preferred_lft forever
```

tim-moody, 2017-09-02:
looks to me like dhcpcd does a good job and we should let it be except for static addresses, which should also go in dhcpcd.conf

jvonau, 2017-09-02:
Yea I was going back and forth between iiab and dhcpcd.conf for static addressing, leaning to keeping iiab. JK showed /etc/network/interfaces worked when showing dhcp/dns issues as the place for static, just moving that to iiab as we are the only ones who are putting that file in place and adjusting dhcpcd.conf to not react on interfaces in #178

tim-moody, 2017-09-03:
added 
    /etc/wpa_supplicant/wpa_supplicant.conf below and rebooted

(no change to interfaces)
```
root@raspberrypi:~# cat /etc/wpa_supplicant/wpa_supplicant.conf
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
network={
    proto=RSN
    key_mgmt=WPA-PSK
    pairwise=TKIP
    auth_alg=OPEN
    ssid="SunRoom"
    psk="xxx"
}
```
ip a
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: enxb827eb45ad20: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:45:ad:20 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:10:f8:75 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.37/24 brd 192.168.0.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::cac1:e06d:fcf6:5aea/64 scope link
       valid_lft forever preferred_lft forever
4: enx3c8cf8f64bfd: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 3c:8c:f8:f6:4b:fd brd ff:ff:ff:ff:ff:ff
5: wlx40a5ef055c87: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 40:a5:ef:05:5c:87 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.28/24 brd 192.168.0.255 scope global wlx40a5ef055c87
       valid_lft forever preferred_lft forever
    inet6 fe80::b355:ddc0:557b:f0be/64 scope link
       valid_lft forever preferred_lft forever

```
both wifi adapters connected!

jvonau, 2017-09-03:
where interface is the default route pointing at?  ip route

tim-moody, 2017-09-03:
```
root@raspberrypi:~# ip route
default via 192.168.0.1 dev wlan0 src 192.168.0.37 metric 304
default via 192.168.0.1 dev wlx40a5ef055c87 src 192.168.0.28 metric 305
192.168.0.0/24 dev wlan0 proto kernel scope link src 192.168.0.37 metric 304
192.168.0.0/24 dev wlx40a5ef055c87 proto kernel scope link src 192.168.0.28 metric 305
```

jvonau, 2017-09-04:
Think the **ansible_default_ipv4** facts section of the iiab.log file would be needed to be seen with the 2 wifi gateway. Think we should be ok and the first one above would be selected for that detection as the metric is lower so traffic should use that interface first, which one to make the AP would be the question.

jvonau, 2017-09-04:
mode=2 enables AP mode if compiled in according to https://askubuntu.com/questions/845211/ap-mode-with-wpa-supplicant. Looks like the call that dhcpcd uses to start wpa_supplicant might need to be altered to account for the 2 interfaces, I'm thinking different files for the interface?

jvonau, 2017-09-05:
Whatever you choose to solve your issue, this should be the test that all installations should pass. Funny that I have to jump a bar higher than what is in the codebase, I didn't do the port to debian, fix your regressions. Good luck

jvonau, 2017-09-06:
> looks to me like dhcpcd does a good job and we should let it be except for static addresses, which should also go in dhcpcd.conf

Is that all static address or just the external WAN? Sure that is possible, but there would be need to move the creation of br0 over to systemd's native method first.

> Yea I was going back and forth between iiab and dhcpcd.conf for static addressing, leaning to keeping iiab.

As networking.service is currently creating br0 and the static ipaddress to use it. Unless feedback requested is forthcoming, and a test of this mockup is preformed against current master with the iiab-install.log attached to this issue, this goes to the bottom of my priorities.

jvonau, 2017-09-08:
Interesting that there is working client wifi prior to running ansible, can you confirm what version of the wifi firmware is install with apt? #230 Wireless firmware for Pi 3 and Pi 0W modified to address Broadpwn exploit mailing list:
> because the firmware did not work, on the august release.

July 2017 thread https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=187725 We need to know if the .txt file mentioned is present out of the box in either 08-16 or 09-07 images.  [related backstory](https://github.com/RPi-Distro/repo/issues/51)

georgejhunt, 2017-09-08:
I can not, since all the images in my possession have been changed. The
firmware filename was unchanged, so someone thought Jessie and Stretch were
identical.

The lack of function was in AP mode, and I did not test client
wpa_supplicant mode.

jvonau, 2017-09-08:
Great, no clue if the version that is being shipped is fixed or not

holta, 2017-09-08:
There are several ways to get/examine the firmware from prior to 2017-08-16 if that becomes necessary e.g. here is one way:
http://download.iiab.io/6.4/rpi/iiab-6.4-170721-lite-min-g4da87a9.img.zip
http://download.iiab.io/6.4/rpi/README2.html

Aside: I'd offer to help diagnose (whatever needs to be diagnosed here?) if I wasn't swamped with travel/prep this wkd for Open Source Summit (formerly LinuxCon) in LA.

jvonau, 2017-09-08:
The fix was in 2017-08-16 so 170721 would be jessie and have the older firmware I would assume. I've asked for a revert of #160 to get back to the stock out of the box version.

jvonau, 2017-09-09:
https://wiki.archlinux.org/index.php/dhcpcd#10-wpa_supplicant 
Think we would need to use /etc/wpa_supplicant/wpa_supplicant-'interface'.conf to get AP=2 to work without editing  /etc/wpa_supplicant/wpa_supplicant.conf as [raspberrypi-net-mods.service](https://github.com/RPi-Distro/raspberrypi-net-mods/blob/master/debian/raspberrypi-net-mods.service) is running and might clobber /etc/wpa_supplicant/wpa_supplicant.conf.

holta, 2017-09-25:
@tim-moody writes "I suppose 198 has outlived its usefulness" but let's re-open for [IIAB 6.5](https://github.com/iiab/iiab/milestone/2) as nec~

closed 2017-09-25 by holta