iiab/iiab #200: DokuWiki shows nothing at http://box/wiki (Raspbian Stretch)

holta, 2017-09-02:
PR #136 says DokuWiki 2017-02-19b was not smoke-tested and it shows (@georgejhunt tried, but PR #184 doesn't seem to help!)

FYI the very latest installed on Raspbian Stretch Lite on RPi3 using http://download.iiab.io/6.4/rpi/load-vpn.txt shows the following, when browsing to http://box/wiki

```
Index of /wiki
[ICO]     Name     Last modified  Size  Description
[PARENTDIR] Parent Directory	 	- 	 
Apache/2.4.25 (Raspbian) Server at 192.168.0.132 Port 80
```

Please @m-anish can you help, so DokuWiki isn't deferred from [Milestone 6.4](https://github.com/iiab/iiab/milestone/1) to [Milestone 6.5](https://github.com/iiab/iiab/milestone/2) ?

jvonau, 2017-09-03:
see #208

holta, 2017-09-05:
Let's close this issue if DokuWiki is not longer in use.

Or defer to [IIAB 6.5](https://github.com/iiab/iiab/milestone/2) if time runs out?

As of yet, @m-anish (steward of the original DokuWiki on XSCE/IIAB initiative) has not yet offered help~  so let's hold his feet to the fire :-)

holta, 2017-09-13:
Am removing DokuWiki issues (this and #192) to [IIAB 6.4's milestone](https://github.com/iiab/iiab/milestone/1) if we cannot confirm an active "fieldback" community exists to keep this fully fresh.

holta, 2018-09-15:
Long ago fixed :-)

closed 2018-09-15 by holta