iiab/iiab #211: calibre-serve crashes after 10 sec on lastest master (Debian Stretch)

holta, 2017-09-04:
http://box:8080 worked in recent days, when installed onto RPi3 via http://download.iiab.io/6.4/rpi/load-vpn.txt

But now "git pull; ./runansible" shows the following in browser:

`Cannot GET / `

FYI "systemctl restart calibre-serve" temporarily brings the service back from "failed" to "active (running)", but then consistently crashed about 10 seconds later.

jvonau, 2017-09-05:
FYI this playbook is using 'package:' so this is an upstream regression. However I will note calibre's primary developer's response to a different issue https://bugs.launchpad.net/ubuntu/+source/calibre/+bug/1426171. CentOS gets the install from website treatment.

holta, 2017-09-05:
Is it possible our years-old build of calibre-server might be upgraded sometime in 2017 for Raspbian Stretch?

Even as calibre-server documentation here hasn't changed much :)

> "The calibre Content server — calibre 3.7.0 documentation"
> https://manual.calibre-ebook.com/server.html

Tangentially the creator of Calibre (Kovid Goyal) suggests there might be a reason to ["wait until debian releases a calibre 3.x package"](https://www.mobileread.com/forums/showthread.php?t=288398) ?

holta, 2017-09-05:
We may not have an ARM compilation yet (!) but this Calibre 3.0 announcement from mid-June 2017 sounds [Substantially More Optimistic](https://calibre-ebook.com/new-in/twelve):

> The biggest new feature, which was in development for almost that entire period [2014-2017] is a completely re-written calibre Content server.
> 
> The Content server allows you to wirelessly browse your calibre books on any modern phone/tablet and even read the books right in your phone browser. The books are stored in off-line storage on the phone so you do not have to keep calibre running or have an always on internet connection to read the books.
> 
> _The new Content server_
> 
> _You can start the Content server by clicking the Connect/share button in calibre. Then just open up a browser and type http://localhost:8080 in the address bar. You will see a list of calibre libraries. Click on one and browse the books. Click on a book and you will have the option to read or download it. To learn more about using the server to read on your devices, see the [User Manual](https://manual.calibre-ebook.com/server.html)._

jvonau, 2017-09-05:
Some update in the dependency chain broke the install. On a working machine use apt to check what is offered to be updated(but don't really update) to get an idea of what changed against the list at https://packages.debian.org/stretch/calibre. Or wait and try again in a while.

jvonau, 2017-09-05:
> Tangentially the creator of Calibre (Kovid Goyal) suggests there might be a reason to "wait until debian releases a calibre 3.x package" ?

The buildroot would be the issue https://github.com/kovidgoyal/build-calibre underneath it uses https://github.com/containers/build 
> acbuild can only be run on a Linux system, and has only been tested on the amd64 architecture.

That is why cross-compiling is recommend. I would look at the source .deb file for how the buildroot is handed as that should be compiling for arm. Good luck

jvonau, 2017-09-11:
Has a crash of the start-up extracted from 10.8.0.XX 
[warning.txt](https://github.com/iiab/iiab/files/1291640/warning.txt)

holta, 2017-09-13:
> calibre-serve.service.j2 Question becomes does calibre crash on non-rpi with current master.

**Open Call** for Ubuntu/Debian/CentOS implementers to please help, _for the sake a e-book readers everywhere!_

holta, 2017-09-13:
I cannot reproduce this bug anymore, on a brand new install of http://download.iiab.io/6.4/rpi/load-vpn.txt onto Raspbian Stretch Lite 2019-09-07, Good Riddance!

closed 2017-09-13 by holta