iiab/iiab #214: "elgg : change ownership" fails on master (Raspbian Stretch)

holta, 2017-09-04:
This was attempted using "git pull; ./runansible" on an existing installation from 2.5 days ago, originally using http://download.iiab.io/6.4/rpi/load-vpn.txt

```
TASK [elgg : change ownership] *************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "recurse option requires state to be 'directory'", "path": "/opt/elgg-2.3.3", "state": "absent"}
```

PS "git pull; ./runansible" worked earlier this morning on this very same RPi3, but no longer &mdash; so perhaps this error arising from PR #202 ?

holta, 2017-09-04:
Good News: if I "rm /opt/elgg /opt/elgg-1.12.16" then "./runansible" Ansible actually completes!  (And so presumably would also on a clean install?)

Bad News: the page http://box/elgg is mishapen / showing test-only, and so must be missing its CSS?

tim-moody, 2017-09-04:
there is a mysql elgg database that may need dropping along with removing the directory in order to reset the install.

holta, 2017-09-04:
Also note that Elgg login fails when attempted with Admin / changeme (as operators should use initially!)

Not that the credential is bad, but the login process erroneously redirects to:

http://schoolserver.lan/elgg/action/login

holta, 2017-09-04:
@tim-moody how do I drop the elgg database?

(Or is it easier to clean-install everything...)

tim-moody, 2017-09-04:
mysql
drop database elgg;
quit

holta, 2017-09-04:
```
MariaDB [(none)]> drop database elgg;
ERROR 1008 (HY000): Can't drop database 'elgg'; database doesn't exist
```

tim-moody, 2017-09-04:
sorry, the name is elggdb

holta, 2017-09-04:
Good News: Login redirects to http://box/elgg/action/login instead of http://schoolserver.lan/elgg/action/login

Bad News @georgejhunt might be able to help with: CSS is still missing on http://box/elgg (no colors/images, only text appears..and barely!)

Unpleasant News: Login forces redirects from http://<ANY NAME OR IP>/elgg to http://box/...

closed 2017-09-04 by holta

holta, 2017-09-04:
Ref PR #218 (thought elgg state=directory was in)