iiab/iiab #217: Elgg 2.3.3 CSS/images missing (master, Raspbian Stretch)

holta, 2017-09-04:
Pages are almost unusably ugly!

Can the CSS be repaired so images etc appear?

Background: installed using http://download.iiab.io/6.4/rpi/load-vpn.txt (master) onto Raspbian Stretch Lite on RPi3.

ASIDE: the forced redirect from http://<IP OR NAME>/elgg to http://box/elgg/action/login during Elgg login makes this unusable in most LAN/campus environments, per #214 (in case anybody's so lucky as to crack that!)

holta, 2017-09-04:
Ref PR #218 (thought elgg state=directory was in)

jvonau, 2017-09-04:
> ASIDE: the forced redirect from http:///elgg to http://box/elgg/action/login during Elgg login makes this unusable in most LAN/campus environments

ProxyPass?

holta, 2017-09-23:
To solve this CSS problem (nothing much appears at http://box/elgg) here is @georgejhunt's workaround that worked on an older install of Elgg on NUC:

1. As root, go into the command line mysql tool "mysql"
2. show databases;
3. drop database elggdb;
4. show databases;
5. quit
6. go to /opt/iiab/iiab
7. ./runtags elgg
8. test whether the CSS is fixed

Unfortunately this does not work on master ([IIAB 6.4 RC2](https://github.com/iiab/iiab/releases/) +) on RPi3!  _Any similar ideas out there?_

jvonau, 2017-09-23:
What other services are being installed at the same time? attach the local_vars to this issue. Think there might be a missing dependence that might be covered in a different play.

holta, 2017-09-23:
All tests were done with our ["medium-sized" local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml)

Aside/Recap from earlier in September: this is building off #202 (Elgg upgrade 1.12.16 LTS to 2.3.3; Nextcloud - add php's zip mbstring).

holta, 2017-09-23:
Clarif: our ["medium-sized" local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml) was of course enhanced from:
```
elgg_install: True
elgg_enabled: False
```
...to:
```
elgg_install: True
elgg_enabled: True
```
And then "cd /opt/iiab/iiab; ./runtags elgg" was run.

holta, 2017-09-23:
I'd thought #188 (Elgg 1.12.12 & 1.12.16 don't show anything in browser (500 Internal Server Error)) was solved.

But is this still relevant?

jvonau, 2017-09-23:
see my comments in #188 it's a dns issue that might be able to solved with some url rewrite trickery

[image]

holta, 2017-09-23:
Thanks @jvonau you've confirmed this is more than a CSS issue!

e.g. http://10.8.0.10/elgg and http://192.168.0.131/elgg etc appear to insist on loading all CSS/images from http://box (which never load) hence causing the ugly/unusable Elgg pages.

jvonau, 2017-09-23:
Elgg should work perfect from the LAN side as 'box' will resolve correctly, proxypass or mod-rewrite should be able to capture "some base url" that is in the browser and change that to "box" for php to use.

jvonau, 2017-09-23:
oh yea /opt/elgg s/b owned by www-data:www-data otherwise you get a 403 prohibited error

holta, 2017-09-24:
fyi 1 - @jvonau is working on a [(partial for now) fix](https://github.com/iiab/iiab/pull/305/files#diff-1460b7c549aa9ce1a2a72a38aed3cd65) that might enable Elgg across all URL's, regardless of http://box/elgg, http://192.168.0.x/elgg, http://10.8.0.y/elgg etc.

fyi 2 - NGINX may obviate the need for this in [IIAB 6.5](https://github.com/iiab/iiab/milestone/2) or [IIAB 6.6](https://github.com/iiab/iiab/milestone/2).

holta, 2017-09-25:
Ref #309 (Elgg - Fix permissions & handle upgrades)

closed 2017-09-25 by holta