iiab/iiab #221: x-large pizza+coke now arriving in Winnipeg within ~5min

holta, 2017-09-05:
Vern's Pizza is unfortunately closed for Labo(u)r Day so I hope Flying Pizza suffices — the driver is "currently en route" (2 hours after I ordered :-) and has already been tipped — and is now trying to debug the following code:

"Please leave pizza on front doorstep as nec ([Actual @jvonau Address]) as this is a gift for a close friend, Thank You!"

jvonau, 2017-09-05:
I was wondering why a pizza showed up and my son was not home... Thought he had ordered it on his way home... After a half hour so I checked my email... Thanks.

jvonau, 2017-09-05:
This whole project now feels like interacting with sugarlabs when I was paid to do so and put up with it. The pizza was a nice thought, but please remove my name from http://wiki.laptop.org/go/IIAB (SW architecture) as I did not architect what is in the current code base, and quite frankly I am ashamed of the state that the codebase is currently in.

closed 2017-09-18 by holta