iiab/iiab #224: Nginx alongside Apache for better analytics & (Captive Portal too?)

holta, 2017-09-05:
Ref #210 @georgejhunt & @tim-moody are exploring.

If this works in the end, Apache might possibly be preserved alongside.

holta, 2017-09-24:
See @georgejhunt's ongoing work here for [IIAB 6.5](https://github.com/iiab/iiab/milestone/2) or [IIAB 6.6](https://github.com/iiab/iiab/milestone/3) &mdash; which MIGHT even permit Moodle/WordPress/Elgg to work(*) cleanly campuswide:

#285 Tinynginx
#286 AWStats nginx
#287 KA Lite nginx
#288 Kiwix nginx
#289 Munin nginx
#290 Nextcloud nginx
#291 OSM nginx
#292 WordPress nginx

(*) via WAN side of IIAB server, if NGINX is essentially translating headers of most all http traffic from http://localhost/x to http://campuswide-server-name/x etc).

holta, 2019-11-01:
FYI there have been many attempts here, since this ticket was opened 26 months ago.

Here are some of the latest:
- PR #2009 "Nginx shim for local_vars_big" (by @georgejhunt)
- PR #2017 (refines the above, by @jvonau)

holta, 2020-01-04:
Time to close this soon, if NGINX is converging properly for IIAB 7.1 anticipated to be released 2020-01-30 ?

Note that Apache will supported alongside, for the indefinite future, or at least for a while to come~

holta, 2020-01-21:
TO DO: substantially clean up NGINX's README.md here:

https://github.com/iiab/iiab/tree/master/roles/nginx#transition-to-nginx

Starting with things like PR #2141's https://github.com/jvonau/iiab/blob/restarts/roles/nginx/README.md

holta, 2020-03-30:
@jvonau @georgejhunt @tim-moody 

Is it almost time to declare victory now, closing this ticket in favor of smaller NGINX-associated cleanup tickets going forward?

holta, 2020-04-06:
Not sure whether to declare victory.

So I'll move this to IIAB 7.2's milestone for now.

jvonau, 2020-09-26:
April -> Sept either it good enough for 7.2 and open new issues defining specific remaining issues.

holta, 2020-09-26:
> open new issues defining specific remaining issues.

Not a bad idea.

Also rename https://github.com/iiab/iiab/tree/master/roles/nginx#transition-to-nginx from "Transit to NGINX" to "NGINX README" (as with other roles) when the transition is more complete.

jvonau, 2021-07-20:
close

closed 2021-07-21 by holta