iiab/iiab #226: Does yr DNS disappear intermittently? (after Ansible glitches especially?)

holta, 2017-09-07:
Thanks all for please helping with ongoing research!

...including Joshua Kanani, @floydianslips & @tim-moody who will try to test install today~

More background in #182 (unable to get static ip on debian 9 bug)

tim-moody, 2017-09-07:
what local vars should we use?

holta, 2017-09-07:
> what local vars should we use?

Joshua Kanani uses:
http://wiki.laptop.org/go/IIAB/local_vars.yml

As invoked by:
http://download.iiab.io/6.4/rpi/load-vpn.txt

holta, 2017-09-07:
Testing with the more complete ~20 apps would also be useful of course:

http://wiki.laptop.org/go/IIAB/local_vars_big.yml
http://download.iiab.io/6.4/rpi/load-big.txt

jvonau, 2017-09-07:
Notes from RPi3 requires a reboot + a 2nd "./runansible" email exchange
> +++On Fri, Sep 1, 2017 at 4:39 PM, Jerry Vonau XX wrote:
    ++On Fri, Sep 1, 2017 at 3:18 PM, Adam Holt XX wrote:
     +  On Fri, Sep 1, 2017 at 4:12 PM, Jerry Vonau XX wrote:
     +The root cause needs to be addressed not papered over at the router​
     ++Absolutely but each of our home routers (might!) very well help us expose what's the pattern here?
     e.g. my many subsequent reboots after IIAB install work flawlessly (DHCP assignment over Ethernet works as it should every time) 
    ++​+The initial dhcp assignment is one issue, the other is dns failing to resolve after some point after booting is another issue but they are related. Adam reported the dns going away after some time.  I think that is related to the release renewal time, having both dhclient used by "iface {{ discovered_wan_iface }} inet dhcp" which might be doing a one-time dhcp-request to get the network up and having the dhcpcd daemon running in the background causing issues. We know that dhcpcd will not start if there is any mention of dhcp in /etc/network/interfaces PR 172, but will start if you are using static addressing, issue 182 has the details.
Adam Holt XX wrote:
"some time" ?  Certainly my DNS disappeared every time, consistently ~15min after the very first ./runansible on Raspbian Stretct does its thing and fails on schedule (all done 3 days ago).  I retained IP access to the full Internet but not DNS.`

JK:
> Did you try rebooting your PI again?   and if so, did everything go back to the "nice stage" as it were before? 
Because i just did mine and am back to my initial problems`

tim-moody, 2017-09-07:
I used whatever runansible gave me

jvonau, 2017-09-08:
#172 dhcpcd will not run if entries are in /etc/network/interfaces
> ​While poking at JK's Ubuntu machine I saw 2 both dhcpcd & dhclient running, that was with dhcp in the interfaces or br0 file. I was not sure what client would be present coming out of the box, that was the reasoning behind suppressing installation of dhcpcd.

> Turns out the command line for starting dhclient is hard coded in the /sbin/ifup binary. So I'm looking for alternative solutions at this point.
https://raspberrypi.stackexchange.com/questions/3332/alter-dhclient-to-retry-fetching-ip-address

So ifup is calling dhclient as shown in the logs for networking.service and appears to be a one-time try for an ip address from above link. We know if 'dhcp' is not present in the /etc/network/interfaces file dhcpcd is used if present and enabled and adding the interface to /etc/network/interfaces is not currently recommended for rpi. Lets stop adding unneeded entries.

holta, 2017-09-25:
Resolving as I have not seen this bug in a long while (if ever?!)

@jvonau feel free to re-open this if new evidence arises -

closed 2017-09-25 by holta