iiab/iiab #228: Internet+in+a+Box appears in /etc/iiab/config_vars.yml

holta, 2017-09-08:
Unclear where all these plus signs ("+") are arising from, as default_vars.yml and local_vars.yml and Admin Console all show "Internet in a Box" correctly.  Nevertheless many buggy plus signs are appearing as follows:

```
   xovis_chart_heading: "My+School:+Usage+Data+Visualization"
   host_ssid: "Internet+in+a+Box"
```

@floydianslips has pasted in his entire /etc/iiab/config_vars.yml here:
https://paste.fedoraproject.org/paste/zoZnLjcEC9G-3Lx7jbIWgw

Context: Josh installed master using http://download.iiab.io/6.4/rpi/local_vars.yml on Raspbian Stretch on RPi3, presumably using http://download.iiab.io/6.4/rpi/load.txt or similar.

holta, 2017-09-08:
@jvonau requests @floydianslips post /opt/iiab/iiab/iiab-install.log to...

#193 = Ansible fails all subsequent runs on Raspbian Stretch: Sugarizer "npm : Depends: node-gyp (>= 0.10.9)"

jvonau, 2017-09-08:
/etc/iiab/config_vars.yml is written by [iiab-admin-console](https://github.com/iiab/iiab-admin-console)
>  JD: funny how the SSID switches to Internet-in-a-Box after installing configured options
told to check /etc/iiab/config_vars.yml for +
JD: there they are
 and on xovis_chart_heading if that matters at all

I asked JD to check the gui log to see if the run completed and it appears to fail as in #193 but maybe for a different reason so use that issue to track the resolution.

holta, 2017-09-08:
Hey @floydianslips 

#193 is now resolved thanks to @jvonau's help, so please post here in #228 instead!

holta, 2017-09-08:
Ref iiab/iiab-admin-console/issues/24

holta, 2017-09-09:
@floydianslips can u confirm this is solved with https://github.com/iiab/iiab-admin-console/pull/25 ?

closed 2017-09-09 by holta