iiab/iiab #229: /etc/iiab/uuid is never created if SD's 1st boot is in RPi Zero W

holta, 2017-09-08:
FYI this works on RPi3, but not on RPi Zero W.

Context: I (and others in Haiti etc) follow George's routine to "rm /etc/iiab/uuid /etc/iiab/handle" on the source image, and then use "dd" to clone IIAB to many other microSD's.

A /etc/iiab/uuid is then automatically created(*) during the first boot of each clone IF the 1st boot occurs in an RPi3.

However there appears to be a timing-or-similar bug when 1st boot occurs in an RPi Zero W machine, such that /etc/iiab/uuid is never created.

jvonau, 2017-09-08:
From what I can see a pass through 1-prep is required to recreate the uuid . I'd double check the behaviour on a freshly cloned rpi3 it should behave the same if runansible or the console has not been used yet. Fix for in the field  `./runtags prep`

holta, 2017-09-16:
@georgejhunt reminds me that xscenet@openvpn will most likely fail when the file /etc/iiab/uuid is absent.

jvonau, 2017-09-19:
#271

holta, 2017-11-23:
@georgejhunt, can you explain what auto-creates /etc/iiab/uuid during RPi3 boot, when none exists after dd / cloning?

Is [roles/openvpn/templates/announcer](https://github.com/iiab/iiab/blob/master/roles/openvpn/templates/announcer) somehow involved?

(As compared to the proposed approach @ PR #271 for example!)

georgejhunt, 2017-11-25:
At one earlier time, uuid was created when admin-console  gathered facts.
But Tim did not like using the fact gathering script for the purpose of
making sure that uuid existed. And at this point we do not have another
start up script that we can all agree upon. A simple place to put it would
be in rc.local -- but that has not been successful as a suggestion for this
sort of thing

I've written a PR which creates a "iiab-startup.sh" script in /usr/libexec
that is stimulated by systemd unit file. We can put the check for uuid
there.

holta, 2017-11-27:
TODO, now that PR #559 's been merged:
* Someone please report back if /etc/iiab/uuid is being created on RPi Zero W, Thanks!
* All 5 lines in [roles/2-common/tasks/iiab-startup.yml](https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/iiab-startup.yml#L8-L26) that say:
```
when: startup_unit.stat.exists is defined and not startup_unit.stat.exists 
```
...can likely be abbreviated to:
```
when: not startup_unit.stat.exists 
```

holta, 2018-06-10:
Appears fixed: @tim-moody & @ georgejhunt can you confirm?

closed 2018-06-10 by holta