iiab/iiab #230: Raspbian Stretch 2017-09-07 fixes...firmware AP Mode & more?

holta, 2017-09-08:
@georgejhunt reports that AP mode was broken in Raspbian Stretch's 2017-08-16 firmware, until he swapped in Jessie firmware as follows: https://github.com/iiab/iiab/commit/ee4df858b3b423380b121db256b078e9896ae796

_Hopefully the new Raspbian Stretch (2017-09-07) fixes many things!_  In any case, here's an excerpt from http://downloads.raspberrypi.org/raspbian/release_notes.txt as we get started here:

```
2017-09-07:
  * Disable predictable network interface names for Ethernet devices
  * Bug fix for keyboard settings dialog in Raspberry Pi Configuration
  * Bug fix for crash on some videos and animations in Chromium
  * Bug fix for taskbar crash when running RealVNC server
  * Bug fix for reloading projects with extensions in Scratch 2
  * Bug fix for MAC address problem in Bluetooth
  * Simple mode and new icons in Thonny
  * New Japanese translations in Raspberry Pi Configuration
  * Install fonts-droid-fallback for international fonts
```

jvonau, 2017-09-08:
> reports that AP mode was broken in Raspbian Stretch's 2017-08-16 firmware

Is there an upstream bug filed anywhere? If not, open one so this issue is brought up with the primary developers? I'm concerned about downgrading because of this entry in linked release-notes.
2017-08-16 
  * Wireless firmware for Pi 3 and Pi 0W modified to address Broadpwn exploit

jvonau, 2017-09-08:
July 2017 thread https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=187725 We need to know if the .txt file mentioned is present out of the box in either 08-16 or 09-07 images.

jvonau, 2017-09-08:
https://www.raspberrypi.org/forums/viewtopic.php?t=191061

jvonau, 2017-09-08:
https://www.raspberrypi.org/blog/raspbian-stretch/
> Nick says:	
18th Aug 2017 at 8:33 pm	
install of latest raspbian lite on Pi 0 W
no wireless
errors in syslog :
raspberrypi dhcpcd[262]: wlan0: ctrl_interface not defined in /etc/wpa_supplicant/wpa_supplicant.conf
wired ethernet works

jvonau, 2017-09-08:
should be reversed with #231

jvonau, 2017-09-08:
and update the docs to start out with 2017-09-07

holta, 2017-09-09:
Ref #233 (remove workarounds for august stretch bugs)

holta, 2017-09-13:
@georgejhunt am hereby closing this (or please re-open if there are major outstanding Raspbian Stretch issues?)

closed 2017-09-13 by holta